zephyrproject-rtos/zephyr #89494: kernel: min-heap data structure

pedrohqr, 2025-05-05:
**Is your feature request related to a problem? Please describe.**

The adding of *min-heap* data structure is helpful to supply API functions to developers.

**Describe the solution you'd like**

Implementation of _min-heap_ header file in _kernel_ sections as API development tool to external use.

**Describe alternatives you've considered**

**Additional context**

sayooj-aerlync, 2025-05-16:
I would like to work on this. I will come up with a design proposal soon.

sayooj-aerlync, 2025-05-16:
`**Public Min-Heap API**`
Feature Description for the addition of a lightweight, efficient min-heap data structure as part of Zephyr’s public API. This utility will provide developers with a flexible, general-purpose priority queue mechanism for a variety of use cases.

**`Motivation`**
Currently, Zephyr lacks a built-in, public-facing min-heap implementation. Many embedded applications, including custom schedulers, timer queues, and priority-based resource managers, would benefit from a reusable min-heap structure. By exposing a min-heap API:
* Developers can avoid re-implementing similar logic across projects.
* Implementing system-level applications and libraries becomes easier.
* It aligns with Zephyr’s philosophy of providing modular and reusable components.

**`Proposed Design`**
The min-heap is implemented using a statically allocated array of nodes, where each node holds a `void *data` pointer and a `uint32_t priority`. Lower priority values indicate higher urgency.
```
struct z_heap_node {
    void *data;
    uint32_t priority;
};
struct z_min_heap {
    struct z_heap_node *nodes;
    size_t capacity;
    size_t size;
};
```
A typical usage pattern would look like this:
```
#define CONFIG_MIN_HEAP_CAPACITY 10  // This will go to Kconfig
static struct z_heap_node buffer[CONFIG_MIN_HEAP_CAPACITY];
static struct z_min_heap heap;

z_min_heap_init(&heap, buffer, CONFIG_MIN_HEAP_CAPACITY);
z_min_heap_insert(&heap, my_task, priority);
void *top = z_min_heap_extract_min(&heap);
```
**`API`**
Header: include/sys/min_heap.h
```
void z_min_heap_init(...) — Initialize the heap with a static buffer
int z_min_heap_insert(...) — Insert an item with a priority
void *z_min_heap_peek_min(...) — Peek at the top item
void *z_min_heap_extract_min(...) — Remove the top item
bool z_min_heap_is_empty(...) — Check if the heap is empty
```

**Thread Safety has to be handled by the application layer by using spinlocks, or it can be extended later.**
I would need your suggestions on the following:
* Whether this fits better in lib or kernel
* Any naming or API structure recommendations
* Whether dynamic resizing should be considered (currently avoided for embedded safety) - We can do it as extension if needed.

sayooj-aerlync, 2025-05-19:
https://github.com/zephyrproject-rtos/zephyr/pull/90130

Introduced minimum heap
Locking for thread safety
Helper API to initialise heap.
APIs to Init, Add, delete, peek, re-arrange the memory pool

Please have a look.
Notify @pedrohqr

pedrohqr, 2025-05-19:
It seem's to be OK. One question: why use `malloc()` instead `k_malloc()`?

sayooj-aerlync, 2025-05-21:
Please check : https://github.com/zephyrproject-rtos/zephyr/pull/90251

@pedrohqr @JarmouniA