zephyrproject-rtos/zephyr #89507: doc: Documentation issue in 'samples/subsys/usb/mass/README'

sintrones, 2025-05-06:
**Describe the bug**
Example not buildable

<< Please describe the issue here >>
```
$ west build --force -b adafruit_feather_nrf52840_sense zephyr/samples/subsys/usb/mass -- -DCONFIG_APP_MSC_STORAGE_FLASH_FATFS=y

-- west build: generating a build system
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/wols/Documents/Wols/zephyrproject/zephyr/samples/subsys/usb/mass
-- CMake version: 3.20.5
-- Cache files will be written to: /home/wols/.cache/zephyr
-- Zephyr version: 4.1.0-rc2 (/home/wols/Documents/Wols/zephyrproject/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: xiao_rp2040, qualifiers: rp2040
-- Found host-tools: zephyr 0.17.0 (/home/wols/zephyr-sdk-0.17.0)
-- Found toolchain: zephyr 0.17.0 (/home/wols/zephyr-sdk-0.17.0)
-- Found BOARD.dts: /home/wols/Documents/Wols/zephyrproject/zephyr/boards/seeed/xiao_rp2040/xiao_rp2040.dts
-- Generated zephyr.dts: /home/wols/Documents/Wols/zephyrproject/build/zephyr/zephyr.dts
-- Generated pickled edt: /home/wols/Documents/Wols/zephyrproject/build/zephyr/edt.pickle
-- Generated devicetree_generated.h: /home/wols/Documents/Wols/zephyrproject/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/wols/Documents/Wols/zephyrproject/build/zephyr/dts.cmake
Parsing /home/wols/Documents/Wols/zephyrproject/zephyr/samples/subsys/usb/mass/Kconfig
Loaded configuration '/home/wols/Documents/Wols/zephyrproject/build/zephyr/.config'
No change to configuration in '/home/wols/Documents/Wols/zephyrproject/build/zephyr/.config'
No change to Kconfig header in '/home/wols/Documents/Wols/zephyrproject/build/zephyr/include/generated/zephyr/autoconf.h'
CMake Warning at ../../../../subsys/usb/device/CMakeLists.txt:22 (message):
  CONFIG_USB_DEVICE_VID has default value 0x2FE3.

  This value is only for testing and MUST be configured for USB products.

-- Using ccache: /usr/bin/ccache
CMake Warning at ../../../../CMakeLists.txt:1011 (message):
  No SOURCES given to Zephyr library: drivers__disk

  Excluding target from build.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/wols/Documents/Wols/zephyrproject/build
-- west build: building application
[1/97] Performing build step for 'second_stage_bootloader'
ninja: no work to do.
[15/94] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /home/wols/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -DPICO_RP2040 -DZEPHYR_CONFIG_OVERRIDE=zephyr_fatfs_config.h -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/wols/Documents/Wols/zephyrproject/zephyr/samples/subsys/usb/common -Izephyr/include/generated/zephyr -I/home/wols/Documents/Wols/zephyrproject/zephyr/include -Izephyr/include/generated -I/home/wols/Documents/Wols/zephyrproject/zephyr/soc/raspberrypi/rpi_pico -I/home/wols/Documents/Wols/zephyrproject/zephyr/soc/raspberrypi/rpi_pico/common/. -I/home/wols/Documents/Wols/zephyrproject/zephyr/soc/raspberrypi/rpi_pico/rp2040/. -I/home/wols/Documents/Wols/zephyrproject/zephyr/subsys/usb/device -I/home/wols/Documents/Wols/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/wols/Documents/Wols/zephyrproject/zephyr/modules/cmsis/. -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/common/pico_base_headers/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/boot_bootrom_headers/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_base/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_clocks/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_watchdog/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_xosc/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_pll/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_irq/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_sync/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_timer/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_resets/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_boot_lock/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_ticks/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_sync_spin_lock/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/pico_bootrom/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/pico_flash/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/pico_platform_compiler/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/pico_platform_sections/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/pico_platform_panic/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/common/boot_picoboot_headers/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/common/boot_picobin_headers/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2040/hardware_regs/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2040/hardware_structs/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2040/pico_platform/include -I/home/wols/Documents/Wols/zephyrproject/zephyr/modules/hal_rpi_pico -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_gpio/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_flash/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_xip_cache/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/rp2_common/hardware_pio/include -I/home/wols/Documents/Wols/zephyrproject/modules/hal/rpi_pico/src/common/hardware_claim/include -I/home/wols/Documents/Wols/zephyrproject/zephyr/modules/fatfs -I/home/wols/Documents/Wols/zephyrproject/modules/fs/fatfs/include -isystem /home/wols/Documents/Wols/zephyrproject/zephyr/lib/libc/common/include -fno-strict-aliasing -Os -imacros /home/wols/Documents/Wols/zephyrproject/build/zephyr/include/generated/zephyr/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m0plus -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/home/wols/zephyr-sdk-0.17.0/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/wols/Documents/Wols/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/wols/Documents/Wols/zephyrproject/zephyr/samples/subsys/usb/mass=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/wols/Documents/Wols/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/wols/Documents/Wols/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=picolibc.specs -std=c99 -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj -c /home/wols/Documents/Wols/zephyrproject/zephyr/samples/subsys/usb/mass/src/main.c
/home/wols/Documents/Wols/zephyrproject/zephyr/samples/subsys/usb/mass/src/main.c:36:2: error: #error No supported disk driver enabled
   36 | #error No supported disk driver enabled
      |  ^~~~~
[24/94] Building C object zephyr/drivers/usb/device/CMakeFiles/drivers__usb__device.dir/usb_dc_rpi_pico.c.obj
ninja: build stopped: subcommand failed.
```
<< You may also want to update the automatically generated issue title above. >>

**Environment**

* Page: `samples/subsys/usb/mass/README`
* Version: v4.1.0
* SHA-1: bb4af13da09525926043fc87a6e2ab94b7b850b7

danieldegrasse, 2025-05-06:
@sintrones running `west build --force -b adafruit_feather_nrf52840_sense samples/subsys/usb/mass -- -DCONFIG_APP_MSC_STORAGE_FLASH_FATFS=y` on SHA bb4af13da09 I do not see the error- could you confirm the SHA and build command you are using? It would be great if you could try this on the latest `main` as well

jfischer-no, 2025-05-13:
>west build --force -b adafruit_feather_nrf52840_sense zephyr/samples/subsys/usb/mass -- -DCONFIG_APP_MSC_STORAGE_FLASH_FATFS=y

>-- Board: xiao_rp2040, qualifiers: rp2040

```
west build --help
...
  -f, --force           ignore any errors and try to proceed
...
```

It looks like a user error.