zephyrproject-rtos/zephyr #89537: Json library tests failing on nrf bsim

wearyzen, 2025-05-06:
**Describe the bug**
Test [libraries.encoding.json](https://github.com/zephyrproject-rtos/zephyr/blob/4a31b8036d9c20349a0ee9ab83fa86147fdf983e/tests/lib/json/testcase.yaml#L2) is fails with below error : 

> d_00: @00:00:00.000093  ===================================================================
> d_00: @00:00:00.000093  START - test_json_enums
> d_00: @00:00:00.000093  
> d_00: @00:00:00.000093      Assertion failed at WEST_TOPDIR/zephyr/tests/lib/json/src/main.c:2179: lib_json_test_test_json_enums: (!memcmp(&enums, &enums_decoded, sizeof(enums)) is false)
> d_00: @00:00:00.000093  Enums not decoded correctly
> d_00: @00:00:00.000093   FAIL - test_json_enums in 0.000 seconds
> d_00: @00:00:00.000093  ===================================================================
> 

Issue observed on PR https://github.com/zephyrproject-rtos/zephyr/pull/89370 but can be reproduced on main branch with same steps.
Boards the issue is observed on: 
- nrf54l15bsim/nrf54l15/cpuapp
- nrf5340bsim/nrf5340/cpunet

**To Reproduce**
Run below commands on main branch to reproduce the issue:
```
west twister -p nrf54l15bsim/nrf54l15/cpuapp -s libraries.encoding.json --no-detailed-test-id
west twister -p nrf5340bsim/nrf5340/cpunet -s libraries.encoding.json --no-detailed-test-id

```
or if the test if filtered out in twister then run below command to reproduce the issue:
```
west build -p -b nrf54l15bsim/nrf54l15/cpuapp tests/lib/json/ -t run
west build -p -b nrf5340bsim/nrf5340/cpunet tests/lib/json/ -t run

```

**Expected behavior**

Tests passes and PR is not blocked by this test.

**Impact**
PR https://github.com/zephyrproject-rtos/zephyr/pull/89370 is blocked by this test.

**Logs and console output**
- https://github.com/zephyrproject-rtos/zephyr/actions/runs/14856947600/job/41712752912?pr=89370#step:12:1913
- https://github.com/zephyrproject-rtos/zephyr/actions/runs/14856947600/job/41712752506?pr=89370#step:12:1924

**Environment (please complete the following information):**

 - OS: Github CI or locally on linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used : 7edd59ee5a68459be29c2b77365ee8f7699e7473 from the PR but can also be reproduced with main branch 4a31b8036d9c20349a0ee9ab83fa86147fdf983e

aescolar, 2025-05-06:
Assigning @kartben as he already submitted a patch

kartben, 2025-05-06:
> Assigning [@kartben](https://github.com/kartben) as he already submitted a patch

yeah but I'm a bit confused as to the root cause, even the patch seems to fix things (note that I couldn't reproduce the test_json_enums failure locally, but have been getting a similar one with test_json_limits.

The other failure https://github.com/zephyrproject-rtos/zephyr/actions/runs/14856947600/job/41712751562?pr=89370 is also somewhat suspicious. Might be worthwhile trying to bisect, @wearyzen, if you have a chance? I wonder if some compiler settings have changed or something? Also, why would this fail only on this board? :/

aescolar, 2025-05-06:
> yeah but I'm a bit confused as to the root cause

padding :)
```
struct test_int_limits {
        int                        int_max;              /*     0     4 */
        int                        int_cero;             /*     4     4 */
        int                        int_min;              /*     8     4 */
        int64_t                    int64_max;            /*    12     8 */
        int64_t                    int64_cero;           /*    20     8 */
        int64_t                    int64_min;            /*    28     8 */
        uint64_t                   uint64_max;           /*    36     8 */
        uint64_t                   uint64_cero;          /*    44     8 */
        uint64_t                   uint64_min;           /*    52     8 */
        uint32_t                   uint32_max;           /*    60     4 */
        /* --- cacheline 1 boundary (64 bytes) --- */
        uint32_t                   uint32_cero;          /*    64     4 */
        uint32_t                   uint32_min;           /*    68     4 */
        int16_t                    int16_max;            /*    72     2 */
        int16_t                    int16_cero;           /*    74     2 */
        int16_t                    int16_min;            /*    76     2 */
        uint16_t                   uint16_max;           /*    78     2 */
        uint16_t                   uint16_cero;          /*    80     2 */
        uint16_t                   uint16_min;           /*    82     2 */
        int8_t                     int8_max;             /*    84     1 */
        int8_t                     int8_cero;            /*    85     1 */
        int8_t                     int8_min;             /*    86     1 */
        uint8_t                    uint8_max;            /*    87     1 */
        uint8_t                    uint8_cero;           /*    88     1 */
        uint8_t                    uint8_min;            /*    89     1 */

        /* size: 92, cachelines: 2, members: 24 */
        /* padding: 2 */
        /* last cacheline: 28 bytes */
};
struct test_enums {
        enum int8_enum             i8;                   /*     0     1 */
        enum uint8_enum            u8;                   /*     1     1 */
        enum int16_enum            i16;                  /*     2     2 */
        enum uint16_enum           u16;                  /*     4     2 */

        /* XXX 2 bytes hole, try to pack */

        enum int32_enum            i32;                  /*     8     4 */
        enum uint32_enum           u32;                  /*    12     4 */

        /* size: 16, cachelines: 1, members: 6 */
        /* sum members: 14, holes: 1, sum holes: 2 */
        /* last cacheline: 16 bytes */
};
``` 
> why would this fail only on this board? :/

Chance. It is one of the very few that are runtime tested. And the exact memory layout will determine what garbage happens to be in the stack in that place before

wearyzen, 2025-05-06:
@kartben I didn't get a chance to do a git bisect yet, I'll try it tomorrow and let you know, and the other issue about gpio seems to be a known issue reported [here](https://github.com/zephyrproject-rtos/zephyr/issues/88293) but hasn't been fixed yet.

wearyzen, 2025-05-06:
> > [@kartben](https://github.com/kartben) I didn't get a chance to do a git bisect yet, I'll try it tomorrow and let you know, and the other issue about gpio seems to be a known issue reported [here](https://github.com/zephyrproject-rtos/zephyr/issues/88293) but hasn't been fixed yet.
> 
> yeah nevermind, don't bother with bisecting, based on [@aescolar](https://github.com/aescolar)'s analysis it just happens that we probably never hit this in CI yet is all. Feel free to check if the linked PR fixes your issue (esp. as I couldn't necessarily reproduce the failure locally) and I will be happy to merge it as a hotfix if that's the case.

👍 I can confirm that PR #89542 fixes this issue locally

closed 2025-05-06 by kartben