zephyrproject-rtos/zephyr #89545: tests/cmake/yaml leaves test data files in the tree

aescolar, 2025-05-06:
**Describe the bug**
Running tests/cmake/yaml creates and leaves test data files in the tree.
```
tests/cmake/yaml/test_setting_escaped_genexes_1.yaml
tests/cmake/yaml/test_setting_genexes_1.yaml
tests/cmake/yaml/test_setting_genexes_2.yaml
```

**To Reproduce**
1. twister -T tests/cmake/yaml/
2. git status

**Expected behavior**
No funny files in the tree

**Impact**
Annoyance for developers

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain (e.g Zephyr SDK, ...) cmake 3.28
 - Commit SHA or Version used main as of now 72509c385e20410eebdc09d8ca60a206a063c064

**Additional context**
3741a0b74441
a90870da4c67

aescolar, 2025-05-06:
CC @tejlmand

pillo79, 2025-05-06:
@aescolar thanks for pointing this out. The linked PR fixes this by moving the intermediate files in the current binary directory.

PR #89548 (2025-05-08): yaml: save intermediate files in the build directory
Make sure to provide full paths when saving the intermediate files in `yaml_save`, to prevent them from being saved in the current (source) directory. Fixes #89545.