zephyrproject-rtos/zephyr #89557: NXP LPSPI Async implementation very dodgy

Finomnis, 2025-05-06:
**Describe the bug**
I'm on a custom imxrt1176 and noticed that LPSPI and tried to debug the experimental fpga_ice40 driver.

I realized that there's a very short `spi_write_dt` followed by quite a long `sp_write_dt`; and because the driver controls CS manually, it relies on the fact that `spi_write_dt` is blocking. It, however, did not seem to block for the second longer transfer.

I decided to do a code review of the driver combined with printf debugging and stumbled across multiple issues in `spi_nxp_lpspi.c`; to be specific, surrounding the `lpspi_isr` function:

- `lpspi_end_xfer(dev)` gets called when `spi_context_rx_len_left(ctx) == 0`, regardless of whether interrupts are still enabled or not. This can cause `lpspi_end_xfer` to be called multiple times during a single transfer, which preemptively flags the next future transfer as finished. Which is exactly what happened in my case.
  To be specific, the `RDIE` interrupt was still enabled. `RDIE` gets disabled in `lpspi_handle_rx_irq`, but no `RDF` interrupt happened yet.
- The interrupt handler calls `lpspi_end_xfer`, which in turn calls `lpspi_wait_tx_fifo_empty`. Is a blocking call inside of an ISR really intended?
- In `lpspi_end_xfer`, the `spi_context_complete` gets called **before** `lpspi_wait_tx_fifo_empty`. This means if `CONFIG_SPI_ASYNC` is on, the `spi_write` function returns **before** all bytes have physically been transmitted on the bus. Which is incompatible with the note `This function is synchronous.` on `spi_transceive, on which other drivers rely that, like in the ICE40 case, handle CS manually.

**Expected behavior**
- Interrupts return immediately without blocking
- The `lpspi_end_xfer` function only gets called once
- Once `spi_write` returns, the physical transfer on the bus is completed

**Impact**
Crashes, hangs, undefined behavior of peripherals

**Logs and console output**
If required, I can attempt to reproduce on an imxrt1176 dev board, but I don't have one on me right now.

**Environment (please complete the following information):**

 - OS: Windows
 - Toolchain: Zephyr SDK 0.17.0
 - Zephyr Git Hash: 8b1f06af052f465e7d7094a49764d6aba76b4655

Finomnis, 2025-05-06:
I feel like all of those problems boil down to these base issues:

- Things happen in the interrupt handler that are not connected to interrupt flags. The proper way would be to only have an if/else cascade that checks for the various interrupt sources (the SR flags), and everything that happens is directly linked to the SR status flags.
- The end-of-transfer condition is computed in a weird way that is not related to the ISR at all, imo, causing the mentioned `wait` call inside of the ISR. The proper way would be to utilize the `MBF` (module busy) flag, which does not have a interrupt, though; or the `WCF`/`FCF`/`TCF`  (word/frame/transfer complete) flags, which do have an interrupt and are therefore the proper solution in my opinion.

decsny, 2025-05-06:
> * The end-of-transfer condition is computed in a weird way that is not related to the ISR at all, imo, causing the mentioned `wait` call inside of the ISR. The proper way would be to utilize the `MBF` (module busy) flag, which does not have an interrupt, though; or the `WCF`/`FCF`/`TCF`  (word/frame/transfer complete) flags, which do have a interrupts and are therefore the proper solution in my opinion.

Not possible to use those flags because LPSPI hardware transfer does not mean the same thing as zephyr spi bus transfer. I wish it were that simple, but no.

> * Things happen in the interrupt handler that are not connected to interrupt flags. The proper way would be to only have an if/else cascade that checks for the various interrupt sources (the SR flags), and everything that happens is directly linked to the SR status flags.

The only flags we are using is data ready for rx and data request for tx. This is because zephyr transfer can be arbitrarily long, and therefore we need to use the continuous hardware mode of LPSPI, so that the hardware doesnt deassert CS on frame boundaries. So the other flags can't be used. It's not that simple, otherwise I would have done it that way.

decsny, 2025-05-06:
> * which in turn calls `lpspi_wait_tx_fifo_empty`. Is a blocking call inside of an ISR really intended?

it is not a very consequential block time wise, but very necessary, the point is to make sure the transmit command was consumed by the LPSPI hardware before continuing. It isn't waiting for actual data to be sent on the bus or anything

Finomnis, 2025-05-06:
> It isn't meant to be used for waiting for actual data to be sent on the bus or anything

I absolutely agree. However, in the current implementation it does actually wait for the data to be sent on the bus, because it gets called too early. My change fixed that.

PR #89559 (2025-05-16): drivers: spi: nxp_lpspi: Fix race condition in ISR
Fixes #89557.

The main architectural change is that the RX/TX halves of a transmission now signal that they are finished by disabling their interrupts.

This in turn will then be picked up further down in the ISR as the signal that the entire transmission is over.

The order of operations inside the `handle_tx_irq` had to be changed, so that the interrupt only gets disabled once both of those are fulfilled:
- no more TX data available
- physical transmission finished

Before, the IRQ got disabled while the physical transmission was not finish [... cut]