zephyrproject-rtos/zephyr #89561: clang-format requests fixing lines that are not part of change but surround it in diff

de-nordic, 2025-05-06:
**Describe the bug**
When running  ./scripts/ci/check_compliance.py on changes to be posted for review clang format requests fixing lines that the commits do not actually change.
I am using here https://github.com/zephyrproject-rtos/zephyr/pull/88467 PR to as an example, after rebasing it to Zephyr main 2707fa2fcaf83cf21ba7f493d339b1783463f2c8.

Using command:
`git diff HEAD~3`
from Zephyr root, I got diff that contains:

[image]

Where you can clearly see that macro `spi_nor_cmd_write` has not been altered; yet, when compliance check is run with:
`git diff HEAD~3 | ./scripts/ci/check_compliance.py`
part of output contains:

[image]
which errors out on the same macro and requests fixing it, even though author of the commit did not touch the line.

**To Reproduce**
```
git fetch origin pull/88467/head
git checkout -b check_88467 FETCH_HEAD
git fetch origin main
git rebase origin/main
git diff HEAD~3 | ./scripts/ci/check_compliance.py
```

You probably also want to run pip update with the `scripts/requirements-complience.txt` as input.
Note that you now have extra branch `check_88467 ` you probably want to remove.

**Expected behavior**
Stick to changed lines only.

**Impact**
Getting annoyed, by request to fix things that have not been touched, with failing compliance check.

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain 
 - Commit SHA: https://github.com/zephyrproject-rtos/zephyr/pull/88467 rebased on 2707fa2fcaf83cf21ba7f493d339b1783463f2c8

pdgendt, 2025-05-14:
This check is excluded in CI, you can exclude it locally too?

de-nordic, 2025-05-14:
> This check is excluded in CI, you can exclude it locally too?

That can probably be done, but the issue with taking incorrect lines from diff remains.

closed 2025-09-28 by github-actions[bot]