zephyrproject-rtos/zephyr #89581: MCXN947 socket network transfer content is wrong

groncarolonxp, 2025-05-07:
**Describe the bug**

If I send 1KB of data to echo server demo on MCXN947 the board does not answer with the same content

** steps **

- compile eco demo
- west build --pristine -b frdm_mcxn947/mcxn947/cpu0  zephyr/samples/net/sockets/echo/
- telnet 192.168.10.2 4242
- copy paste the content of the file ( ~1000 0 characters)
- capture the network traffic with wireshark 

Below the packet going to the board (wireshare text dump)

[error_1.txt](https://github.com/user-attachments/files/20077028/error_1.txt)

[error_1.txt](https://github.com/user-attachments/files/20077033/error_1.txt)
[error_mnxn947.pcapng.txt](https://github.com/user-attachments/files/20077032/error_mnxn947.pcapng.txt)
[zeros.txt](https://github.com/user-attachments/files/20077034/zeros.txt)

 ```

No.     Time           Source                Destination           Protocol Length Info

      1 0.000000000    192.168.10.1          192.168.10.2          TCP      822    46770 → 4242 [ACK] Seq=1 Ack=1 Win=63984 Len=768

 

Frame 1: 822 bytes on wire (6576 bits), 822 bytes captured (6576 bits) on interface enx3c8cf8ea5203, id 0

Ethernet II, Src: TRENDnet_ea:52:03 (3c:8c:f8:ea:52:03), Dst: ae:9a:22:97:c9:16 (ae:9a:22:97:c9:16)

Internet Protocol Version 4, Src: 192.168.10.1, Dst: 192.168.10.2

Transmission Control Protocol, Src Port: 46770, Dst Port: 4242, Seq: 1, Ack: 1, Len: 768

Data (768 bytes)

 

0000  30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000

0010  30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000

0020  30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000

0030  30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000

0040  30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000

0050  30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000

0060  30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000

0070  30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000

0080  30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000

0090  30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000

00a0  30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000

00b0  30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000

00c0  30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000

00d0  30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000

00e0  30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000

[... 350 lines cut ...]
xyz:~/siemens/os$ telnet 192.168.10.2 4242

Trying 192.168.10.2...

Connected to 192.168.10.2.

Escape character is '^]'.

frdm_rw612

frdm_rw612

00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

**Expected behavior**
the content answered should be the same

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK
 - V4.1.0

theadib, 2025-05-07:
Hello @groncarolonxp what is the maximum size of the sentence that is echoed correctly?

when I copy paste in my telnet 1000 bytes my FRDM_MXCN947 get stuck with no response.
I need to go down to 74 characters in 1(one) chunk. Then it will echo correctly.

groncarolonxp, 2025-05-12:
Hello @theadib, in case of RW612 sending 981 charactehs works OK.
In case of MCXN947 i would say it starts breaking with 500 charactes in my case.

I open this with the echo example because it is the fastest way to cho the issue, futhermore the same exact test with 1000 chars works OK on another platform, but you optain the same result opening a socket and transferring binary data

groncarolonxp, 2025-05-12:
Hi,

I tested again adding DBG log and increasing log stack 
```

[579.062893 0.943418] [00Connection #0 from ::ffff:192.168.10.1
[579.066433 0.003540] :00:06.895,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Created RX pkt 0x3000bd0c
[579.075030 0.008597] [00:00:06.895,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Receiving RX packet
[579.083503 0.008473] [00:00:06.895,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Recycling RX buf
[579.091573 0.008070] [00:00:06.895,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: Setting up TX descriptors for packet 0x3000b954
[579.102383 0.010810] [00:00:06.895,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: Starting TX DMA on packet 0x3000b954
[579.112408 0.010025] [00:00:06.895,000] <dbg> eth_nxp_enet_qos_mac: tx_dma_done: TX DMA completed on packet 0x3000b954
[579.121494 0.009086] [00:00:06.895,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Created RX pkt 0x3000bd0c
[579.130376 0.008882] [00:00:06.895,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Receiving RX packet
[579.138784 0.008408] [00:00:06.895,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Recycling RX buf
[579.146905 0.008121] [00:00:06.895,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: Setting up TX descriptors for packet 0x3000b9dc
[579.157726 0.010821] [00:00:06.895,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: Starting TX DMA on packet 0x3000b9dc
[579.167603 0.009877] [00:00:06.895,000] <dbg> eth_nxp_enet_qos_mac: tx_dma_done: TX DMA completed on packet 0x3000b9dc
[579.176852 0.009249] [00:00:06.896,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Created RX pkt 0x3000bd0c
[579.185766 0.008914] [00:00:06.896,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Receiving RX packet
[579.194158 0.008392] [00:00:06.896,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Recycling RX buf
[583.931367 4.737210] [00:00:11.765,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Created RX pkt 0x3000bd0c
[583.940095 0.008728] [00:00:11.765,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Receiving RX packet
[583.948526 0.008431] [00:00:11.765,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Recycling RX buf
[583.956635 0.008109] [00:00:11.766,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Created RX pkt 0x3000bd94
[583.965554 0.008919] [00:00:11.766,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Receiving RX packet
[583.973927 0.008373] [00:00:11.766,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Recycling RX buf
[583.982071 0.008144] [00:00:11.766,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: Setting up TX descriptors for packet 0x3000b9dc
[583.992884 0.010813] [00:00:11.766,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: Starting TX DMA on packet 0x3000b9dc
[584.002766 0.009882] [00:00:11.766,000] <dbg> eth_nxp_enet_qos_mac: tx_dma_done: TX DMA completed on packet 0x3000b9dc
[584.012017 0.009251] [00:00:11.766,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: Setting up TX descriptors for packet 0x3000b9dc
[584.022834 0.010816] [00:00:11.766,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: Starting TX DMA on packet 0x3000b9dc
[584.032679 0.009845] [00:00:11.766,000] <dbg> eth_nxp_enet_qos_mac: tx_dma_done: TX DMA completed on packet 0x3000b9dc
[584.041946 0.009267] [00:00:11.766,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Created RX pkt 0x3000bdd8
[584.050850 0.008904] [00:00:11.766,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Receiving RX packet
[584.059242 0.008392] [00:00:11.766,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Recycling RX buf
[584.067376 0.008134] [00:00:11.766,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Created RX pkt 0x3000be1c
[584.076301 0.008925] [00:00:11.766,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Receiving RX packet
[584.084696 0.008395] [00:00:11.766,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Recycling RX buf
[584.092833 0.008137] [00:00:11.766,000] <err> eth_nxp_enet_qos_mac: TX packet too large
[584.099675 0.006841] [00:00:11.767,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: Setting up TX descriptors for packet 0x3000b9dc
[584.110472 0.010797] [00:00:11.767,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: Starting TX DMA on packet 0x3000b9dc
[584.120336 0.009864] [00:00:11.767,000] <dbg> eth_nxp_enet_qos_mac: tx_dma_done: TX DMA completed on packet 0x3000b9dc
[585.199599 1.079264] [00:00:12.035,000] <err> eth_nxp_enet_qos_mac: TX packet too large
[585.206006 0.006407] [00:00:12.439,000] <err> eth_nxp_enet_qos_mac: TX packet too large
[585.212968 0.006962] [00:00:13.043,000] <err> eth_nxp_enet_qos_mac: TX packet too large
[587.112347 1.899379] [00:00:13.949,000] <err> eth_nxp_enet_qos_mac: TX packet too large
[588.470949 1.358602] [00:00:15.309,000] <err> eth_nxp_enet_qos_mac: TX packet too large
[590.507859 2.036910] [00:00:17.347,000] <err> eth_nxp_enet_qos_mac: TX packet too large
[593.563269 3.055410] [00:00:20.404,000] <err> eth_nxp_enet_qos_mac: TX packet too large
[598.145687 4.582418] [00:00:24.990,000] <err> eth_nxp_enet_qos_mac: TX packet too large
[604.288515 6.142828] error: recv: 116
[604.289671 0.001155] Connection from ::ffff:192.168.10.1 closed
[605.018828 0.729157] [00:00:31.867,000] <err> eth_nxp_enet_qos_mac: TX packet too large
```

groncarolonxp, 2025-05-12:
the problem comes from the `frags_count`

```
[970.646792 0.008649] [00:00:07.505,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: RNC NUM_TX_BUFDESC 4
[970.655392 0.008600] [00:00:07.505,000] <err> eth_nxp_enet_qos_mac: TX packet too large 438 frags_count 5
[970.663770 0.008378] [00:00:07.505,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Created RX pkt 0x3000bdd8
[970.672694 0.008924] [00:00:07.505,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Receiving RX packet
[970.681109 0.008415] [00:00:07.505,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_rx: Recycling RX buf
[970.689208 0.008099] [00:00:07.505,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: RNC max_frame_len 1518
[970.697874 0.008667] [00:00:07.505,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: RNC NUM_TX_BUFDESC 4
[970.706353 0.008479] [00:00:07.505,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: Setting up TX descriptors for packet 0x3000b9dc
[970.717176 0.010823] [00:00:07.505,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: Starting TX DMA on packet 0x3000b9dc
[970.727044 0.009868] [00:00:07.505,000] <dbg> eth_nxp_enet_qos_mac: tx_dma_done: TX DMA completed on packet 0x3000b9dc
[970.736299 0.009255] [00:00:07.715,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: RNC max_frame_len 1518
[970.744940 0.008641] [00:00:07.715,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: RNC NUM_TX_BUFDESC 4
[970.753419 0.008479] [00:00:07.715,000] <err> eth_nxp_enet_qos_mac: TX packet too large 438 frags_count 5
[971.967121 1.213702] [00:00:08.030,000] <dbg> eth_nxp_enet_qos_mac: eth_nxp_enet_qos_tx: RNC max_frame_len 1518
```

if I define it as 
```
CONFIG_ETH_NXP_ENET_QOS_TX_BUFFER_DESCRIPTORS=8
```
I have no errors but the content is still wrong

as a ref this is what i added in my proj.conf
```
CONFIG_LOG_MODE_IMMEDIATE=y
CONFIG_LOG=y
CONFIG_ETHERNET_LOG_LEVEL_DBG=y
CONFIG_LOG_PROCESS_THREAD_STACK_SIZE=8192
CONFIG_ETH_NXP_ENET_QOS_TX_BUFFER_DESCRIPTORS=8
```

groncarolonxp, 2025-05-12:
using this
```
CONFIG_ETH_NXP_ENET_QOS_RX_BUFFER_DESCRIPTORS=16
CONFIG_ETH_NXP_ENET_QOS_TX_BUFFER_DESCRIPTORS=16
CONFIG_NET_PKT_RX_COUNT=16
CONFIG_NET_PKT_TX_COUNT=16
CONFIG_NET_BUF_RX_COUNT=64
CONFIG_NET_BUF_TX_COUNT=64
CONFIG_NET_BUF_DATA_SIZE=96
```

the resuls is correct 
```
tito@nostromo:~$ telnet 192.168.10.2 4242
Trying 192.168.10.2...
Connected to 192.168.10.2.
Escape character is '^]'.
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

theadib, 2025-05-12:
Hi @decsny this is maybe also related to MCXN driver.

I could see potentially 3 main issues there:
1. buffer size overflow on DMA RX transfer
  the DMA RX buffer size is initialized in `ener_qos_rx_desc_init()` 
  `base->DMA_CH[0].DMA_CHX_RX_CTRL |=
		/* Set DMA receive buffer size. The low 2 bits are not entered to this field. */
		ENET_QOS_REG_PREP(DMA_CH_DMA_CHX_RX_CTRL, RBSZ_13_Y, NET_ETH_MAX_FRAME_SIZE >> 2);`
 but the target buffer size is allocated by:
`buf = net_pkt_get_reserve_rx_data(CONFIG_NET_BUF_DATA_SIZE, K_NO_WAIT);`
CONFIG_NET_BUF_DATA_SIZE <-- 0x100
NET_ETH_MAX_FRAME_SIZE <-- (NET_ETH_MTU + _NET_ETH_MAX_HDR_SIZE)
**NET_ETH_MAX_FRAME_SIZE is greater than CONFIG_NET_BUF_DATA_SIZE**

2. order of processing DMA buffer in RX queue `eth_nxp_enet_qos_rx()`
  - the DMA buffers is a linear array of NUM_RX_BUFDESC
  - the DMA engine fills from tail buffer position
  - the filled buffers are from head position to tail position
  - but the workqueue that reads the buffer out reads the buffers **from 0 to MAX and not from head to tail**

3. The workqueue has suspicious error handling `eth_nxp_enet_qos_rx`
   - the RX workqueue loops all buffers
   - when it finds an error i.e. can not allocate buffer it completely quits and returns
   - handle the individual error and continue to process the other buffer elements

This is my personal feeling. I did touched a bit (point 3) but overall I have no solution.
I test using a lot (NUM_RX_BUFDESC) of individual packets #89435
The MCXN947 is new to me and the datasheet is a monster, so it is quit difficult to understand the way of working ...

Adib

ekwus, 2025-05-12:
Hey

I've got this same error "TX Buffer Too Large" in the http_server sample and in our own i.MX RT1062 tested to be working code.

I'm planning to do a "send only" and "receive only" test to see if it is inbound, outbound or both directions but I'll add @groncarolonxp configuration changes tonight and see how/if it changes my behaviour.

Cheers

Dave

PR #91312 (2025-06-11): eth_nxp_enet_qos fixes on MCXN947
This PR fixes the following in ENET QoS driver:

- out of order processing of DMA-prepared buffers (previously started from index 0, now from the next position prepared by DMA in the ring of descriptors)
- receiving of frames split into multiple buffers (previously dropped, now net_pkt is composed of multiple buffers if frame is longer than single buffer size now and passed to network stack)
- increases the default configuration for number of buffers, so that at least 1 frame of maximum size can be received (from 4 to 16)

Together with https://github.com/zeph [... cut]
closed 2025-06-11 by dkalowsk