zephyrproject-rtos/zephyr #89623: Null pointer is not checked in adc_nrfx_saadc.c check_buffer_size()

otzy, 2025-05-07:
https://github.com/zephyrproject-rtos/zephyr/blob/77bb19c2a94a4c528a7e817c48b712825d570224/drivers/adc/adc_nrfx_saadc.c#L469

zephyr documentation says that sequence->options can be null, but here this is not checked, so `neede_buffer_size` is sometimes extremely big and adc call fails.

pdgendt, 2025-05-08:
Please use the bug issue template to report these issues.

pdgendt, 2025-05-08:
I don't get the issue though, the pointer is checked right in the line above:

https://github.com/zephyrproject-rtos/zephyr/blob/77bb19c2a94a4c528a7e817c48b712825d570224/drivers/adc/adc_nrfx_saadc.c#L468-L470

otzy, 2025-05-08:
hm, you are right, I did not read carefully enough, but this means that the options is assigned somewhere else? I could not find where, but I observed extra_samplings values in the range from 5000 to 47000 and there were pretty random.
And it was different after each call of adc_read()

To make it working I had to create and assign the options object manually like this:
```
    struct adc_sequence_options options = {};
    struct adc_sequence sequence;
    sequence.options = &options;
```

otzy, 2025-05-08:
I use NRF Connect SDK 2.9.1, board nrf52840dongle