zephyrproject-rtos/zephyr #89684: MQTT-SN configuration

mpecar, 2025-05-08:
In a recent change, CONFIG_MQTT_SN_KEEPALIVE introduced new upper limit of uint8_t - 255.

There are two problems with this:
1. The configuration option should accept values up to uint16_t
2. 0 should be a valid option and keep alive should be disabled in this case

Reference: v1.2 spec:
	•	Section 5.2.6 – CONNECT
“The duration of the Keep Alive Timer is application specific. If the value is 0, the timer is disabled.”
“Keep Alive Timer is a 2-octet unsigned integer representing the number of seconds between the point at which the Client finishes transmitting one Control Message and the point it starts sending the next.”

henrikbrixandersen, 2025-05-11:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

mpecar, 2025-05-11:
@henrikbrixandersen 

Describe the bug
CONFIG_MQTT_SN_KEEPALIVE needs to accept values from 0 to 65535, it is only accepting values from 1 to 255.

To Reproduce
Set the value outside the enforced range. It is a recent change, it used to accept values from 1 to 65535, which is also wrong, but better than current state.

Expected behavior
Reference: v1.2 spec:
• Section 5.2.6 – CONNECT
“The duration of the Keep Alive Timer is application specific. If the value is 0, the timer is disabled.”
“Keep Alive Timer is a 2-octet unsigned integer representing the number of seconds between the point at which the Client finishes transmitting one Control Message and the point it starts sending the next.”

Impact
Increased unwanted traffic for PING requests

Logs and console output
-

Environment (please complete the following information):

OS: (e.g. Linux, MacOS, Windows)
MacOS

Toolchain (e.g Zephyr SDK, ...)
NCS 3.0.0

Commit SHA or Version used
Zephyr v4.0.99-ncs1

Additional context
-

rlubos, 2025-05-12:
> “The duration of the Keep Alive Timer is application specific. If the value is 0, the timer is disabled.”

@mpecar Could you clarify the source of this quote? I've been checking the spec here:
https://groups.oasis-open.org/higherlogic/ws/public/document?document_id=66091

But I could find no mention of **disabling** Keep Alive timer for MQTT-SN. Extending the config range to `65535` is quite obvious, but I'm not so sure if we should allow to disable Keep Alive mechanism for MQTT-SN specifically given it's UDP based.

mpecar, 2025-05-12:
@rlubos 
Sorry, section 5.2.6 does not even exist in MQTT-SN specs. Perhaps I mixed up the specs with MQTT specs or ChatGPT mislead me...

Anyway, extending this value back to 2 bytes also solves the issue.

rlubos, 2025-05-12:
Since the documentation is a bit vague in this regard, I've checked what Paho MQTT-SN client does (https://github.com/eclipse-paho/paho.mqtt-sn.embedded-c/) and it does seem to allow to disable keep alive. I guess we could do the same in our implementation.

mpecar, 2025-05-12:
Well, AFAIK if gateway does not know the client disconnected, the unconfirmed message to the client will be lost. OTOH there will always be some time when the gateway does not know that the client disconnected, so...

While you mention Paho, one more observation from my side: By default, Paho uses port 10000 for data and 1883 for discovery and Zephyr MQTT-SN sample does the same. IMO this is probably a typo in default Paho config and 1883 should be used for data (as is for MQTT) and 10000 for discovery.

rlubos, 2025-05-12:
> While you mention Paho, one more observation from my side: By default, Paho uses port 10000 for data and 1883 for discovery and Zephyr MQTT-SN sample does the same. IMO this is probably a typo in default Paho config and 1883 should be used for data (as is for MQTT) and 10000 for discovery.

IDK, that would be a lot of typos on the Paho side (this port configuration is referenced in several places in the repo), and MQTT-SN seems to leave such details to the applications to decide (it makes no mention on recommended port). Anyway, the sample provides Kconfigs to reconfigure it if needed, but I would keep the defaults as they are to retain interoperability.

mpecar, 2025-05-12:
I would also not change anything, it works out of the box as is. Just wanted to say that default ports are "strange".

dkalowsk, 2025-05-13:
Closing as it does not appear to be a bug.

mpecar, 2025-05-13:
@dkalowsk @kartben 
What is described in the issue is a bug and @rlubos confirmed it.
Just the port selection is not a bug/problem.

rlubos, 2025-05-14:
And it will be fixed with https://github.com/zephyrproject-rtos/zephyr/pull/89825.

closed 2025-05-14 by kartben