zephyrproject-rtos/zephyr #89687: Display Issue with LVGL Basic Sample on ESP32-C3 and SSD1306 in Zephyr 4.1.0

r-pfitzner-truma, 2025-05-08:
Description: When using the LVGL basic sample with ESP32-C3 and SSD1306 display in Zephyr 3.7.1, the label "Hello World" is correctly shown on the display. However, in Zephyr 4.1.0, only a white screen is displayed. Additionally, I2C communication shows that in Zephyr 3.7.1, the write pixel data for the text is transmitted correctly, while in Zephyr 4.1.0, only 0xFF as pixel data is transmitted.

Steps to Reproduce:

Build the LVGL basic sample with the overlay for SSD1306:
```
west build -b esp32c3_devkitm samples/subsys/display/lvgl/ -- -DDTC_OVERLAY_FILE=/path/to/overlay/esp32c3_devkitm.overlay -DCONFIG_DISPLAY=y -DCONFIG_LVGL=y -DCONFIG_LV_COLOR_DEPTH_1=y -DCONFIG_LV_Z_BITS_PER_PIXEL=1 -DCONFIG_LV_Z_VDB_SIZE=64
```

Overlay File Used:
```
/ {
	chosen {
		zephyr,display = &ssd1306;
	};
};

&pinctrl {
	i2c0_default: i2c0_default {
		group1 {
			pinmux = <I2C0_SDA_GPIO0>,
					 <I2C0_SCL_GPIO1>;
			bias-pull-up;
			drive-open-drain;
			output-high;
		};
	};
};

&i2c0 {
	status = "okay";
	clock-frequency = <I2C_BITRATE_STANDARD>;
	pinctrl-0 = <&i2c0_default>;
	pinctrl-names = "default";
	ssd1306: ssd1306@3c {
		compatible = "solomon,ssd1306fb";
		reg = <0x3c>;
		width = <128>;
		height = <64>;
		segment-offset = <0>;
		page-offset = <0>;
		display-offset = <0>;
		multiplex-ratio = <63>;
		segment-remap;
		com-invdir;
		prechargep = <0x22>;
	};
};
```

Run the build with Zephyr 4.1.0 and observe the white screen.
Run the build with Zephyr 3.7.1 and observe the "Hello World" label displayed correctly.
Expected Behavior: The label "Hello World" should be displayed on the SSD1306 screen in Zephyr 4.1.0, as it is in Zephyr 3.7.1.

Actual Behavior: In Zephyr 4.1.0, only a white screen is displayed, and I2C communication shows 0xFF as pixel data instead of the correct write pixel data for the text.

Environment:

Zephyr SDK: 0.17.0
Zephyr Versions: 3.7.1 and 4.1.0
Board: ESP32-C3 DevKitM
Display: SSD1306

Attached I2C Data Log:
[Zephyr-3.7.1.txt](https://github.com/user-attachments/files/20110373/Zephyr-3.7.1.txt)
[Zephyr-4.1.0.txt](https://github.com/user-attachments/files/20110374/Zephyr-4.1.0.txt)

faxe1008, 2025-05-09:
@r-pfitzner-truma I just tested it on the native_sim since I do not own a SSD1306, did not encounter an issue. Can you maybe try the basic display sample to rule out any issues with the display driver?

r-pfitzner-truma, 2025-05-09:
build the display example with:

```
west build -b esp32c3_devkitm samples/drivers/display/ -- -DDTC_OVERLAY_FILE=/path/to/overlay/esp32c3_devkitm.overlay -DCONFIG_HEAP_MEM_POOL_SIZE=16384 -DCONFIG_LOG=y -DCONFIG_DISPLAY=y
```

The result is that I'm seeing a white cross, and the bottom left corner is flickering with rectangles

[image]

[image]

shiwzhi, 2025-05-09:
https://github.com/zephyrproject-rtos/zephyr/discussions/51023#discussioncomment-3826394 I need to set these configs when use LVGL and SSD1306.
```
CONFIG_LV_USE_THEME_DEFAULT=y
CONFIG_LV_Z_VDB_SIZE=64
CONFIG_LV_DPI_DEF=116
CONFIG_LV_Z_BITS_PER_PIXEL=1
CONFIG_LV_COLOR_DEPTH_1=y
```

r-pfitzner-truma, 2025-05-09:
nice the CONFIG_LV_USE_THEME_DEFAULT did the trick!
with CONFIG_LV_COLOR_DEPTH_1 && !CONFIG_LV_CONF_MINIMAL the default Theme used is CONFIG_LV_USE_THEME_MONO this will cause the white screen.

r-pfitzner-truma, 2025-05-09:
@faxe1008  I tried adjusting the threshold, but it doesn't change the appearance. The only difference is that when the threshold is too low, all pixels are interpreted as off

sylvioalves, 2025-05-15:
I guess this isn't SoC/board issue, so I'll remove the platform label. Add again if needed.

faxe1008, 2025-05-30:
@Snevzor try using `CONFIG_LV_USE_THEME_DEFAULT ` or `CONFIG_LV_USE_THEME_SIMPLE `

faxe1008, 2025-05-30:
Zephyr 3.7
[image]
Zephyr 4.1
[image]

I tested the font aliasing with both versions and yes there is a difference. Both versions will default to montserrat font at 14 size, and the glyph bitmap is exactly the same in both versions, so it comes down to rendering:
```
0xe, 0xa0, 0xd9, 0xd, 0x90, 0xc8, 0xc, 0x80,
    0xb7, 0xa, 0x60, 0x11, 0xb, 0x80, 0xd9,
```

Snevzor, 2025-08-02:
> Zephyr 3.7 [image] Zephyr 4.1 [image]
>
> I tested the font aliasing with both versions and yes there is a difference. Both versions will default to montserrat font at 14 size, and the glyph bitmap is [...]

Any update on this sub-issue?

closed 2025-12-09 by github-actions[bot]