zephyrproject-rtos/zephyr #89724: Review rules in Sonarqube

nashif, 2025-05-09:
List of rules that should  considered for addition / removal in the sonaqube tool:

Proposal to add:

-

Proposl be remove:

- [Redundant pairs of parentheses should be removed
](https://sonarcloud.io/organizations/zephyrproject-rtos/rules?open=c%3AS1110&rule_key=c%3AS1110)

jhedberg, 2025-05-09:
I'd propose to remove this:

 - [Array declarations should include an explicit size specification](https://sonarcloud.io/organizations/zephyrproject-rtos/rules?open=cpp%3AS834&q=array+size)

The reason being that this adds redundancy and risk of bugs in most cases where a human has to correctly count the elements and make sure the initializer size matches the size of given initializer elements. Some discussion about it here: https://github.com/zephyrproject-rtos/zephyr/pull/87450#discussion_r2081157914

nashif, 2025-05-09:
> I'd propose to remove this:
> 
> * [Array declarations should include an explicit size specification](https://sonarcloud.io/organizations/zephyrproject-rtos/rules?open=cpp%3AS834&q=array+size)
> 
> The reason being that this adds redundancy and risk of bugs in most cases where a human has to correctly count the elements and make sure the initializer size matches the size of given initializer elements. Some discussion about it here: [#87450 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/87450#discussion_r2081157914)

if we agree on this, we need to list this for removal from the [coding guidelines.](https://docs.zephyrproject.org/latest/contribute/coding_guidelines/index.html) as well.

nashif, 2025-05-12:
> * ["#undef" should be used with caution](https://sonarcloud.io/organizations/zephyrproject-rtos/rules?open=c%3AS959&rule_key=c%3AS959) (undefining not defined macros is safe, forcing users to check if a macro is defined before undefining it just adds pointless verbosity to code)

ok, although this is a misra rule, it is not in the subset of rules we maintain, deactivating

dottspina, 2025-05-12:
Probably not relevant for Zephyr since ["#undef" should be used with caution](https://sonarcloud.io/organizations/zephyrproject-rtos/rules?open=c%3AS959&rule_key=c%3AS959)  is deactivated, but I don't know where to report that (I do not have access to MISRA publications, nor am I a Sonar customer).

IMHO there's a typo in the example of compliant use:
``` c
#define LEVEL(i) int const i = #i
LEVEL(Debug);
LEVEL(Warning);
LEVEL(Error);
#undef LEVEL
```

Shouldn't it be something like `#define LEVEL(l) char const *l = #l` ?

Thalley, 2025-05-20:
I can't seem to find a check for https://gitlab.com/MISRA/MISRA-C/MISRA-C-2012/Example-Suite/-/blob/master/R_14_04.c , but it's pretty clearly not being handling by Sonar

Thalley, 2026-01-06:
Not sure if we need to modify the macro in Zephyr or do something in Sonar, but there are a lot of macros like this that gets warnings from Sonar: 

[image]

Thalley, 2026-02-12:
> Not sure if we need to modify the macro in Zephyr or do something in Sonar, but there are a lot of macros like this that gets warnings from Sonar:
> <img alt="Image" width="1291" height="117" src="https://private-user-images.githubusercontent.com/2520413/532365976-9cd19888-e83f-4dfb-8d64-bb5432bb0430.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NzA5MTQzNzcsIm5iZiI6MTc3MDkxNDA3NywicGF0aCI6Ii8yNTIwNDEzLzUzMjM2NTk3Ni05Y2QxOTg4OC1lODNmLTRkZmItOGQ2NC1iYjU0MzJiYjA0MzAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI2MDIxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNjAyMTJUMTYzNDM3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzNiM2U1NmZmNjNlZDg0MzllYTYxOGJlZjUzM2VhZjcyYzUxZDJmYzljOWE3NTYyOWFjZDVmMTk4ZWI0YzIzYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.9Ti5Jw-IZRd-eVHT1dh5DHy5VUisXmaF6wIe7ukPFCs">

@nashif Can we please disable the GNU Statement expression check in Sonar? It spams nearly all PRs, and makes it less likely that authors will actually go through the reported issues (that may be actual issues)