zephyrproject-rtos/zephyr #89728: Remove rule 58 from coding guidelines

nashif, 2025-05-09:
## Introduction

[image]

### Problem description

### Proposed change
Remove rule

keith-zephyr, 2025-05-14:
I think the SonarQube error reported on https://github.com/zephyrproject-rtos/zephyr/pull/87450 is actually a bug in SonarQube.

The MISRA rule is explicitly for when array uses designated initializers.  

```
   int32_t a1[ ]    = { [0] = 1 };    /* Non-compliant - probably unintentional to have single element */
```

But https://github.com/zephyrproject-rtos/zephyr/pull/87450 didn't specify array indexes using designated initializers.

```
	static const struct {
		int code;
		const char *str;
	} reason_strings[] = {
		{ WLAN_REASON_NO_REASON, "No reason specified" },
...
```

@jhedberg for visibilty.

nashif, 2025-05-19:
hmm, i think it is hitting 

```MISRA C:2012, 8.11 - When an array with external linkage is declared, its size should be explicitely specified```

which we do not have in our guidelines, but sonar combines the two: 9.5 and 8.11

keith-zephyr, 2025-05-21:
> hmm, i think it is hitting
>
> `MISRA C:2012, 8.11 - When an array with external linkage is declared, its size should be explicitely specified`
>
> which we do not have in our guidelines, but sonar [...]

PR https://github.com/zephyrproject-rtos/zephyr/pull/87450 declared a static array inside a function.  So the external linkage of rule 8.11 shouldn't apply.

I'm not necessarily against removing rule 58, but I haven't seen an instance where this rule is causing a problem.

@simhein what are your thoughts on removing this rule?

de-nordic, 2025-05-21:
I think this is intended to avoid errors like:
```
some_type some_array[] = { [0] = { init_0 }, [1] = { init_1}, [22] = { init_2 },  [3] = { init_3 } };
```
where you would end up with an array of 23 elements, while expecting 4. I guess that would get detected quickly?

nashif, 2025-05-22:
> PR [#87450](https://github.com/zephyrproject-rtos/zephyr/pull/87450) declared a static array inside a function. So the external linkage of rule 8.11 shouldn't apply.

yeah, I was not referring to this PR, was talking in general, it seems the sonarqube rule does combine two MISRA rules, one which is in our guidelines, the other is not, so this is confusing.

> I'm not necessarily against removing rule 58, but I haven't seen an instance where this rule is causing a problem.
> 
> [@simhein](https://github.com/simhein) what are your thoughts on removing this rule?

@keith-zephyr  Simon is not active in the project any more. This is something we need to descide on collectively with input from the safety/security cmtes and maintainers in general.

closed 2026-09-23 by github-actions[bot]