zephyrproject-rtos/zephyr #89736: Bluetooth: Controller: Missing CIS Established event with default ACL buffer size

Thalley, 2025-05-09:
**Describe the bug**
If `CONFIG_BT_BUF_ACL_RX_SIZE`/`CONFIG_BT_CTLR_DATA_LENGTH_MAX` is set to their default value of 27 then it seems that there's an issue with generating/providing the CIS Established event to the host. 

**To Reproduce**

Thalley, 2025-05-09:
Possibly related issue: https://github.com/zephyrproject-rtos/zephyr/issues/59173
Possibly related PR: https://github.com/zephyrproject-rtos/zephyr/pull/89258

cvinayak, 2025-05-11:
Diff:

```
diff --git a/samples/bluetooth/bap_unicast_client/prj.conf b/samples/bluetooth/bap_unicast_client/prj.conf
index 6d6658d728b..f1d8fadc77f 100644
--- a/samples/bluetooth/bap_unicast_client/prj.conf
+++ b/samples/bluetooth/bap_unicast_client/prj.conf
@@ -17,3 +17,5 @@ CONFIG_BT_BAP_UNICAST_CLIENT_ASE_SRC_COUNT=2
 CONFIG_BT_KEYS_OVERWRITE_OLDEST=y
 
 CONFIG_BT_EXT_ADV=y
+
+CONFIG_BT_USE_DEBUG_KEYS=y
diff --git a/samples/bluetooth/bap_unicast_client/src/main.c b/samples/bluetooth/bap_unicast_client/src/main.c
index fdc0ed249ff..d7e5277e2e2 100644
--- a/samples/bluetooth/bap_unicast_client/src/main.c
+++ b/samples/bluetooth/bap_unicast_client/src/main.c
@@ -193,7 +193,7 @@ static void device_found(const bt_addr_le_t *addr, int8_t rssi, uint8_t type,
        printk("Device found: %s (RSSI %d)\n", addr_str, rssi);
 
        /* connect only to devices in close proximity */
-       if (rssi < -50) {
+       if (rssi < -60) {
                return;
        }
 

```

Build client:
```
$ west build -b nrf5340bsim/nrf5340/cpuapp --sysbuild ../../samples/bluetooth/bap_unicast_client -- -DCONFIG_BT_ISO_LOG_LEVEL_DBG=y
```

Build Server:
```
$ west build -b nrf5340bsim/nrf5340/cpuapp --sysbuild ../../samples/bluetooth/bap_unicast_server -- -DCONFIG_BT_ISO_LOG_LEVEL_DBG=y
```

Running (`sh ~/test.sh > log.txt 2>&1`) where `test.sh` is:
```
#! /bin/sh

./build/bap_unicast_client/zephyr/zephyr.exe -s=temp -RealEncryption=1 -d=0 &
../bap_unicast_server/build/bap_unicast_server/zephyr/zephyr.exe -s=temp -RealEncryption=1 -d=1

```

and (in `tools/bsim/bin` folder): 

```
./bs_2G4_phy_v1 -s=temp -D=2 -sim_length=30e6
```

Logs (snippet):

```
d_00: @00:00:08.239712  (CPU:0): Enabling streams
d_01: @00:00:08.308005  (CPU:0): Enable: stream 0x8184ca0 meta_len 4
d_00: @00:00:08.378656  (CPU:0): Audio Stream 0x81997c0 enabled
d_01: @00:00:08.448005  (CPU:0): Enable: stream 0x8184cc4 meta_len 4
d_00: @00:00:08.518656  (CPU:0): Audio Stream 0x81997e8 enabled
d_01: @00:00:08.588006  (CPU:0): Enable: stream 0x8184d00 meta_len 4
d_00: @00:00:08.658655  (CPU:0): Audio Stream 0x8199810 enabled
d_00: @00:00:08.658655  (CPU:0): Streams enabled
d_00: @00:00:08.658655  (CPU:0): Connecting streams
d_00: @00:00:08.658655  (CPU:0): [00:00:08.658,630] [0m<dbg> bt_iso: bt_iso_chan_set_state_debug: chan 0x819e0c0 iso 0x819dd00 disconnected -> connecting[0m
d_01: @00:00:08.798090  (CPU:0): [00:00:08.798,065] [0m<dbg> bt_iso: hci_le_cis_req: acl_handle 0 cis_handle 7 cig_id 0 cis 0[0m
d_01: @00:00:08.798090  (CPU:0): [00:00:08.798,065] [0m<dbg> bt_iso: iso_accept: 0x8188980[0m
d_01: @00:00:08.798090  (CPU:0): [00:00:08.798,065] [0m<dbg> bt_iso: bt_iso_chan_add: iso 0x8188980 chan 0x81890e0[0m
[... 70 lines cut ...]
d_00: @00:00:09.428203  (CPU:0): [00:00:09.428,192] [0m<inf> lc3: Setting up LC3 encoder[0m
d_00: @00:00:09.428203  (CPU:0): [00:00:09.428,192] [0m<inf> stream_tx: Registered 0x81997e8 for TX[0m
d_01: @00:00:09.429884  (CPU:0): RX stream 0x8184ca0 len 0
d_01: @00:00:09.439884  (CPU:0): RX stream 0x8184ca0 len 0
d_01: @00:00:09.449885  (CPU:0): RX stream 0x8184ca0 len 0
d_01: @00:00:09.459884  (CPU:0): RX stream 0x8184ca0 len 0
d_01: @00:00:09.469885  (CPU:0): RX stream 0x8184ca0 len 0
d_01: @00:00:09.479885  (CPU:0): RX stream 0x8184ca0 len 0
d_01: @00:00:09.489884  (CPU:0): RX stream 0x8184ca0 len 0
d_01: @00:00:09.492105  (CPU:0): RX stream 0x8184cc4 len 0
d_01: @00:00:09.497986  (CPU:0): Start: stream 0x8184d00
d_01: @00:00:09.497986  (CPU:0): Failed to send audio data on streams[0] (0x8184d00): (-74)
d_01: @00:00:09.497986  (CPU:0): Audio Stream 0x8184d00 started
d_01: @00:00:09.499885  (CPU:0): RX stream 0x8184ca0 len 0
d_01: @00:00:09.502106  (CPU:0): RX stream 0x8184cc4 len 0
d_01: @00:00:09.508027  (CPU:0): Sending mock data with len 2 on streams[0] (0x8184d00)
d_01: @00:00:09.509885  (CPU:0): RX stream 0x8184ca0 len 0
d_01: @00:00:09.512106  (CPU:0): RX stream 0x8184cc4 len 0
d_01: @00:00:09.518067  (CPU:0): Sending mock data with len 3 on streams[0] (0x8184d00)
d_01: @00:00:09.519885  (CPU:0): RX stream 0x8184ca0 len 0
d_01: @00:00:09.522106  (CPU:0): RX stream 0x8184cc4 len 0

```

@Thalley from the bsim output of the samples, both client and server have each received `hci_le_cis_established`  as per the logs. I am I missing something (other than the logs having `len 0` for `Rx stream`) ?.

cvinayak, 2025-05-14:
I am seeing bus/usage/mem fault in the hci_ipc on the nrf5340 audio dk net core running the said samples on the PR: 89845

[image]

PR #89845 (2025-05-19): Bluetooth: Controller: Fix MIC failures, assertions and bus faults
Fix missing removal of redundant ISO Rx size in ACL Rx PDU
allocation size.

Related to commit https://github.com/zephyrproject-rtos/zephyr/commit/7c89f1fe9f2ed89b666dd9f7df0b95f94c7849be ("Bluetooth: controller:
Support for separate ISO RX data path").

Fixes #89736
PR #90396 (2025-06-13): Bluetooth: Controller: nRF53x: Fix up ISO Rx Buffer size
Fix up the bus fault by not violating the MAXPACKETSIZE value range of NRF_CCM h/w peripheral.

Fix up relates to commit 920117922bc8 ("Bluetooth: Controller: nRF53x: Fix NRF_CCM MAXPACKETSIZE value").

Fixes #89736.
closed 2025-05-19 by kartben