zephyrproject-rtos/zephyr #89774: Twister ignores the contents of the "Test Configuration File"

Pipoderoso, 2025-05-11:
**Describe the bug**

Twister ignores the "Test Configuration File". When calling twister with:

`west twister --test-config=<PATH_TO_CONFIG_FILE> ...`

the testplan script ignores the given configuration file.

**Environment (please complete the following information):**

 - OS: Linux
 - Zephyr v4.1.0

**Additional context**

It seems like the problem was introduced in Zephyr v4.1.0 in *scripts/pylib/twister/twisterlib/testplan.py*
I checked the script and what it seems to be the problem is that `self.add_configurations()` is being called before `self.parse_configuration(config_file=self.env.test_config)`.

As comparison:
- [v3.7.1](https://github.com/zephyrproject-rtos/zephyr/blob/v3.7.1/scripts/pylib/twister/twisterlib/testplan.py#L195)
- [v4.1.0](https://github.com/zephyrproject-rtos/zephyr/blob/v4.1.0/scripts/pylib/twister/twisterlib/testplan.py#L194)

nashif, 2025-05-13:
care to post your configuration? does it involve setting default platforms?

Pipoderoso, 2025-05-13:
Yes indeed, this is the configuration file:

```yaml
platforms:
  override_default_platforms: true
  default_platforms:
    - nrf52840dk/nrf52840
levels:
  - name: unit
    adds:
      - kernel.common.*
```

When I run `west twister --test-config tests/my_test_config.yml --level unit` I expect to build and run tests **only** for nrf52840dk/nrf52840 in *tests/kernel/common/*. However, twister starts building the expected tests but for **all** default platforms.

closed 2025-07-14 by dkalowsk