zephyrproject-rtos/zephyr #89782: CMake warning when building for Teensy 4.1

raveious, 2025-05-11:
**Describe the bug**
There appears to be a build warning coming from CMake when building for the Teensy 4.1. I've tried building for the native sim, RPi Pico, and an ESP32s3 board, none of which show the build warning. The issue does not appear in the v4.1 release tag.

**To Reproduce**

```
west build -p always -b teensy41 ./samples/hello_world
```

**Expected behavior**
No build warning.

**Impact**
Build output contains warnings.

**Logs and console output**

```
west build -p always -b teensy41 ./samples/hello_world
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: <snip>/zephyr/samples/hello_world
-- CMake version: 4.0.1
-- Found Python3: /Projects/zephyr-venv/bin/python (found suitable version "3.13.3", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: /home/<snip>/.cache/zephyr
-- Zephyr version: 4.1.99 (<snip>/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: teensy41, qualifiers: mimxrt1062
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.17.0 (/home/<snip>/zephyr-sdk-0.17.0)
-- Found toolchain: zephyr 0.17.0 (/home/<snip>/zephyr-sdk-0.17.0)
-- Found Dtc: /home/<snip>/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: <snip>/zephyr/boards/pjrc/teensy4/teensy41.dts
-- Generated zephyr.dts: <snip>/zephyr/build/zephyr/zephyr.dts
-- Generated pickled edt: <snip>/zephyr/build/zephyr/edt.pickle
-- Generated devicetree_generated.h: <snip>/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: <snip>/zephyr/build/zephyr/dts.cmake
Parsing <snip>/zephyr/Kconfig
Loaded configuration '<snip>/zephyr/boards/pjrc/teensy4/teensy41_defconfig'
Merged configuration '<snip>/zephyr/samples/hello_world/prj.conf'
Configuration saved to '<snip>/zephyr/build/zephyr/.config'
Kconfig header saved to '<snip>/zephyr/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/<snip>/zephyr-sdk-0.17.0/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/<snip>/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Looking for device MIMXRT1062 in <snip>/modules/hal/nxp/mcux/mcux-sdk-ng/devices/
-- Found device folder: <snip>/modules/hal/nxp/mcux/mcux-sdk-ng/devices/RT/RT1060/MIMXRT1062
CMake Warning at <snip>/zephyr/CMakeLists.txt:1034 (message):
  No SOURCES given to Zephyr library: drivers__memc

  Excluding target from build.

-- Configuring done (11.3s)
-- Generating done (1.0s)
-- Build files have been written to: <snip>/zephyr/build
-- west build: building application
[1/138] Preparing syscall dependency handling

[3/138] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.1.99 (<snip>/zephyr), build: v3.7.0-15669-g244888c830bc
[138/138] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       30522 B         8 MB      0.36%
             RAM:        4288 B       256 KB      1.64%
            ITCM:          0 GB       128 KB      0.00%
            DTCM:          0 GB       128 KB      0.00%
           OCRAM:          0 GB       256 KB      0.00%
          OCRAM2:          0 GB       512 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from <snip>/zephyr/build/zephyr/zephyr.elf for board: teensy41
```

**Environment:**

 - OS: Linux
 - Toolchain: Zephyr SDK 0.17.0
 - Commit SHA: fee8eddc5baa2eb68a6f021f72625d2551b495a2

vignesh-aerlync, 2025-05-16:
The Teensy 4.1 microcontroller uses the NXP iMXRT1062 SoC, which supports an external memory slot. In the Zephyr source code for this SoC, `CONFIG_MEMC` is enabled by default for all iMXRT series except the iMXRT118X, since that series does not have FLEXRAM.

I don’t have the board to test on, provided a patch below that disables `CONFIG_MEMC` by default instead of enabling it. I’m not sure how this will affect other NXP SoC boards. Without the patch, you can build the project for Teensy 4.1 and manually disable MEMC using:

```
west build -p always -b teensy41 ./samples/hello_world -- -DCONFIG_MEMC=n
```

Here’s the patch:

```diff
diff --git a/soc/nxp/imxrt/Kconfig.defconfig b/soc/nxp/imxrt/Kconfig.defconfig
index 78469d24fee..96ce6d6f184 100644
--- a/soc/nxp/imxrt/Kconfig.defconfig
+++ b/soc/nxp/imxrt/Kconfig.defconfig
@@ -59,7 +59,7 @@ config SYSTEM_WORKQUEUE_STACK_SIZE
 endif # ETH_NXP_ENET

 config MEMC
-       default y if !SOC_SERIES_IMXRT118X # RT118X lacks FLEXRAM
+       default n

 DT_CHOSEN_Z_DTCM := zephyr,dtcm

```

decsny, 2025-05-21:
the correct patch is to just delete those two lines, it was missed when I moved flexram out of the memc folder, I just haven't had time to form the PR, I think @dleach02 said he was planning to

PR #90301 (2025-05-27): soc: imxrt: remove MEMC config selection
MEMC no longer needs to be default y on RT builds.

Fixes #89782
commit c52bea0a (2025-05-27): soc: imxrt: remove MEMC config selection

MEMC no longer needs to be default y on RT builds.

Fixes #89782