zephyrproject-rtos/zephyr #89786: Introduce Stepper Motion Controller <-> Stepper Driver

jilaypandya, 2025-05-11:
**Is your enhancement proposal related to a problem? Please describe.**

Currently Stepper API combines **Motion Controller** and  **Stepper Driver** under a single api with following functions.

[image]

However, as one can see in the block diagram of TMC5041(Dual controller/driver for up to two 2-phase bipolar stepper motors with Integrated motion controller with SPI interface) is a controller with **MotionController + Stepper Driver**.

Additionally all the STEP/DIR Drivers such as the tmc2209 are just stepper drivers which has a "Software Motion Controller" integrated within them.

[image]

**Describe the solution you'd like**

Hence the cleaner approach would be to seperate "Motion Controller" from the actual "Stepper Driver" with the following API structure.
![controller->driver](https://www.plantuml.com/plantuml/png/bP9HQnin3CVVzIdqjeTBM0aPM_Om6XeOVUWOpTZ2NWx7foIcFilOkXjbxBlFFYUzDgo1lGH5-lzradw-ZgA2TAq5iBGJ5CPWzWV1nWJIOjW1H27lATIQdGIsTHDCd_ugRt0w_OIdyfW616rLZ7Z1-HkgbdkgXO_1zX6gq3ceeh1FloHrjjV62OLUMQZsAQMqTChMdgF9mnLTe1q5SfhEyYqt19M9AUYDsvSazUIarihQhT8Fy0V0iH1kMOHRvDs5eM7vzHPt7F2GJZIrr9CzgR5bIOMZvoDCyPwieOZi_d6WqaI5vfSAD8sUjDaPZPnKAeyVBnuzTjvpa8XFHjDwIoqvAMwqve7W9_d3OvYiV-3wEEDDwOMheHSgr-1oTJlvTdVpxcgpUKK6b6JvHF0IwXbCScfh2Qh6n24e08mEN306YeNWUIxhniTMvKA6yWQ_i53CbnhtB9-Chn1xJkepEGhAua1i16FJCtY9B6PMwzD5ynPPyxQiaQ9KlSiJ3wRi3XJPTddOR2IHd7nQ80vNFOb7ZF-FBr6PpQuM4rpTV_ymdyqdk9mltY-6_fdOPYHs8lKCrNKAkzR-1G00)

**Describe alternatives you've considered**

**Additional context**

Further context can be understood by the following image, here TMC429 is a **Stepper Motion Controller Driver**, whereas TMC2660 **Stepper Driver**.

| Stepper Motion Controller Driver  | Stepper Driver   | Notes      |
|-----------------|----------------|----------------|
| TMC429  | TMC2660  | TMC2209 and other step/dir driver might be compatible as well  |
| zephyr,stepper-control  | TMC22xx, DRV84xx, A4979  |  zephyr stepper control driver is a sw driver |
| TMC50XX Stepper Motion Controller  | TMC5XXX Stepper Driver  |  TMC5XXX Series are a combined packagage (something like TMC429 + TMC2660 packed together) |

[image]

jilaypandya, 2025-05-11:
Could you share your opinion on this?  
@bjarki-andreasen  @dipakgmx @fabiobaltieri @faxe1008

dipakgmx, 2025-05-11:
Guess, we are coming close to the MFD approach [suggested back then](https://github.com/zephyrproject-rtos/zephyr/pull/68548#pullrequestreview-1867193733) from the first stepper RFC.

jbehrensnx, 2025-05-12:
First of all, I am in favor of restructuring the current stepper api in this way.
By effectively inverting the whole step-dir implementation, it is actually possible to extend it beyond just adding new timing sources, see my PRs [#87800](https://github.com/zephyrproject-rtos/zephyr/pull/87800) and [#88342](https://github.com/zephyrproject-rtos/zephyr/pull/88342). This should make integrating them easier.
It is, in my opinion, also better structured from a logical and architectural perspective.
New stepper drivers should also be easier to implement.

One point that needs further discussion is, in my opinion, a stepper controller api capability to configure the acceleration profile. Yes, currently only one exists, but more will come, see PRs [#88342](https://github.com/zephyrproject-rtos/zephyr/pull/88342) and and [#88564](https://github.com/zephyrproject-rtos/zephyr/pull/88564) as examples. While a controller specific option  for fine control is useful and makes sense to exist, it somewhat goes against the generalist driver intention of Zephyr.
In addition, most users will probably want to set a speed and either an acceleration time or a maximum acceleration. The stepper controller driver should then translate this into its specific acceleration profile configuration.

One point I disagree with is the step-function for drivers. It is in my opinion unnecessary, the first indication for this can already be found in the fact that the tmc50xx doesn't implement it.
Most stepper controllers will not use it, in fact the only usefulness it has is, that it allows the existing workqueue and counter controllers (the latter not yet being modified) use the gpio bitbang driver. For the step-dir drivers a gpio-pin is driven, which can be done directly in the controller.
In addition, controlling the step pin in software is something that most practical real world applications will not do, instead having the step pin driven by a timer (see PR [#87800](https://github.com/zephyrproject-rtos/zephyr/pull/87800)) or using another hardware method. Something like this would also be used for any serious implementation that requires driving the h-bridges directly.
And spi-based stepper-drivers (like tmc50xx) feature either an integrated controller or are intended to be used with an additional hardware stepper controller like the TMC429, again making the step function unnecessary (and highly unperformant in that case).

jbehrensnx, 2025-05-12:
Also @jilaypandya as the maintainer, could you possibly contact people from Analog Devices and TI on the Zephyr Project? They are both Zephyr members and developers/producers of stepper drivers/controllers we have written drivers for, so getting feedback from their experts would be useful. I feel like the significant planned changes to the api offer a good opportunity for this.

If stepper support in Zephyr is not on their agenda, that is good to know, but raising awareness and possibly getting them involved would be useful.

faxe1008, 2025-05-12:
Generally makes sense to me, need to browse the reference PR some more. Regarding the naming I'd suggest `stepper_motion_controller_api` (adding motion in there somewhere), since imo it signals that the stepper driver itsself provides the base building blocks and configuration and any motion triggering etc. shall be done by that driver. 

Also a thing that I was wondering about is whether or not this can be generic enough to accomodate for DC motors and make it rather a subsystem then a driver? Allow for things like encoders for position detection etc. Maybe thats too amibitious tho.

jbehrensnx, 2025-05-12:
> Also a thing that I was wondering about is whether or not this can be generic enough to accomodate for DC motors and make it rather a subsystem then a driver? Allow for things like encoders for position detection etc. Maybe thats too amibitious tho.

Probably to ambitious for know. I think it is better we get a good stepper motion control api for know. I think we need to wait at least for a dc-motor api and possibly for a dc-motion-control api before we think about a general api. Plus for a general motion api we will want experts of that field to get involved as well.

jbehrensnx, 2025-05-12:
Another thing that needs to be considered is fault/error handling. Both the ti84xx and the tmc2209 drivers as well as the tmc2130 driver (I am currently working on the latter), as well as many more, have fault-gpio-pins. In addition the tmc drivers have (driver specific) error bit in their spi read values.
We need some way for controllers to manage this. One idea would be to have separate driver and controller event handling, with the controller attaching itself to the driver event handling and processing it.

jilaypandya, 2025-05-12:
> I like it, this way we can have a common "soft stepper controller" which works with various and numerous stepper drivers, could remove a bit of duplicate code from the "stepper driver drivers" :)

Reducing duplication of Code was one of the motivation :)

> Generally makes sense to me, need to browse the reference PR some more. Regarding the naming I'd suggest `stepper_motion_controller_api` (adding motion in there somewhere), since imo it signals that [...]

`stepper_motion_control_api` how does that sound :) ?

> One point I disagree with is the step-function for drivers. It is in my opinion unnecessary, the first indication for this can already be found in the fact that the tmc50xx doesn't implement it.

TMC429(Motion Controller) -- s/d -> TMC2260(Driver), I have already added that block diagram in the RFC.
`TMC5072-LA 2-Axis Stepper Motor Controller/Driver I2C, SPI, Step/Dir` - TMC5072 Datasheet.

I will start now with the refactoring of driver step-by-step :)

jbehrensnx, 2025-05-13:
Had an offline discussion with a colleague about the proposed changes. Some of the things that came up:

- There were arguments that while using a proper controller driver there is no need for using a dedicated driver-driver, as a controller should handle these things(enable/disable, handling microstep resolution) as part of motion control. The physical driver specific parts might be handled by a specific module.
- Again affirming that the step-function needs to be dropped as any serious real world use case will not use it and instead generate the step signal another way.
- Consequence of this is that any step-dir-controller driver will need to drive the step pin directly - at which point it can be argued that it is a stepper-driver-driver and not a controller any longer, as anything that directly configures a physical stepper driver or controls its pins is a driver driver (which means that a stepper controller is a physical stepper driver driver).
- Discussion about using the new architecture simply for step-dir drivers, with the proposed controller being a stepper-driver-driver and the proposed driver-driver being a config module plugged into the driver-driver that is used for configuration. While this option offers better extendability than the old one, there is the question about the driver self-configuring while it is running (e.g. current regulation during acceleration and coasting). There was no clear conclusion if the current or the possible new architecture would be better for this.

jilaypandya, 2025-05-13:
Hi @jbehrensnx  I kindly request you to please read the RFC again, and have a look at `TMC429 -- s/d --> TMC2660` setup, A stepper motion controller and stepper driver are per say two different entities. 

Just for instance, if somebody has a better software based stepper_motion_controller implementation downstream, they can still use a upstream stepper_driver. Also If you have a custom downstream tick_source, you can just do stepper_step and drive the steppers. A lot of high-end trinamic stepper drivers even provide a step/dir interface, as i already mentioned in the previous comment.

The block diagrams i have shown show a clear separation of concerns between stepper_motion_controller and stepper_driver. As @bjarki-andreasen has already pointed out moving all the control specific code to stepper_motion_controller would highly reduce code duplication.

I humbly request you to please have a look at RFC again.

jilaypandya, 2025-05-13:
> One completely different issue that I have noticed is that the current stepper api as well as your new proposed apis all work on a 1 stepper controller -> 1 stepper driver -> 1 stepper motor model, when each stepper controller might control multiple stepper drivers (TMC429: 3) and each stepper driver might drive multiple stepper motors.

From what I have seen there is always a one-one relation between a motion controller and driver 
[image]

[image]

jbehrensnx, 2025-05-14:
> From what I have seen there is always a one-one relation between a motion controller and driver

False. The TMC429 for example can control 3 drivers independently, with the set of registers you have shown existing separately for each driver. There are 3 ramp generators and the outputs allow for driving 3 stepper-drivers via either spi or step-dir interface. Its datasheet also explicitly states, that the three drivers can be configured independently. The TMC5041 also has 2 sets of driver hardware, as can be seen in the block diagram. Several of its registers exist twice to allow for driving the two motors separately.

> regarding stepper_step function. I'll make the dt-binding step-pin as not required. If it present in the dt then the driver will toggle it otherwise return -ENOTSUP.
Hence motion controllers like tmc429 or your STM32 based controller that are directly connected to the step pin of a stepper driver shall not call step function or even if they do no effect will take place.

Not sure i like this solution. The step-pin should always remain with the stepper-driver and not the dt-binding of the stepper-controller (of course it would be missing with hardware controllers). The controller could still access it via child node properties or something similar.
I still think dropping the function and just creating a separate controller for the gpio-driver is the best solution. That said, even that solution is not without its issues, as the stepper-driver api would be without any way to move the driver and would only be relevant for configuration, at which point the question comes up, if it makes sense for it to remain a full independent api.
I really would like an softwarearchitects feedback on this and the software-controller could be considered a stepper-driver question I raised earlier.

jilaypandya, 2025-05-14:
> False. The TMC429 for example can control 3 drivers independently, with the set of registers you have shown existing separately for each driver. There are 3 ramp generators and the outputs allow for [...]

TMC429 is a spi-device which has internally 3 motion controllers. TMC5041 is a spi-device which has internally 2 motion controllers. There is a one-one correspondence between a motion controller -> stepper in both the cases. @dipakgmx @bjarki-andreasen Is my understanding incorrect?

> > regarding stepper_step function. I'll make the dt-binding step-pin as not required. If it present in the dt then the driver will toggle it otherwise return -ENOTSUP.

> The step-pin should always remain with the stepper-driver and not the dt-binding of the stepper-controller (of course it would be missing with hardware controllers)

i mean in the dt-binding of stepper driver and not stepper motion controller, since stepper_step shall return -ENOTSUP.

> I really would like an softwarearchitects feedback on this

Could you please elaborate on this? Almost 3 years in Zephyr Project, i have not this kind of request yet.

dipakgmx, 2025-05-14:
> I really would like an softwarearchitects feedback on this and the software-controller could be considered a stepper-driver question I raised earlier.

I am not sure what are you trying to suggest here. A few of the collaborators have already provided their feedback here.

jbehrensnx, 2025-05-14:
> TMC429 is a spi-device which has internally 3 motion controllers. TMC5041 is a spi-device which has internally 2 motion controllers. There is a one-one correspondence between a motion controller -> stepper in both the cases. dipakgmx bjarki-andreasen Is my understanding incorrect?

Personally, I would consider the  TMC429 to be one controller, as it has a single spi interface and some shared resources and configuration. For example you have to decide if you want to control all connected stepper-drivers via spi or via step-dir, you cant decide on a per driver basis.

> i mean in the dt-binding of stepper driver and not stepper motion controller, since stepper_step shall return -ENOTSUP.

I think we misunderstood each other a bit. What I meant is that, unless the step pin is connected to and controlled by a hardware controller (e.g. TMC429) the step pin definition should only be in the stepper-driver dt-binding. The software controller should not have a step-pin binding, instead getting it if needed from the connected driver binding. That said, thinking more about this, this actually becomes a tricky question with the stm-driver (and probably similar ones in the future as well), as the step-signal output is defined in the controller output via pincontrol, even if no step-pin is ever defined. So I am not sure how best resolve this from an architectural point.

> The generic zephyr stepper motion controller would be able to drive both h-bridge and step/dir drivers using stepper_step function.

If I understood you in a previous discussion, sorry I don't remember which one, correctly, you said that the gpio-driver is more of an demonstrator that intended to be a serious driver. At least that was my takeaway. In that case I would consider the gpio-drivers compatibility with the workqueue/counter software-controllers of a low priority, compared to to a clean architecture. This is why I consider the gpio-driver unimportant for that discussion.

> I really would like an softwarearchitects feedback on this

> I am not sure what are you trying to suggest here. A few of the collaborators have already provided their feedback here.

Maybe I didn't find the correct words, but effectively I want somebody with softwarearchitecture experience to take a look at this and give feedback. This isn't intended as a slight against any of you and if any of you do have experience in that filed, I am sorry. 
This is also connected to my question if @jilaypandya could get people from ADI and TI involved. They are some of the domain experts on stepper drivers and the two companies are members of the Zephyr Project, but we have heard nothing from them concerning this api so far.
I think the question about the involvement of an software architect came up in the original PR for the api during a very heated part of the discussion.

jilaypandya, 2025-05-14:
> Personally, I would consider the TMC429 to be one controller, as it has a single spi interface and some shared resources and configuration. For example you have to decide if you want to control all connected stepper-drivers via spi or via step-dir, you cant decide on a per driver basis.

TMC429 Motion Controller does not send target_position VIA SPI to the TMC2660 stepper driver. Stepping always happens via step/dir. That SPI Line from TMC429 to TMC2660 is for setting stallguard,chopconf and so on. 

snippet from TMC22660 datasheet
[image]

From Arch perspective, the Split is quite clear, motion controller generates step signals and stepper driver gets step signals. Some trinamic chips combines these two in one package and supply it as an integrated solution. But in the datasheet you will see that there is always one motion controller per driver.

andre-stefanov, 2025-05-14:
> If I understood you in a previous discussion, sorry I don't remember which one, correctly, you said that the gpio-driver is more of an demonstrator that intended to be a serious driver. At least [...]

I would strongly disagree. Even thought extra hardware solutions like tmc429 or tmc5xxx drivers (with integrated ramps) have better performance and timing accuracy, the software solutions (especially counter based) are very widely used in open source projects around the world. It does not matter if it is just a demo or a final product. There are definitely some advantages of counter based solutions:

- Cost efficiency: most 32 bit MCUs already have counters built in and some of them will be probably unused. Dedicated ICs cost extra money.
- Simple applications: if the requirement is just to move a stepper at "normal" speed with a simple acceleration ramp (or even without any acceleration), it is doable without a specialized IC.
- Highly customizable: If an application has very unique movement requirements like very slow stepping speeds, special ramps, synchronization with other events etc. a counter offer maximum flexibility.
- Board space: Sometimes there is no space for extra ICs

Further i don't see how the implementation of counter/worker timing sources is contradicting a clean architecture.

> Personally, I would consider the TMC429 to be one controller, as it has a single spi interface and some shared resources and configuration. For example you have to decide if you want to control all connected stepper-drivers via spi or via step-dir, you cant decide on a per driver basis.

Even if TMC429 uses a single SPI interface to address single steppers, it still can and should be considered as parent of 3 motion controllers from zephyr's point of view. Implementing it as a single device would introduce the need of duplicating code for other controllers which support different amount of steppers (1, 2 ... etc) and making device trees less reusable and the implementation (and DTS API) more inconsistent with rising amount of supported drivers. Instead defining it just as 3 separate (equal) motion controllers which share the same parent stepper controller allows the flexibility of reusing these in other drivers. The parent (stepper controller) would be then in charge of configuring SPI/stepdir for all it's motion controllers.

> Except that they have shared resources like the single microstep lookuptable available in the TMC429.

Since multiple motion controllers will still share the same parent, the implementation of these would be straight forward with the proposed solution imo.

jbehrensnx, 2025-05-15:
> I would strongly disagree. Even thought extra hardware solutions like tmc429 or tmc5xxx drivers (with integrated ramps) have better performance and timing accuracy, the software solutions (especially counter based) are very widely used in open source projects around the world.

I think there was a bit of an misunderstanding. I am not considering softwarecontrollers or even software h-bridge drivers (like the gpio-driver) as unimportant. The issue is with the implementation of the ones currently available in Zephyr, and more specifically the fact that they are toggling the gpio-pins for steps in software. Instead, the step-pin is usually controlled by an pwm or a counter/timer output. The stm-timer PR that I linked does exactly that, but is vendor specific to allow for efficient step counting. These kinds of software-stepper-controllers are definitely used.

> Further i don't see how the implementation of counter/worker timing sources is contradicting a clean architecture.

The issue is, that every software that directly controls a stepper-driver - be it by toggling pins or sending spi-messages is by definition a stepper-driver-driver. The proposed timing source controllers circumvent this through the step function, but as mentioned, that is not sustainable. Thus what we currently would consider software-controllers would become stepper-driver-drivers (the stm-timer controller would be it from the start). That is the architectural issue.

> Even if TMC429 uses a single SPI interface to address single steppers, it still can and should be considered as parent of 3 motion controllers from zephyr's point of view. Implementing it as a [...]

>> Except that they have shared resources like the single microstep lookuptable available in the TMC429.

> Since multiple motion controllers will still share the same parent, the implementation of these would be straight forward with the proposed solution imo.

Except that this would be a three level architecture instead of the two level architecture that is actually proposed (there is no stepper-controller above the motion-controller that would handle these issues). It would also just put the issue one level higher, at that is the level the user would actually use.

jilaypandya, 2025-05-15:
> Except that this would be a three level architecture instead of the two level architecture that is actually proposed (there is no stepper-controller above the motion-controller that would handle these issues). 

You recognized it correclty. Its a three level arch, it's something i tested yesterday and i'll soon create a PR.

as far as `stepper_step` is concerned, i am going to keep it in stepper driver api, and there is a use-case with counter/workq for h-bridge and quite a few step/dir drivers out there, as @andre-stefanov  already mentioned. If the sophisticated motion-controllers don't need it, they shall not use it.

jilaypandya, 2025-05-18:
Hi Everyone,

long story short, we  (@dipakgmx , @andre-stefanov  and I ) tried shifting around blocks and kinda tried to distribute various blocks (motion-controller, stepper driver) of tmc controllers into various device drivers.

However, we eventually came to the conclusion that the step-dir/h-bridge drivers, as such lack a motion-controller and the ansatz would be to inject one in it. We are kinda already doing it but the goal would be to unify the timing source api from step-dir lib and step-counting in one so called "software motion controller". (@andre-stefanov  has already some PoC Code in that direction)

closed 2025-05-18 by jilaypandya

jilaypandya, 2025-06-21:
So, a small update to keep everyone informed in which direction we are heading/aiming with stepper area.
 
Motion Controllers like the "Zephyr,SW-Stepper-Motion-Controller", probably, we need to think of a nice name for it :), ingrained and embedded behind the api in the drivers has to be brought to the device driver level. 

#26221, the very first stepper API did this as well, directly implemented a stepper motion controller on device driver level. This would also enable downstream controllers to be developed which could still use some of the upstreamed step-dir-drivers.

PR https://github.com/zephyrproject-rtos/zephyr/pull/91909 has further helped in separating concerns.

|     | stepper driver | stepper motion control driver   |
|----------|-----|-----------|
|enable / disable    |  x |       |
|microstep_resolution     |  x |        |
|step/dir  |  x |    |
|move_to/move_by/run/stop  |   |  x  |
|set_reference_position  |   |  x  |
|*is_moving |   |  x  | 
|get_actual_position|   |  x  |
|set_reference_position|  |  x  |

*is_moving, (need a better name for this one, is_active/is_stopped)

There is one function `stepper_set_event_callback` which would have to be split as following.

|     | stepper driver | stepper motion control driver   |
|----------|-----|-----------|
|set_fault_event_callback   |  x |       |
|set_stepper_motion_event_callback  |    |   x |

Now how do we move ahead :) We need to handle the split in a step-wise manner (_love the word step, kind of fits everywhere :_), since refactoring couple of thousands of lines of code is not an easy task.

**Phase 1:**
- [ ] Introduce set_fault_event_callback, remove fault_event from the current stepper_event enum and rename stepper_event_callback to stepper_motion_control_event_callback.
- [ ] Introduce set_step/set_dir functions in stepper api so that the zephyr, stepper-motion-controller can be brought to a device driver level without affecting the existing drivers like tmc5xxx.