zephyrproject-rtos/zephyr #89826: drivers: flash: common: drivers.flash.common.disable_spi_nor fails

katgiadla, 2025-05-12:
**Describe the bug**
The test `tests/drivers/flash/common/drivers.flash.common.disable_spi_nor` fails.

Observed for:
- `nrf54l15dk/nrf54l05/cpuapp`
- `nrf54l15dk/nrf54l10/cpuapp`
- `nrf54l15dk/nrf54l15/cpuapp`

**To Reproduce**
Steps to reproduce the behavior:
1. go to your zephyr dir
2. call `./scripts/twister -s drivers.flash.common.disable_spi_nor -p nrf54l15dk/nrf54l15/cpuapp --device-testing --device-serial /dev/ttyACM1 -v --inline-logs --west-flash="--erase"`
3. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v4.1.0-3766-gb2c6d960cbd1 ***
Running TESTSUITE flash_driver
===================================================================
START - test_flash_copy
Test will run on device rram-controller@5004b000
No devices with erase requirement present

    Assertion failed at zephyr/tests/drivers/flash/common/src/main.c:403: test_flash_copy_inner: flash_flatten(flash_dev, page_info.start_offset + page_info.size, page_info.size) is non-zero

 FAIL - test_flash_copy in 0.246 seconds
===================================================================
START - test_flash_erase
Test will run on device rram-controller@5004b000
No devices with erase requirement present
 PASS - test_flash_erase in 0.155 seconds
===================================================================
START - test_flash_fill
Test will run on device rram-controller@5004b000
No devices with erase requirement present
 PASS - test_flash_fill in 0.423 seconds
===================================================================
START - test_flash_flatten
Test will run on device rram-controller@5004b000
No devices with erase requirement present
 PASS - test_flash_flatten in 0.351 seconds
===================================================================
START - test_flash_page_layout
Test will run on device rram-controller@5004b000
No devices with erase requirement present
start_offset=0x15c000size=4096index=348
page_count=357
 PASS - test_flash_page_layout in 0.014 seconds
===================================================================
START - test_get_size
Test will run on device rram-controller@5004b000
No devices with erase requirement present
 SKIP - test_get_size in 0.009 seconds
===================================================================
START - test_read_unaligned_address
Test will run on device rram-controller@5004b000
No devices with erase requirement present
 PASS - test_read_unaligned_address in 0.013 seconds
===================================================================
TESTSUITE flash_driver failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  83.33% [flash_driver]: pass = 5, fail = 1, skip = 1, total = 7 duration = 1.211 seconds
 - FAIL - [flash_driver.test_flash_copy] duration = 0.246 seconds
 - PASS - [flash_driver.test_flash_erase] duration = 0.155 seconds
 - PASS - [flash_driver.test_flash_fill] duration = 0.423 seconds
 - PASS - [flash_driver.test_flash_flatten] duration = 0.351 seconds
 - PASS - [flash_driver.test_flash_page_layout] duration = 0.014 seconds
 - SKIP - [flash_driver.test_get_size] duration = 0.009 seconds
 - PASS - [flash_driver.test_read_unaligned_address] duration = 0.013 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.17.0
 - Commit SHA or Version used: `b2c6d960cbd1de8258d18acfcf947ca13231e84d`

**Additional information:**
Probably introducedy by commit d8be41c00a1ca2dcdc97031436dc851a6f5137b9

FRASTM, 2025-05-13:
Seen on fk7b0m1_vbt6 board when building tests/drivers/flash/common -T drivers.flash.common.disable_spi_nor
```
in function `flash_erase':
~/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/syscalls/flash.h:90: undefined reference to `__device_dts_ord_132'
~/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(main.c.obj): in function `z_impl_flash_read':
~/zephyrproject/zephyr/include/zephyr/drivers/flash.h:255: undefined reference to `__device_dts_ord_132'
~/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(main.c.obj): in function `z_impl_flash_get_size':
~/zephyrproject/zephyr/include/zephyr/drivers/flash.h:354: undefined reference to `__device_dts_ord_132'
~/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(main.c.obj): in function `flash_driver_before':
~/zephyrproject/zephyr/tests/drivers/flash/common/src/main.c:124: undefined reference to `__device_dts_ord_132'
~/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(main.c.obj): in function `_flash_driver_test_flash_page_layout_wrapper':
~/zephyrproject/zephyr/tests/drivers/flash/common/src/main.c:344: undefined reference to `__device_dts_ord_132'
```

kartben, 2025-05-15:
@decsny can you please take a look?

decsny, 2025-05-15:
Most likely it is those nrF platforms just failing the test. It seems especially likely given that only one test fails.

I specifically added this test case because I discovered that most SOC flashes were not being tested due to the way the test was written to prioritize testing the SPI flash if one is present.

I added the case to expose the bugs by adding coverage but will not be fixing all the bugs of all the platforms that weren't tested before.