zephyrproject-rtos/zephyr #89831: USB high speed not working on U5

bearsh, 2025-05-12:
**Describe the bug**
USB high-speed is not working on a NUCLEO-U5A5ZJ-Q and STM32U5A9J-DK. On full-speed works

**To Reproduce**

**Expected behavior**
The board should use high-speed usb

**Logs and console output**

kernel: usb 3-6.1.1.4: new full-speed USB device number 43 using xhci_hcd
kernel: usb 3-6.1.1.4: config 1 interface 1 altsetting 0 endpoint 0x82 has invalid maxpacket 512, setting to 64
kernel: usb 3-6.1.1.4: config 1 interface 1 altsetting 0 endpoint 0x1 has invalid maxpacket 512, setting to 64
kernel: usb 3-6.1.1.4: New USB device found, idVendor=2fe3, idProduct=0001, bcdDevice= 4.01
kernel: usb 3-6.1.1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
kernel: usb 3-6.1.1.4: Product: Zephyr CDC ACM sample
kernel: usb 3-6.1.1.4: Manufacturer: ZEPHYR
kernel: usb 3-6.1.1.4: SerialNumber: 4231500120393441
kernel: cdc_acm 3-6.1.1.4:1.0: ttyACM0: USB ACM device

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit: eeb1d92df475dae7f79f4c30f468e5a12736854c

**Additional context**
see also https://github.com/zephyrproject-rtos/zephyr/pull/86922#discussion_r2084562504

bearsh, 2025-05-12:
**proposed fix**

the speed selection preprocessor logic in `usb_dc_stm32.c` and `udc_stm32.c` does not work as intended:
- check for `defined(USB_OTG_HS_EMB_PHYC)` is always true as it just got defined a couple of lines above
- when looking at https://github.com/zephyrproject-rtos/zephyr/pull/86922/files: `usb_dc_stm32_get_maximum_speed()`, `USB_OTG_SPEED_HIGH_IN_FULL` should only be set if `high-speed` is **not** set in dt

```patch
diff --git a/drivers/usb/device/usb_dc_stm32.c b/drivers/usb/device/usb_dc_stm32.c
index 16b91470dee..c41c6a689dc 100644
--- a/drivers/usb/device/usb_dc_stm32.c
+++ b/drivers/usb/device/usb_dc_stm32.c
@@ -91,8 +91,8 @@ static const struct gpio_dt_spec ulpi_reset =
  * The following defines are used to map the value of the "maxiumum-speed"
  * DT property to the corresponding definition used by the STM32 HAL.
  */
-#if defined(CONFIG_SOC_SERIES_STM32H7X) || defined(USB_OTG_HS_EMB_PHYC) || \
-	defined(USB_OTG_HS_EMB_PHY)
+#if DT_ENUM_IDX(DT_DRV_INST(0), maximum_speed) == 1 && \
+	(defined(CONFIG_SOC_SERIES_STM32H7X) || USB_OTG_HS_EMB_PHYC || USB_OTG_HS_EMB_PHY)
 #define USB_DC_STM32_HIGH_SPEED             USB_OTG_SPEED_HIGH_IN_FULL
 #else
 #define USB_DC_STM32_HIGH_SPEED             USB_OTG_SPEED_HIGH
diff --git a/drivers/usb/udc/udc_stm32.c b/drivers/usb/udc/udc_stm32.c
index bec4860c375..11a6e1e91a5 100644
--- a/drivers/usb/udc/udc_stm32.c
+++ b/drivers/usb/udc/udc_stm32.c
@@ -41,7 +41,10 @@ LOG_MODULE_REGISTER(udc_stm32, CONFIG_UDC_DRIVER_LOG_LEVEL);
 #define UDC_STM32_IRQ		DT_INST_IRQ_BY_NAME(0, UDC_STM32_IRQ_NAME, irq)
 #define UDC_STM32_IRQ_PRI	DT_INST_IRQ_BY_NAME(0, UDC_STM32_IRQ_NAME, priority)
 
-#define USB_OTG_HS_EMB_PHY (DT_HAS_COMPAT_STATUS_OKAY(st_stm32_usbphyc) && \
+#define USB_OTG_HS_EMB_PHYC (DT_HAS_COMPAT_STATUS_OKAY(st_stm32_usbphyc) && \
+			     DT_HAS_COMPAT_STATUS_OKAY(st_stm32_otghs))
+
+#define USB_OTG_HS_EMB_PHY (DT_HAS_COMPAT_STATUS_OKAY(st_stm32u5_otghs_phy) && \
 			    DT_HAS_COMPAT_STATUS_OKAY(st_stm32_otghs))
 
 #define USB_OTG_HS_ULPI_PHY (DT_HAS_COMPAT_STATUS_OKAY(usb_ulpi_phy) && \
@@ -51,7 +54,8 @@ LOG_MODULE_REGISTER(udc_stm32, CONFIG_UDC_DRIVER_LOG_LEVEL);
  * The following defines are used to map the value of the "maxiumum-speed"
  * DT property to the corresponding definition used by the STM32 HAL.
  */
-#if defined(CONFIG_SOC_SERIES_STM32H7X) || defined(USB_OTG_HS_EMB_PHY)
+#if DT_ENUM_IDX(DT_DRV_INST(0), maximum_speed) == 1 && \
+	(defined(CONFIG_SOC_SERIES_STM32H7X) || USB_OTG_HS_EMB_PHYC || USB_OTG_HS_EMB_PHY)
 #define UDC_STM32_HIGH_SPEED             USB_OTG_SPEED_HIGH_IN_FULL
 #else
 #define UDC_STM32_HIGH_SPEED             USB_OTG_SPEED_HIGH
@@ -1006,13 +1010,13 @@ static void priv_pcd_prepare(const struct device *dev)
 	priv->pcd.Instance = (USB_OTG_GlobalTypeDef *)UDC_STM32_BASE_ADDRESS;
 #endif /* USB */
 
-#if USB_OTG_HS_EMB_PHY
+#if USB_OTG_HS_EMB_PHYC || USB_OTG_HS_EMB_PHY
 	priv->pcd.Init.phy_itface = USB_OTG_HS_EMBEDDED_PHY;
 #elif USB_OTG_HS_ULPI_PHY
 	priv->pcd.Init.phy_itface = USB_OTG_ULPI_PHY;
 #else
 	priv->pcd.Init.phy_itface = PCD_PHY_EMBEDDED;
-#endif /* USB_OTG_HS_EMB_PHY */
+#endif /* USB_OTG_HS_EMB_PHYC || USB_OTG_HS_EMB_PHY */
 }
 
 static const struct stm32_pclken pclken[] = STM32_DT_INST_CLOCKS(0);
@@ -1146,7 +1150,7 @@ static int priv_clock_enable(void)
 	LL_AHB1_GRP1_DisableClockLowPower(LL_AHB1_GRP1_PERIPH_OTGHSULPI);
 #endif /* defined(CONFIG_SOC_SERIES_STM32H7X) */
 
-#if USB_OTG_HS_EMB_PHY
+#if USB_OTG_HS_EMB_PHYC
 #if !DT_HAS_COMPAT_STATUS_OKAY(st_stm32n6_otghs)
 	LL_APB2_GRP1_EnableClock(LL_APB2_GRP1_PERIPH_OTGPHYC);
 #endif

```

erwango, 2025-05-12:
@kbidani ^^

erwango, 2025-06-30:
@kbidani ^^

nandojve, 2025-07-15:
Hi,

I can not see USB working on `nucleo_u5a5zj_q` when using usb device-next.
Can someone confirm that it works ?

I'm trying to build usbtest on f52d71c8cf5, which is 10 commits after rc3.

nandojve, 2025-07-16:
Hi @bearsh ,

> [@nandojve](https://github.com/nandojve) did you include the patches from [#89866](https://github.com/zephyrproject-rtos/zephyr/pull/89866)?

No, I'll try and report back.

nandojve, 2025-09-15:
> Hi [@bearsh](https://github.com/bearsh) ,
> 
> > [@nandojve](https://github.com/nandojve) did you include the patches from [#89866](https://github.com/zephyrproject-rtos/zephyr/pull/89866)?
> 
> No, I'll try and report back.

What I made was the #94001