zephyrproject-rtos/zephyr #89841: drivers/usb/udc/Kconfig.stm32 does not select UDC_DRIVER_HAS_HIGH_SPEED_SUPPORT

bearsh, 2025-05-12:
**Describe the bug**
According to `subsys/usb/device_next/Kconfig`, `CONFIG_UDC_DRIVER_HAS_HIGH_SPEED_SUPPORT` is required to select _high-speed_ mode. Unfortunately, this is not selected in `drivers/usb/udc/Kconfig.stm32` whereas other udc drivers (e.g. `drivers/usb/udc/Kconfig.mcux`) do select that.

**Expected behavior**
usb _high-speed_ mode (`UDC_DRIVER_HAS_HIGH_SPEED_SUPPORT=y`) should be selected for stm32 devices which makes it possible to set the max speed to _high-speed_ (`CONFIG_USBD_MAX_SPEED=1`)

**Impact**
currently it is not possible to use _high-speed_ usb for stm32 with the usb-next stack.

erwango, 2025-05-12:
@bearsh Thanks for raising this. Would you be able to contribute this change ?

closed 2025-08-12 by github-actions[bot]

mathieuchopstm, 2025-10-01:
Fixed by #96028