zephyrproject-rtos/zephyr #89850: APDS9306-065 reset throws an I2C error

Kampi, 2025-05-12:
**Describe the bug**

When using the Zephyr driver for the APDS9306 sensor with an APDS9306-065 device, the reset of the chip fails because of an I2C error. This issue occurs in the latest NCS, which contains the latest driver for this sensor.

```
00> [00:04:05.464,042] <dbg> avago_apds9306: apds9306_init: Start to initialize APDS9306...
00> [00:04:05.474,973] <dbg> avago_apds9306: apds9306_sensor_setup: APDS-9306-065 found!
00> [00:04:05.475,276] <err> avago_apds9306: Can not reset the sensor!
00> [00:04:05.475,281] <err> avago_apds9306: Failed to setup device!
```

**Expected behavior**

The initialization must be executed without an issue.

**Environment (please complete the following information):**

 - NCS 3.0
 - Sensor driver commit [2174ea0](https://github.com/nrfconnect/sdk-zephyr/commit/2174ea0ac0c7ca2aef57fbdf6f2a903dcf9ed441)

henrikbrixandersen, 2025-05-13:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

Kampi, 2025-05-14:
Hi @henrikbrixandersen,

I want to fix the issue on Zephyr `main` (will do it later). I just referenced this issue to NCS 3.0 because I´ve figured out the issue with it. It´s very likely that this issue also happens on Zephyr `main` since both drivers are identical. And I´m pretty sure that this issue is not related to nRF only.

Is this okay for you?

Kampi, 2025-05-22:
@carlescufi yes. I have a patch ready, which was tested by me this week in the latest NCS version. I can´t figure out the reason for this behaviour, and the datasheet is not very precise. So I simply remove the reset.

```
From 17d41dd41ace010f48d17632ae2d6bff7c360841 Mon Sep 17 00:00:00 2001
From: Daniel Kampert <DanielKampert@kampis-elektroecke.de>
Date: Wed, 21 May 2025 23:13:59 +0200
Subject: [PATCH] Remove reset from APDS-9306 driver

---
 drivers/sensor/apds9306/apds9306.c | 15 ---------------
 1 file changed, 15 deletions(-)

diff --git a/drivers/sensor/apds9306/apds9306.c b/drivers/sensor/apds9306/apds9306.c
index 87723b21624..00c2ab9537c 100644
--- a/drivers/sensor/apds9306/apds9306.c
+++ b/drivers/sensor/apds9306/apds9306.c
@@ -301,21 +301,6 @@ static int apds9306_sensor_setup(const struct device *dev)
 		LOG_DBG("APDS-9306-065 found!");
 	}
 
-	/* Reset the sensor. */
-	if (i2c_reg_write_byte_dt(&config->i2c, APDS9306_REGISTER_MAIN_CTRL,
-				  APDS9306_BIT_SW_RESET)) {
-		LOG_ERR("Can not reset the sensor!");
-		return -EFAULT;
-	}
-	k_msleep(10);
-
-	/* Perform a dummy read to avoid bus errors after the reset. See */
-	/* https://lore.kernel.org/lkml/ab1d9746-4d23-efcc-0ee1-d2b8c634becd@tweaklogic.com/ */
-	if (i2c_reg_read_byte_dt(&config->i2c, APDS9306_REGISTER_PART_ID, &temp)) {
-		LOG_ERR("Failed reading chip id!");
-		return -EFAULT;
-	}
-
 	return 0;
 }
 
-- 
2.48.1.windows.1
```

closed 2025-05-31 by nashif