zephyrproject-rtos/zephyr #89852: S32K344 halts after a period of time

cmoser-crl, 2025-05-12:
**Describe the bug**

I'm working on an application for the NXP S32K344 and am currently using the MR-CANHUBK3 board for development. After a period of time, the MCU appears to halt/hang. The time it takes to fail is consistent across reboots for a given program. Initially, I thought the issue was with Logging/UART output as it seems to occur faster when that's enabled, but after more testing, it also happens with it disabled.

I've tried to do some debugging with Trace32. After the issue occurs, I'm not able to pause the processor with the break command. If I use the ITM trace with PC sampler, it looks like the cpu hangs on an instruction in the main thread. The instruction it hangs on varries across runs. The cpu continues handling the systick interrupts and then hangs each time it returns to the main thread. Other interrupts such as GPIO interrupts are also handled. After the issue occurs, I'm still able to view the peripheral registers in Trace32 and I've comfirmed that the safety peripherals haven't caused a functional reset.

If I disable threading with `CONFIG_MULTITHREADING=n` or disable data cache with `CONFIG_DCACHE=n`, the issue doesn't appear to occur. I've run builds with both for over 48 hours without any issue.

**To Reproduce**

I've created a repo [here](https://github.com/cmoser-crl/simple-zephyr-example) that consistently reproduces the issue after about a minute. I've also tested with the `samples/basic/blinky` sample program and the issue occurs after ~8.5 hours.

**Expected behavior**

MCU shouldn't halt/hang.

**Impact**

We're able to work around it by modifying existing drivers to not use threads, but it's a pretty big annoyance.

**Logs and console output**

```
[00:01:17.368] hello world 2822
[00:01:17.384] hello world 2823
[00:01:17.416] hello world 2824
[00:01:17.432] hello world 2825
[00:01:17.448] hello world 2826
*stops outputting*
```

**Environment**

- OS: Linux
- Toolchain: Zyphyr sdk 0.16.6 (also tested 0.16.9)
- Zephyr version: v3.7.0 (also tested v4.0.0, v4.1.0)
- Code: https://github.com/cmoser-crl/simple-zephyr-example

**Additional context**

All of the testing was done with the FS26 SBC in [Debug mode](https://docs.zephyrproject.org/latest/boards/nxp/mr_canhubk3/doc/index.html#fs26-sbc-watchdog).

I've tested the same code on a Teensy (MIMXRT1062, also Cortex M7) without seeing this issue.

[image]
> systick is still handled after the main thread hangs

[image]
> the systick handler hangs when trying to return to the main thread

manuargue, 2025-05-14:
Hi @cmoser-crl , this may be related to Cortex-M core trying to make speculative access to certain memory address that do not allow to be read. Would you mind trying this patch with your sample and feedback if the issue still persist?

```
diff --git a/soc/nxp/s32/s32k3/mpu_regions.c b/soc/nxp/s32/s32k3/mpu_regions.c
index acd3ee9d916..66b027e5cc7 100644
--- a/soc/nxp/s32/s32k3/mpu_regions.c
+++ b/soc/nxp/s32/s32k3/mpu_regions.c
@@ -14,6 +14,13 @@ extern char _rom_attr[];

static struct arm_mpu_region mpu_regions[] = {

+       /* Prevent speculative access in entire memory region */
+       {
+               .name = "BACKGROUND",
+               .base = 0,
+               .attr = REGION_RAM_ATTR(REGION_4G | NOT_EXEC | P_RW_U_RW_Msk),
+       },
+
        /* Keep before CODE region so it can be overlapped by SRAM CODE in non-XIP systems */
        {
                .name = "SRAM",
```

cmoser-crl, 2025-05-19:
This patch fixed the issue I was seeing. I checked this morning and the MCU has been running fine for the last 4 days.

manuargue, 2025-05-19:
thanks for the feedback @cmoser-crl , I'll open a pr with this patch

PR #90305 (2025-05-29): fix potential speculative access that might trigger bus faults on Cortex-M7
Due to erratum [1013783-B](https://documentation-service.arm.com/static/665dff778ad83c4754308908?token=), speculative accesses might be performed to normal memory unmapped in the MPU. This can be avoided by using MPU region 0 to cover all unmapped memory and make this region execute-never and inaccessible.

Note that this applies for all Cortex-M7 revisions implementing MPU and D-cache, so although the issue was reported for `mr_canhubk3` board that implements custom SoC MPU regions, I added the fix for any Cortex-M7 S [... cut]
commit c3164023 (2025-05-31): soc: nxp: s32k3: fix erratum ERR011573

Due to erratum ERR011573, speculative accesses might be performed
to normal memory unmapped in the MPU. This can be avoided by using
MPU region 0 to cover all unmapped memory and make this region
execute-never and inaccessible.

Fixes #89852