zephyrproject-rtos/zephyr #89855: changing a binding breaks the build

teburd, 2025-05-12:
**Describe the bug**
Changing a binding should result in rebuilding and regenerating the pickle file, cmake, and generated header. It does not.

**To Reproduce**
1. Build for any sample/board
2. Change related binding
3. build (not pristine!) again
4. Wonder why things don't work

**Expected behavior**
Rebuilds on binding changes should work

**Impact**
Annoyance

**Additional context**
We don't generate a build dependencies file for bindings like we do for devicetree source includes, we likely should. Though its also possible the binding dependencies change on a dts file change so that should be kept in mind.

Loosely Related #89729 as it pertains to another python script that doesn't generate a dependency file and appears to be a repeated mistake

teburd, 2025-05-12:
@bjarki-andreasen noted he too runs into this I believe, and I didn't see any issue for it

dkalowsk, 2025-05-13:
This is possibly more an Enhancement than a Bug as I don't believe this ever worked.

closed 2025-08-12 by github-actions[bot]

teburd, 2026-04-08:
@henrikbrixandersen you mentioned you do not do pristine builds often, this is why I do