zephyrproject-rtos/zephyr #89872: sensor: bme280 regulary fails to fetch samples

visuphi, 2025-05-13:
**Describe the bug**

I have the following setting
- Zephyr 4.1.0
- nRF52840DK 3.0.2
- BME280 Pimoroni Module, connected to I2C on P0.27 and P0.27, VCC and GND.

The BME280 regulary fails to fetch samples, returning the error code `-EAGAIN`. 
It looks like the function `bme280_wait_until_ready` in the bme280 driver regulary runs into a timeout.

```
*** Booting Zephyr OS build v4.1.0 ***
Found device "BME280", getting sensor data
[00:00:00.416,992] <err> BME280: Failed to fetch samples
BME280: sensor_read() failed: -11
```

**To Reproduce**

- connect bme280 module to to I2C0 on P0.27 and P0.27, VCC and GND.
- use the BME280 sample in `samples/sensor/bme280`
- add overlay file for nrf52840dk:

```
&i2c0 {
    status = "okay";

    bme280@76 {
        compatible = "bosch,bme280";
        reg = <0x76>;
        label = "BME280";
        status = "okay";
    };
};
```
- build and flash `west build -b nrf52840dk/nrf52840 && west flash`
- see logoutputs on serial

..or use my sample: https://github.com/visuphi/test_bme280_zephyr

**Expected behavior**

bme280 should be able to fetch sample without running regularly into timeout.

**Impact**

no samples fetched

**Logs and console output**

when using zephyr sample:
```
*** Booting Zephyr OS build v4.1.0 ***
Found device "BME280", getting sensor data
[00:00:00.416,992] <err> BME280: Failed to fetch samples
BME280: sensor_read() failed: -11
```

when using my sample, referenced above:
```
*** Booting Zephyr OS build v4.1.0 ***
[00:00:00.525,482] <err> main: sensor_sample_fetch() bme280 failed: -11
[00:00:00.525,543] <inf> main: temperature: 0.000000
[00:00:00.525,604] <inf> main: humidity: 0.000000
[00:00:00.525,665] <inf> main: pressure: 0.000000
[00:00:02.678,833] <err> main: sensor_sample_fetch() bme280 failed: -11
[00:00:02.678,924] <inf> main: temperature: 0.000000
[00:00:02.678,985] <inf> main: humidity: 0.000000
[00:00:02.679,046] <inf> main: pressure: 0.000000
[00:00:04.746,368] <inf> main: temperature: 23.740000
[00:00:04.746,520] <inf> main: humidity: 34.657226
[00:00:04.746,673] <inf> main: pressure: 96.534425
[00:00:06.759,521] <inf> main: temperature: 23.730000
[00:00:06.759,674] <inf> main: humidity: 34.634765
[00:00:06.759,826] <inf> main: pressure: 96.533644
[00:00:08.913,024] <err> main: sensor_sample_fetch() bme280 failed: -11
[00:00:08.913,146] <inf> main: temperature: 23.730000
[00:00:08.913,299] <inf> main: humidity: 34.634765
[00:00:08.913,452] <inf> main: pressure: 96.533644
[00:00:11.064,300] <inf> main: temperature: 23.720000
[00:00:11.064,453] <inf> main: humidity: 34.632812
[00:00:11.064,605] <inf> main: pressure: 96.535851
[00:00:13.077,423] <inf> main: temperature: 23.720000
[00:00:13.077,575] <inf> main: humidity: 34.667968
[00:00:13.077,728] <inf> main: pressure: 96.535695
[00:00:15.090,576] <inf> main: temperature: 23.710000
[00:00:15.090,728] <inf> main: humidity: 34.667968
[00:00:15.090,881] <inf> main: pressure: 96.534429
[00:00:17.103,729] <inf> main: temperature: 23.710000
[00:00:17.103,881] <inf> main: humidity: 34.689453
[00:00:19.257,232] <err> main: sensor_sample_fetch() bme280 failed: -11
```

**Environment (please complete the following information):**

 - OS: macOS Sequoia 15.4.1
 - Zephyr 4.1.0 with Zephyr SDK 0.17.0

**Additional context**

vignesh-aerlync, 2025-05-15:
Hi @visuphi I’m able to reproduce the issue in both your code and the Zephyr sample. The problem occurs because `BME280_MEASUREMENT_TIMEOUT_MS` in *bme.c* is set too low. I’ll raise a PR with the change.

> *** Booting Zephyr OS build v4.1.0-3678-gaf77efb77e91 ***
Found device "BME280", getting sensor data
temp: 24.959991; press: 100.292968; humidity: 0.7812
temp: 24.949981; press: 100.292968; humidity: 0.7812
temp: 24.929992; press: 100.296875; humidity: 0.7812
temp: 24.929992; press: 100.296875; humidity: 0.7812 
temp: 24.939971; press: 100.292968; humidity: 0.7812
temp: 24.939971; press: 100.292968; humidity: 0.7812
temp: 24.949981; press: 100.296875; humidity: 0.7812
temp: 24.949981; press: 100.296875; humidity: 0.7812
[00:00:09.195,709] <err> BME280: Failed to fetch samples

visuphi, 2025-05-15:
hi @vignesh-aerlync 

Thank you for your response. 
The error really occurs less if you increase the timeout. However, I have increased the timeout up to 500ms but the error still occurs in a few cases.

which value for `BME280_MEASUREMENT_TIMEOUT_MS` do you suggest?

ldomaigne, 2025-05-21:
I've observed this kind of behavior when some of the contacts to the BME280 sensor is loose. Have you already eliminated that possibility?

ldomaigne, 2025-05-23:
I could reproduce the issue on the NRF52840DK. My preliminaries investigations let to think that we have some regressions on that platform:

| Zephyr  |  Platform     | Test | 
|--------- | ------------- | ---- | 
| LTS 3.7  | nrf52840dk  | :white_check_mark: |
| LTS 3.7  | mcxn947      | :white_check_mark: |
| 4.1        | nrf52840dk  | :x: | 
| 4.1        | mcxn947      | :white_check_mark: |

Too early to draw any conclusions yet, however.

On a side note, I've observed some suspicious behaviors if a transient I/O errors is occurring on all tested platforms/versions. It's possibly an unrelated issue.

ldomaigne, 2025-05-23:
On my test setup, If BME280 mode is set to forced (CONFIG_BME280_FORCED=y), then the problem disappears. 
@visuphi : do you see the same?

visuphi, 2025-05-26:
Hi @ldomaigne 

I assume you mean `CONFIG_BME280_MODE_FORCED=y` :).

I can confirm that the problem disappears when using this config.

PR #90646 (2025-05-29): drivers: sensor: bme280: fix "config" register intialization issue
A write to the "ctrl_meas" register can cause the sensor to transition from sleep to normal mode (which is default Kconfig settings). As per BME280 datasheet, writes to the "config" register in normal mode may be ignored. This can lead to use BME280_STANDBY_05MS and BME280_FILTER_OFF instead of the values set by the application. To fix this, write to the "config" register before "ctrl_meas" in the sensor init function.

Fixes: #89872.
closed 2025-05-29 by kartben