zephyrproject-rtos/zephyr #89889: Not able to keep PHY at 1M when connecting with iPhone

mahori-ravinder, 2025-05-13:
**Describe the bug**

I (Peripheral) wants to connect with iPhone(Central) and make sure connection PHY is 1M.
Please note that my device (peripheral) supports 2M, i just want to use 1M.
I tried the recommended settings (gathered by reading zephyr doc + google search), but the iPhone moves the PHY to 2M after connection.

**To Reproduce**
To make this happen i set following config params

CONFIG_BT_CTLR_PHY_2M=n
CONFIG_BT_CTLR_PHY_CODED=n
# Tell user about phy update
CONFIG_BT_USER_PHY_UPDATE=y
# Udpate PHY
CONFIG_BT_PHY_UPDATE=n
#https://devzone.nordicsemi.com/f/nordic-q-a/119628/how-to-set-cis-phy-to-1m-in-le-audio
CONFIG_BT_AUTO_PHY_UPDATE=n

The expectation was that connection will happen at 1M.

The initial connection happens at 1M. After that iPhone sends a phy request (preferring 2M), my device (peripheral) responds with PHY response (preferring 2M). This leads to PHY moving to 2M.

See below screen grab of OTA.

[image]

**Expected behavior**
iPhone ends phy request (preferring 2M) and my device (peripheral) responds with PHY response (preferring 1M).

Please note: as a work around i created a fork of zephyr (https://github.com/zephyrproject-rtos/zephyr/pull/87810) . This makes use of LE_SET_DEFAULT_PHY HCI commands. This enables me to achieve right behaviour (See OTA below)

[image]

**Environment (please complete the following information):**
Zephyr OS build ba3251d14138
 

**What configuration i need to do to achieve the desired behaviour?**

dkalowsk, 2025-05-13:
@mahori-ravinder it's not clear from the issue which device this is on.  Is it a Nordic device or another?

cvinayak, 2025-05-14:
The preferred PHY is hard-coded to 2M when using `CONFIG_BT_AUTO_PHY_UPDATE=y` in the Zephyr Host implementation:

https://github.com/zephyrproject-rtos/zephyr/blob/3c7e43f4e2ba87c2427bb74d6fa5fa6b3311891d/subsys/bluetooth/host/conn.c#L1800-L1810

But when `CONFIG_BT_AUTO_PHY_UPDATE=n` then the Zephyr Controller (nRF and may be OpenISA SoC variants) power-on defaults of using preferred 2M PHY is used. **There is no Zephyr Host support to update the Controller defaults using HCI LE Set Default Phy Command**, this will be a enhancement request to Zephyr Host. Thank you for sending #87810.

CC @jhedberg  @alwa-nordic 

Finally, as a intermediate solution, with `CONFIG_BT_USER_PHY_UPDATE=y` you can use `bt_conn_le_phy_update()` in your application to switch to any supported PHY on the connection at runtime, like in this sample:

https://github.com/zephyrproject-rtos/zephyr/blob/3c7e43f4e2ba87c2427bb74d6fa5fa6b3311891d/samples/bluetooth/central_gatt_write/src/gatt_write_common.c#L113-L134

mahori-ravinder, 2025-05-14:
> [@mahori-ravinder](https://github.com/mahori-ravinder) it's not clear from the issue which device this is on. Is it a Nordic device or another?

This is Infineon 20829.

mahori-ravinder, 2025-05-14:
> The preferred PHY is hard-coded to 2M when using `CONFIG_BT_AUTO_PHY_UPDATE=y` in the Zephyr Host implementation:
> 
> [zephyr/subsys/bluetooth/host/conn.c](https://github.com/zephyrproject-rtos/zephyr/blob/3c7e43f4e2ba87c2427bb74d6fa5fa6b3311891d/subsys/bluetooth/host/conn.c#L1800-L1810)
> 
> Lines 1800 to 1810 in [3c7e43f](/zephyrproject-rtos/zephyr/commit/3c7e43f4e2ba87c2427bb74d6fa5fa6b3311891d)
> 
>  if (IS_ENABLED(CONFIG_BT_AUTO_PHY_UPDATE) && BT_FEAT_LE_PHY_2M(bt_dev.le.features) && 
>      !uses_symmetric_2mbit_phy(conn)) { 
>  	err = bt_le_set_phy(conn, 0U, BT_HCI_LE_PHY_PREFER_2M, BT_HCI_LE_PHY_PREFER_2M, 
>  			    BT_HCI_LE_PHY_CODED_ANY); 
>  	if (err) { 
>  		LOG_ERR("Failed LE Set PHY (%d)", err); 
>  	} 
>  	if (conn->state != BT_CONN_CONNECTED) { 
>  		return; 
>  	} 
>  } 
> But when `CONFIG_BT_AUTO_PHY_UPDATE=n` then the Zephyr Controller (nRF and may be OpenISA SoC variants) power-on defaults of using preferred 2M PHY is used. **There is no Zephyr Host support to update the Controller defaults using HCI LE Set Default Phy Command**, this will be a enhancement request to Zephyr Host. Thank you for sending [#87810](https://github.com/zephyrproject-rtos/zephyr/pull/87810).
> 
> CC [@jhedberg](https://github.com/jhedberg) [@alwa-nordic](https://github.com/alwa-nordic)
> 
> Finally, as a intermediate solution, with `CONFIG_BT_USER_PHY_UPDATE=y` you can use `bt_conn_le_phy_update()` in your application to switch to any supported PHY on the connection at runtime, like in this sample:
> 
> [zephyr/samples/bluetooth/central_gatt_write/src/gatt_write_common.c](https://github.com/zephyrproject-rtos/zephyr/blob/3c7e43f4e2ba87c2427bb74d6fa5fa6b3311891d/samples/bluetooth/central_gatt_write/src/gatt_write_common.c#L113-L134)
> 
> Lines 113 to 134 in [3c7e43f](/zephyrproject-rtos/zephyr/commit/3c7e43f4e2ba87c2427bb74d6fa5fa6b3311891d)
> 
>  struct bt_conn_le_phy_param conn_phy_param; 
>   
>  if (conn_info.role == BT_CONN_ROLE_CENTRAL) { 
>  	conn_phy_param.options = phy_param[phy_param_idx].options; 
>  	conn_phy_param.pref_tx_phy = phy_param[phy_param_idx].pref_tx_phy; 
>  	conn_phy_param.pref_rx_phy = phy_param[phy_param_idx].pref_rx_phy; 
>  } else { 
>  	conn_phy_param.options = phy_param[phy_param_idx].options; 
>  	conn_phy_param.pref_tx_phy = phy_param[phy_param_idx].pref_rx_phy; 
>  	conn_phy_param.pref_rx_phy = phy_param[phy_param_idx].pref_tx_phy; 
>  } 
>   
>  printk("%s: PHY Update requested %u %u (%u)\n", __func__, 
>         conn_phy_param.pref_tx_phy, 
>         conn_phy_param.pref_rx_phy, 
>         conn_phy_param.options); 
>   
>  err = bt_conn_le_phy_update(conn, &conn_phy_param); 
>  if (err) { 
>  	printk("Failed to update PHY (%d).\n", err); 
>  	return; 
>  }

> The preferred PHY is hard-coded to 2M when using `CONFIG_BT_AUTO_PHY_UPDATE=y` in the Zephyr Host implementation:
> 
> [zephyr/subsys/bluetooth/host/conn.c](https://github.com/zephyrproject-rtos/zephyr/blob/3c7e43f4e2ba87c2427bb74d6fa5fa6b3311891d/subsys/bluetooth/host/conn.c#L1800-L1810)
> 
> Lines 1800 to 1810 in [3c7e43f](/zephyrproject-rtos/zephyr/commit/3c7e43f4e2ba87c2427bb74d6fa5fa6b3311891d)
> 
>  if (IS_ENABLED(CONFIG_BT_AUTO_PHY_UPDATE) && BT_FEAT_LE_PHY_2M(bt_dev.le.features) && 
>      !uses_symmetric_2mbit_phy(conn)) { 
>  	err = bt_le_set_phy(conn, 0U, BT_HCI_LE_PHY_PREFER_2M, BT_HCI_LE_PHY_PREFER_2M, 
>  			    BT_HCI_LE_PHY_CODED_ANY); 
>  	if (err) { 
>  		LOG_ERR("Failed LE Set PHY (%d)", err); 
>  	} 
>  	if (conn->state != BT_CONN_CONNECTED) { 
>  		return; 
>  	} 
>  } 
> But when `CONFIG_BT_AUTO_PHY_UPDATE=n` then the Zephyr Controller (nRF and may be OpenISA SoC variants) power-on defaults of using preferred 2M PHY is used. **There is no Zephyr Host support to update the Controller defaults using HCI LE Set Default Phy Command**, this will be a enhancement request to Zephyr Host. Thank you for sending [#87810](https://github.com/zephyrproject-rtos/zephyr/pull/87810).
> 
> CC [@jhedberg](https://github.com/jhedberg) [@alwa-nordic](https://github.com/alwa-nordic)
> 
> Finally, as a intermediate solution, with `CONFIG_BT_USER_PHY_UPDATE=y` you can use `bt_conn_le_phy_update()` in your application to switch to any supported PHY on the connection at runtime, like in this sample:
> 
> [zephyr/samples/bluetooth/central_gatt_write/src/gatt_write_common.c](https://github.com/zephyrproject-rtos/zephyr/blob/3c7e43f4e2ba87c2427bb74d6fa5fa6b3311891d/samples/bluetooth/central_gatt_write/src/gatt_write_common.c#L113-L134)
> 
> Lines 113 to 134 in [3c7e43f](/zephyrproject-rtos/zephyr/commit/3c7e43f4e2ba87c2427bb74d6fa5fa6b3311891d)
> 
>  struct bt_conn_le_phy_param conn_phy_param; 
>   
>  if (conn_info.role == BT_CONN_ROLE_CENTRAL) { 
>  	conn_phy_param.options = phy_param[phy_param_idx].options; 
>  	conn_phy_param.pref_tx_phy = phy_param[phy_param_idx].pref_tx_phy; 
>  	conn_phy_param.pref_rx_phy = phy_param[phy_param_idx].pref_rx_phy; 
>  } else { 
>  	conn_phy_param.options = phy_param[phy_param_idx].options; 
>  	conn_phy_param.pref_tx_phy = phy_param[phy_param_idx].pref_rx_phy; 
>  	conn_phy_param.pref_rx_phy = phy_param[phy_param_idx].pref_tx_phy; 
>  } 
>   
>  printk("%s: PHY Update requested %u %u (%u)\n", __func__, 
>         conn_phy_param.pref_tx_phy, 
>         conn_phy_param.pref_rx_phy, 
>         conn_phy_param.options); 
>   
>  err = bt_conn_le_phy_update(conn, &conn_phy_param); 
>  if (err) { 
>  	printk("Failed to update PHY (%d).\n", err); 
>  	return; 
>  }

Vinayak - Thanks for your comment. Runtime phy update is not possible in my specific scenario. The reason being, iPhone rejects my request to move phy to 1M.

I see you have given specific comments on my PR to add coverage. Will update that so that this enhancement can go in. Thanks.