zephyrproject-rtos/zephyr #89951: example-application does not build after 4.1.0

bastienjauny, 2025-05-14:
**Describe the bug**
Following the example application README instructions, the build fails if we use a version of Zephyr post 4.1.0. More precisely, [this commit](https://github.com/zephyrproject-rtos/zephyr/commit/f726cb5123eda12feb8746e3acb34a1f7d7b0340) is causing a Kconfig warning, which in turns abort the build.

**To Reproduce**
```
west init -m https://github.com/zephyrproject-rtos/example-application --mr main ws
west update
cd ws
west build -b custom_plank  example_application/app -- -DEXTRA_CONF_FILE=debug.conf
```

**Expected behavior**
The example app builds succesfully.

**Impact**
Minor annoyance but the example application is the entrypoint for most of the newcomers.

**Logs and console output**
```
Parsing /home/bajau/trainings/ws/example-application/app/Kconfig
Loaded configuration '/home/bajau/trainings/ws/build/zephyr/.config'

warning: CMSIS_CORE_HAS_SYSTEM_CORE_CLOCK (defined at soc/nxp/imxrt/imxrt6xx/Kconfig.defconfig:8, modules/cmsis_6/Kconfig:23) has direct dependencies (SOC_SERIES_IMXRT6XX && SOC_FAMILY_NXP_IMXRT) || (HAS_CMSIS_CORE && 0) with value n, but is currently being y-selected by the following symbols:
 - SOC_FAMILY_NORDIC_NRF (defined at soc/nordic/Kconfig.soc:19, soc/nordic/Kconfig:9), with value y, direct dependencies y (value: y), and select condition ARM (value: y)

error: Aborting due to Kconfig warnings

CMake Error at /home/bajau/trainings/ws/zephyr/cmake/modules/kconfig.cmake:365 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/bajau/trainings/ws/zephyr/cmake/modules/zephyr_default.cmake:131 (include)
  /home/bajau/trainings/ws/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/bajau/trainings/ws/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  CMakeLists.txt:8 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/home/bajau/.local/share/pipx/venvs/west/bin/python -B/home/bajau/trainings/ws/build -GNinja -DEXTRA_CONF_FILE=debug.conf -S/home/bajau/trainings/ws/example-application/app
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK 0.17.0
 - Errors happen starting with commit f726cb5123eda12feb8746e3acb34a1f7d7b0340

**Additional context**

bastienjauny, 2025-05-14:
Moving this:
```config CMSIS_CORE_HAS_SYSTEM_CORE_CLOCK
	bool
	help
	  Enable this option if CMSIS SystemCoreClock symbols is available.
```

back to `modules/cmsis/Kconfig` makes the build start but it fails with the following:
```Parsing /home/bajau/trainings/my-workspace/example-application/app/Kconfig
Loaded configuration '/home/bajau/trainings/my-workspace/build/zephyr/.config'
No change to configuration in '/home/bajau/trainings/my-workspace/build/zephyr/.config'
No change to Kconfig header in '/home/bajau/trainings/my-workspace/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found gen_kobject_list: /home/bajau/trainings/my-workspace/zephyr/scripts/build/gen_kobject_list.py
-- Configuring done (2.3s)
-- Generating done (0.0s)
-- Build files have been written to: /home/bajau/trainings/my-workspace/build
-- west build: building application
[7/153] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj 
ccache /home/bajau/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DNRF52840_XXAA -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I/home/bajau/trainings/my-workspace/zephyr/kernel/include -I/home/bajau/trainings/my-workspace/zephyr/arch/arm/include -I/home/bajau/trainings/my-workspace/build/zephyr/include/generated/zephyr -I/home/bajau/trainings/my-workspace/zephyr/include -I/home/bajau/trainings/my-workspace/build/zephyr/include/generated -I/home/bajau/trainings/my-workspace/zephyr/soc/nordic -I/home/bajau/trainings/my-workspace/zephyr/soc/nordic/nrf52/. -I/home/bajau/trainings/my-workspace/zephyr/soc/nordic/common/. -I/home/bajau/trainings/my-workspace/example-application/include -I/home/bajau/trainings/my-workspace/modules/hal/nordic/nrfx -I/home/bajau/trainings/my-workspace/modules/hal/nordic/nrfx/drivers/include -I/home/bajau/trainings/my-workspace/modules/hal/nordic/nrfx/mdk -I/home/bajau/trainings/my-workspace/zephyr/modules/hal_nordic/nrfx/. -isystem /home/bajau/trainings/my-workspace/zephyr/lib/libc/common/include -fno-strict-aliasing -Og -imacros /home/bajau/trainings/my-workspace/build/zephyr/include/generated/zephyr/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/home/bajau/zephyr-sdk-0.17.0/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/bajau/trainings/my-workspace/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/bajau/trainings/my-workspace/example-application/app=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/bajau/trainings/my-workspace/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/bajau/trainings/my-workspace=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=picolibc.specs -std=c99 -fno-lto -MD -MT zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -c /home/bajau/trainings/my-workspace/zephyr/arch/arm/core/offsets/offsets.c
In file included from /home/bajau/trainings/my-workspace/zephyr/include/zephyr/arch/arm/asm_inline.h:18,
                 from /home/bajau/trainings/my-workspace/zephyr/include/zephyr/arch/arm/arch.h:30,
                 from /home/bajau/trainings/my-workspace/zephyr/include/zephyr/arch/cpu.h:19,
                 from /home/bajau/trainings/my-workspace/zephyr/include/zephyr/kernel_includes.h:36,
                 from /home/bajau/trainings/my-workspace/zephyr/include/zephyr/kernel.h:17,
                 from /home/bajau/trainings/my-workspace/zephyr/arch/arm/core/offsets/offsets_aarch32.c:28,
                 from /home/bajau/trainings/my-workspace/zephyr/arch/arm/core/offsets/offsets.c:9:
/home/bajau/trainings/my-workspace/zephyr/include/zephyr/arch/arm/asm_inline_gcc.h:24:10: fatal error: cmsis_core.h: No such file or directory
   24 | #include <cmsis_core.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/bajau/trainings/my-workspace/build
```

raphaZahnd, 2025-05-14:
Our automated build started to fail without changing any of our code. Must be related to west update which we do on our automated builds every-time:

10/188] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj 
ccache /opt/toolchains/zephyr-sdk-0.16.9/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DCORE_CM7 -DHSE_VALUE=25000000 -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -DSTM32H743xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I/workdir/zephyr/kernel/include -I/workdir/zephyr/arch/arm/include -I/workdir/CB2MCU/app/build/zephyr/include/generated/zephyr -I/workdir/zephyr/include -I/workdir/CB2MCU/app/build/zephyr/include/generated -I/workdir/zephyr/soc/st/stm32 -I/workdir/zephyr/lib/posix/options/getopt -I/workdir/zephyr/soc/st/stm32/common/. -I/workdir/zephyr/drivers -I/workdir/zephyr/soc/st/stm32/stm32h7x/. -I/workdir/zephyr/subsys/shell/modules/kernel_service/thread/.. -I/workdir/CB2MCU/include -I/workdir/modules/hal/stm32/stm32cube/stm32h7xx/soc -I/workdir/modules/hal/stm32/stm32cube/stm32h7xx/drivers/include -I/workdir/modules/hal/stm32/stm32cube/common_ll/include -isystem /workdir/zephyr/lib/libc/common/include -fno-strict-aliasing -Os -imacros /workdir/CB2MCU/app/build/zephyr/include/generated/zephyr/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m7 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/opt/toolchains/zephyr-sdk-0.16.9/arm-zephyr-eabi/arm-zephyr-eabi -imacros /workdir/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/workdir/CB2MCU/app=CMAKE_SOURCE_DIR -fmacro-prefix-map=/workdir/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/workdir=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=picolibc.specs -D_POSIX_THREADS -std=c99 -fno-lto -MD -MT zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -c /workdir/zephyr/arch/arm/core/offsets/offsets.c
In file included from /workdir/zephyr/include/zephyr/arch/arm/asm_inline.h:18,
                 from /workdir/zephyr/include/zephyr/arch/arm/arch.h:30,
                 from /workdir/zephyr/include/zephyr/arch/cpu.h:19,
                 from /workdir/zephyr/include/zephyr/kernel_includes.h:36,
                 from /workdir/zephyr/include/zephyr/kernel.h:17,
                 from /workdir/zephyr/arch/arm/core/offsets/offsets_aarch32.c:28,
                 from /workdir/zephyr/arch/arm/core/offsets/offsets.c:9:
/workdir/zephyr/include/zephyr/arch/arm/asm_inline_gcc.h:24:10: fatal error: cmsis_core.h: No such file or directory

joelguittet, 2025-05-14:
Hello, I reported the error with the example-application on Discord and @bastienjauny thanks again for creating this issue and analyzing this.

I also opened a PR few days ago now and just rebased the branch on top of main to solve merge conflicts. This is also not building on top of main outside of example-application, e.g. building samples inside the zephyr directory as explained by the getting started guide to build the hello-world sample.

So it's very annoying and totally blocking developments now.... :(

joelguittet, 2025-05-14:
As example the hello-world sample doesn't build anymore on current top main commit 6d8eb270fc68dec39ffd2c76b17070a69318355a 👎 

```
(.venv) ubuntu@ubuntu-20:~/zephyrproject/zephyr$ west build -d build-hello -b nucleo_l4a6zg samples/hello_world
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/ubuntu/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.28.3
-- Found Python3: /home/ubuntu/zephyrproject/.venv/bin/python3 (found suitable version "3.12.3", minimum required is "3.10") found components: Interpreter 
-- Cache files will be written to: /home/ubuntu/.cache/zephyr
-- Zephyr version: 4.1.99 (/home/ubuntu/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: nucleo_l4a6zg, qualifiers: stm32l4a6xx
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.8 (/home/ubuntu/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (/home/ubuntu/zephyr-sdk-0.16.8)
-- Found Dtc: /home/ubuntu/zephyr-sdk-0.16.8/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/ubuntu/zephyrproject/zephyr/boards/st/nucleo_l4a6zg/nucleo_l4a6zg.dts
-- Generated zephyr.dts: /home/ubuntu/zephyrproject/zephyr/build-hello/zephyr/zephyr.dts
-- Generated pickled edt: /home/ubuntu/zephyrproject/zephyr/build-hello/zephyr/edt.pickle
-- Generated devicetree_generated.h: /home/ubuntu/zephyrproject/zephyr/build-hello/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/ubuntu/zephyrproject/zephyr/build-hello/zephyr/dts.cmake
Parsing /home/ubuntu/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/ubuntu/zephyrproject/zephyr/boards/st/nucleo_l4a6zg/nucleo_l4a6zg_defconfig'
Merged configuration '/home/ubuntu/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/ubuntu/zephyrproject/zephyr/build-hello/zephyr/.config'
Kconfig header saved to '/home/ubuntu/zephyrproject/zephyr/build-hello/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/ubuntu/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ubuntu/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Using ccache: /usr/bin/ccache
-- Found gen_kobject_list: /home/ubuntu/zephyrproject/zephyr/scripts/build/gen_kobject_list.py
-- Configuring done (13.9s)
-- Generating done (0.1s)
-- Build files have been written to: /home/ubuntu/zephyrproject/zephyr/build-hello
-- west build: building application
[1/138] Preparing syscall dependency handling

[2/138] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.1.99 (/home/ubuntu/zephyrproject/zephyr), build: v4.1.0-3997-g66a5212f8ac6
[9/138] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj 
ccache /home/ubuntu/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DCORE_CM4 -DHSE_VALUE=8000000 -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -DSTM32L4A6xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I/home/ubuntu/zephyrproject/zephyr/kernel/include -I/home/ubuntu/zephyrproject/zephyr/arch/arm/include -I/home/ubuntu/zephyrproject/zephyr/build-hello/zephyr/include/generated/zephyr -I/home/ubuntu/zephyrproject/zephyr/include -I/home/ubuntu/zephyrproject/zephyr/build-hello/zephyr/include/generated -I/home/ubuntu/zephyrproject/zephyr/soc/st/stm32 -I/home/ubuntu/zephyrproject/zephyr/soc/st/stm32/common/. -I/home/ubuntu/zephyrproject/zephyr/drivers -I/home/ubuntu/zephyrproject/zephyr/soc/st/stm32/stm32l4x/. -I/home/ubuntu/zephyrproject/modules/hal/stm32/stm32cube/stm32l4xx/soc -I/home/ubuntu/zephyrproject/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include -I/home/ubuntu/zephyrproject/modules/hal/stm32/stm32cube/common_ll/include -isystem /home/ubuntu/zephyrproject/zephyr/lib/libc/common/include -fno-strict-aliasing -Os -imacros /home/ubuntu/zephyrproject/zephyr/build-hello/zephyr/include/generated/zephyr/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/home/ubuntu/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/ubuntu/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/ubuntu/zephyrproject/zephyr/samples/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/ubuntu/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/ubuntu/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=picolibc.specs -std=c99 -fno-lto -MD -MT zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -c /home/ubuntu/zephyrproject [... cut ...]
                 from /home/ubuntu/zephyrproject/zephyr/arch/arm/core/offsets/offsets_aarch32.c:28,
                 from /home/ubuntu/zephyrproject/zephyr/arch/arm/core/offsets/offsets.c:9:
/home/ubuntu/zephyrproject/zephyr/include/zephyr/arch/arm/asm_inline_gcc.h:24:10: fatal error: cmsis_core.h: No such file or directory
   24 | #include <cmsis_core.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/ubuntu/zephyrproject/zephyr/build-hello
```

dipakgmx, 2025-05-14:
Is it because of this `west.yml` update? I had to update my manifest to include cmsis_6, and then I could get the main compiling

joelguittet, 2025-05-14:
@bastienjauny fixed by @fabiobaltieri at https://github.com/joelguittet/example-application/commit/06bbb0a5e5adc5b6da0c0bb16919cebea931ee7e

fabiobaltieri, 2025-05-14:
Yup, fixed.

closed 2025-05-14 by fabiobaltieri