zephyrproject-rtos/zephyr #89965: Scheduler: Fails to work

parthitce, 2025-05-14:
**Describe the bug**
- Create 2 threads with same priority
- Run the same entry point with tight while
- Enable `CONFIG_TIMESLICING`
- Only first thread in ready state always runs and scheduler fails to re-schedule second thread

**To Reproduce**
```
#define STACKSIZE       512
#define PRIORITY        7

const k_tid_t thread_a;
const k_tid_t thread_b;
K_THREAD_STACK_DEFINE(thread_a_stack_area, STACKSIZE);
K_THREAD_STACK_DEFINE(thread_b_stack_area, STACKSIZE);

void thread_entry_point(void *dummy1, void *dummy2, void *dummy3)
{
        char *name = (char *)dummy1;

        while (1) {
                printk("%s\n", name);
        }
}

K_THREAD_DEFINE(thread_a, STACKSIZE, thread_entry_point, "a", NULL, NULL, PRIORITY, 0, 0);
K_THREAD_DEFINE(thread_b, STACKSIZE, thread_entry_point, "b", NULL, NULL, PRIORITY, 0, 0);
```

- Compile for any board using `west build` and flash using `west flash`

**Expected behavior**
- Scheduler shall re-schedule thread b and a with equal time slice

**Impact**
- Trying to bisect back until v2.7.0, but still the problem exists

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.7.0  ***
a 
a
a
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit 6e9cfcdcd5272c2b8bd9954099f4a74b89a3421e or as old as 3f826560aaf81a444018293bd6acce3c339fe150

carlescufi, 2025-05-14:
I can reproduce with:
`west build -b qemu_x86 samples/hello_world -- -DCONFIG_TIMESLICING=y`

peter-mitsis, 2025-05-14:
I too have been able to reproduce this. However, the reason is that the Kconfig option CONFIG_TIMESLICE_SIZE was left with its default value (0). When the size of the time slice is 0, time slicing is disabled.

To achieve the desired behavior, the developer must perform at least one of the following two options:

1. At build time, ensure CONFIG_TIMESLICE_SIZE is set to a non-zero positive value.
2. At run time, call k_sched_time_slice_set() to configure the time slice details.

Digging further, I observe that the thread documentation mentions the following time slicing Kconfig options:
 
* CONFIG_TIMESLICING
* CONFIG_TIMESLICE_SIZE
* CONFIG_TIMESLICE_PRIORITY

However, it may be worthwhile to reference those in the time slicing documentation as well to prevent a future misunderstanding.

andyross, 2025-05-14:
Did we break that?  I swear the defaults used to work to just flick timeslicing on without needing to tune the quantum, no?

andyross, 2025-05-14:
Or more strongly: if the default timeslicing configuration doesn't timeslice when timeslicing is enabled, we have a bug.

hasheddan, 2025-05-14:
@andyross I've been tracking down some behavior that I believe originates from this change as well. Went ahead and opened https://github.com/zephyrproject-rtos/zephyr/pull/89970.

peter-mitsis, 2025-05-14:
Also created #89974 to make those time slice Kconfig options more accessible to those reading the time slicing documentation.

PR #89970 (2025-05-16): kernel: sched: default TIMESLICE_SIZE to 2
Sets the default for CONFIG_TIMESLICE_SIZE to 2 to match the historical clamping behavior that was removed in [405611d](https://github.com/zephyrproject-rtos/zephyr/commit/405611dc9e0edb93b6f05bbc1597024fe34be0b7).

Fixes #89965
closed 2025-05-16 by kartben