zephyrproject-rtos/zephyr #89968: STM32: UART: 7-bit mode is broken

jameswalmsley, 2025-05-14:
**Describe the bug**
The UART driver seems to have implemented the function: `uart_stm32_cfg2ll_databits()` incorrectly.
I have been trying to configure my UART with 7-e-1 (7 data bits, even parity and 1 stop bit).

I've found it impossible to make the stm32 uart work in this mode with the zephyr driver. However if I force 7-bit data in the register then it works correctly.

I.e uart_stm32_cfg2ll_databits() returns 0, but it really should return LL_USART_DATAWIDTH_7B.

My dts:

```
&usart1 {
    stop-bits = <1>;
    parity = "even";
    data-bits = <7>;
};
```

jameswalmsley, 2025-05-14:
See PR:
https://github.com/zephyrproject-rtos/zephyr/pull/89972

erwango, 2025-05-15:
Without going into the details, it seems this behavior was specifically fixed and verified by multiple users in https://github.com/zephyrproject-rtos/zephyr/pull/33555.
Can you have a check ?

jameswalmsley, 2025-05-15:
Thanks for the response @erwango 
I will try to setup some tests for all the cases with some nucleo boards I have.

I suspect this might be a difference in how 7-bit mode is handled.

Is it possible there's a difference between the different series? Although I noticed from the other issues they had tested on f746 which is the same as the f750.

closed 2025-09-14 by github-actions[bot]