zephyrproject-rtos/zephyr #89979: LVGL

AravindPosina, 2025-05-15:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

AravindPosina, 2025-05-15:
Hi 
 I'm using Renesas Board of EK_RA8D1 . Here the issuse was the LVGL of MIPI_DSI was not working properly, the flashing procesure and all working working fine , but there was no expected output that we are not getting and also the touch functionality was not working (effectively). And also one more thing that whenever i was enabled the UART1 in the dts the LVGL was flashing but not responding in the board .

The environmental setup for the was as follow,
Zephyr version -4.1.0-rc2
Zephyr SDK version -  0.17.0

erwango, 2025-05-15:
Thanks for raising this issue, but please:
- edit title to make it more specific
- edit the bug template to fill your information. Also adding clear steps for reproduction.

closed 2025-05-20 by dkalowsk