zephyrproject-rtos/zephyr #89990: STM32: USB: USB CDC IN Endpoint stuck forever after receiving Clear feature - endpoint halt request

ptrcr, 2025-05-15:
**Describe the bug**

I have a stm32wb55rg based device with enabled USB CDC, that is connected to PC over a USB hub. After the device reset and successful enumeration in host system, application is opening a port to communicate with the device over CDC. During enumeration process, the device schedules [sending the ZLP packet](https://github.com/zephyrproject-rtos/zephyr/blob/cab2f9d27f3e433d274fd18a6418da4af6dc0920/subsys/usb/device/class/cdc_acm.c#L400).
Normally, after opening a serial port, the host starts polling the In Endpoint right away, reads the ZLP and communication proceeds without any issues. But sometimes before the host starts polling it sends 'Clear feature - endpoint halt' request first, and then starts polling the IN Endpoint. 
This request is handled by calling [usb_dc_ep_clear_stall](https://github.com/zephyrproject-rtos/zephyr/blob/cab2f9d27f3e433d274fd18a6418da4af6dc0920/subsys/usb/device/usb_device.c#L1080). Underneath, it causes [setting the endpoint's status to NAK](https://github.com/zephyrproject-rtos/hal_stm32/blob/6e4716f8ac2c6e6159a958a836e1dcca06804456/stm32cube/stm32wbxx/drivers/src/stm32wbxx_ll_usb.c#L662), which effectively cancels any pending transfer without releasing associated resources, that is the queued ZLP packet transfer and endpoint's write semaphore. As a result, it is impossible to schedule any new transfer, because there is an unfinished transfer that due to NAK status will never be completed.

**To Reproduce**
This situation seems to happen randomly from around once per 5 to once per 100 following cycles:

1. Reset the device
2. Open serial port from host application
3. Send any message from the device to host over CDC

**Expected behavior**
Message sent from the device to the host should be read by the application that opened the serial port.

**Impact**
Unable to perform two way communication with the device. Messages from the host to the device are sent successfully.

**Logs and console output**
The following screenshot presents logs from the described situation. The host sends request b'0201000082000000' and starts polling IN endpoint, but the status is NAK forever, ZLP packet is not read.
[image]

**Environment (please complete the following information):**
I was only able to reproduce this bug on a PC with Debian GNU/Linux 11 bullseye 5.10.0-28-amd64 and the device connected over a USB hub (reproduces with a few different USB hubs). 

**Additional context**

ptrcr, 2025-05-15:
I opened a PR that works for me: https://github.com/zephyrproject-rtos/zephyr/pull/89982

jfischer-no, 2025-05-15:
> But sometimes before the host starts polling it sends 'Clear feature - endpoint halt' request first, and then starts polling the IN Endpoint.

It would not happen without reason. Your tracelog does not show what is really going on, please add a tracelog since the bus reset or Set Configuration request. Kernel cdc-acm driver does not randomly call usb_clear_halt(acm->dev, acm->in).

jfischer-no, 2025-05-20:
@ptrcr Thanks.
Are you using `samples/subsys/usb/cdc-acm` sample to reproduce it? If not, could you please try?
On the host side, could you please enable debug messages if possible (`echo 'file cdc-acm.c +p' > /sys/kernel/debug/dynamic_debug/control`) and post them here? Thanks.

Checking if the endpoint is halted should prevent the lock you observed. Could you please test the change below?

```diff
diff --git a/drivers/usb/device/usb_dc_stm32.c b/drivers/usb/device/usb_dc_stm32.c
index 16b91470dee..4fcefd7a129 100644
--- a/drivers/usb/device/usb_dc_stm32.c
+++ b/drivers/usb/device/usb_dc_stm32.c
@@ -852,6 +852,10 @@ int usb_dc_ep_clear_stall(const uint8_t ep)
                return -EINVAL;
        }
 
+       if (!ep_state->ep_stalled) {
+               return 0;
+       }
+
        status = HAL_PCD_EP_ClrStall(&usb_dc_stm32_state.pcd, ep);
        if (status != HAL_OK) {
                LOG_ERR("HAL_PCD_EP_ClrStall failed(0x%02x), %d", ep,
```

ptrcr, 2025-05-21:
@jfischer-no I was able to reproduce it using `samples/subsys/usb/cdc-acm` on _nucleo_wb55rg_ board just with minor changes that made it easier to reproduce it (if 'R' is sent to the device, it performs a reset after 250ms):
[cdc-acm-sample.patch](https://github.com/user-attachments/files/20370263/cdc-acm-sample.patch)

I enabled debug messages you asked for and the logs when CLEAR HALT is sent are following:
[clear_halt_cdc_acm.txt](https://github.com/user-attachments/files/20370245/clear_halt_cdc_acm.txt)

And for a reference, logs when there is no CLEAR HALT:
[port_open_ok_cdc_acm.txt](https://github.com/user-attachments/files/20370287/port_open_ok_cdc_acm.txt)

Regarding the change you proposed, I tested it and verified that it solves the issue. However I have concerns regarding scenario when the host sends SET FEATURE ENDPOINT HALT and then CLEAR FEATURE ENDPOINT HALT. I think it would also result in endpoint lockup.

ptrcr, 2025-05-23:
@jfischer-no Unfortunately I am able to reproduce it only on this particular machine with this kernel version. The fix you proposed is working on my setup, so I updated the related PR: https://github.com/zephyrproject-rtos/zephyr/pull/89982

PR #89982 (2025-06-02): drivers: usb: Prevent from perpetual locked state
The USB CDC driver is unable to process any bulk IN transfers after receiving 'Clear Feature - Endpoint Halt' request from host due to perpetual locked state caused by previously scheduled transfer, that will never be finished, as the endpoint's state is set to NAK. Fix by cancelling the transfer and releasing the resources blocking from scheduling next transfer upon receiving 'Clear Feature - Endpoint Halt' request.

Fixes #89990
closed 2025-06-02 by kartben