zephyrproject-rtos/zephyr #89994: tests: drivers: spi: spi_controller_peripheral fails at nrf54h20dk

nordic-piks, 2025-05-15:
**Describe the bug**
Tests 
- samples/boards/nordic/spis_wakeup/sample.drivers.spis.wakeup - nrf54h20dk@0.9.0/nrf54h20/cpuapp
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.pm_runtime- nrf54h20dk@0.9.0/nrf54h20/cpuapp
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.pm_runtime- nrf54h20dk@0.9.0/nrf54h20/cpuppr
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.pm_runtime- nrf54h20dk@0.9.0/nrf54h20/cpurad
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_4MHz- nrf54h20dk@0.9.0/nrf54h20/cpuapp
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_4MHz- nrf54h20dk@0.9.0/nrf54h20/cpurad
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_8MHz- nrf54h20dk@0.9.0/nrf54h20/cpuapp
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_8MHz- nrf54h20dk@0.9.0/nrf54h20/cpuppr
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_8MHz- nrf54h20dk@0.9.0/nrf54h20/cpurad
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_fast- nrf54h20dk@0.9.0/nrf54h20/cpuapp
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_mode0- nrf54h20dk@0.9.0/nrf54h20/cpuapp
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_mode0- nrf54h20dk@0.9.0/nrf54h20/cpuppr
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_mode0- nrf54h20dk@0.9.0/nrf54h20/cpurad
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_mode1- nrf54h20dk@0.9.0/nrf54h20/cpuapp
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_mode1- nrf54h20dk@0.9.0/nrf54h20/cpuppr
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_mode1- nrf54h20dk@0.9.0/nrf54h20/cpurad
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_mode2- nrf54h20dk@0.9.0/nrf54h20/cpuapp
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_mode2- nrf54h20dk@0.9.0/nrf54h20/cpuppr
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_mode2- nrf54h20dk@0.9.0/nrf54h20/cpurad
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_mode3- nrf54h20dk@0.9.0/nrf54h20/cpuapp
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_mode3- nrf54h20dk@0.9.0/nrf54h20/cpuppr
- tests/drivers/spi/spi_controller_peripheral/drivers.spi.spi_mode3- nrf54h20dk@0.9.0/nrf54h20/cpurad

**To Reproduce**
Steps to reproduce the behavior:
1. haven rf54h20dk@0.9.0/nrf54h20/cpuapp connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/drivers/spi/spi_controller_peripheral -p nrf54h20dk@0.9.0/nrf54h20/cpuapp --device-testing --device-serial /dev/ttyACM0 -v --inline-logs --west-flash="--erase"
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v4.1.0-3987-g6887b58d4e32 ***
Running TESTSUITE spi_controller_peripheral
===================================================================
START - test_basic
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.17.0
 - Commit SHA or Version used: `v4.1.0-3987-g6887b58d4e32`

nordic-piks, 2025-05-15:
Found after #89602

bjarki-andreasen, 2025-05-15:
issue is identified, working on fix

nordic-piks, 2025-05-21:
Fixed by #90047

closed 2025-05-21 by nordic-piks