zephyrproject-rtos/zephyr #90029: testsuite: boards: unit_testing: 32-bit time_t incompatible with all other platforms

cfriedt, 2025-05-15:
**Describe the bug**

While creating some additional utilities for dealing with time structures, and using the `unit_testing` board (since that is the least expensive of all of our test platforms), I was triggering an unexpected assertion.

Namely:

```cpp
BUILD_ASSERT(sizeof(((struct timespec *)0)->tv_sec) == sizeof(int64_t), "tv_sec must be 64-bit");
```

Just to be sure that this was out of the ordinary, I created a testsuite to check that `time_t` was 64-bit on all other platforms (see 'Additional Context'), and sure enough, every other architecture that is supported by Zephyr supports a 64-bit `time_t`, even `native_sim/native`, which is the 32-bit variant of `native_sim`.

I realize it's a bit tricky due to the current `-m32` CFLAG, and what that might imply for the host platform in terms of other standard C types if we were to move to `-m64` instead.

However, we typically do want to test both 64-bit and 32-bit platforms with our `unit_testing` platform, and for the most part, standard C types should be compatible between our `unit_testing` platform and all other Zephyr platforms.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? `unit_testing`
 - What have you tried to diagnose or workaround this issue? Compared with every other architecture supported by Zephyr
 - Is this a regression? Unsure

**To Reproduce**
1. apply patch below
2. `twister -i -p unit_testing -T tests/unit/timeutil`
3. See error
4. run contrived non-unit-testing testsuite below
5. See success (inconsistence) 

**Expected behavior**
That standard C types should be consistent between the `unit_testing` platform and all other platforms in Zephyr.

**Impact**
The impact to the utilities and testsuite that I'm working on is pretty big, since it should cover most possible values within the `[INT64_MIN, INT64_MAX]` range, but without the support of the `unit_testing` platform itself, 

**Logs and console output**
```shell
In file included from /home/cfriedt/zephyrproject/zephyr/include/zephyr/toolchain.h:52,
                 from /home/cfriedt/zephyrproject/zephyr/include/zephyr/sys/printk.h:11,
                 from /home/cfriedt/zephyrproject/zephyr/subsys/testsuite/ztest/include/zephyr/ztest.h:37,
                 from /home/cfriedt/zephyrproject/zephyr/tests/unit/timeutil/test_timespec.c:12:
/home/cfriedt/zephyrproject/zephyr/include/zephyr/toolchain/gcc.h:87:36: error: static assertion failed: "tv_sec must be 64-bit"
   87 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert((EXPR), "" MSG)
      |                                    ^~~~~~~~~~~~~~
/home/cfriedt/zephyrproject/zephyr/include/zephyr/sys/timespec_util.h:23:1: note: in expansion of macro ‘BUILD_ASSERT’
   23 | BUILD_ASSERT(sizeof(((struct timespec *)0)->tv_sec) == sizeof(int64_t), "tv_sec must be 64-bit");
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK v0.17.0
 - Commit SHA: 4512cbf509ccbe4e54a956d3f821664062c9c408

**Additional context**
basic `testcase.yml` for the `unit_testing` platform and the `test_timeutil.c` below (and added to `tests/unit/timeutil/CMakeLists.txt`).

`tests/unit/timeutil/test_timeutil.c`:
```cpp
BUILD_ASSERT(sizeof(((struct timespec *)0)->tv_sec) == sizeof(int64_t), "tv_sec must be 64-bit");
```

When the same `.c` file is run through a non-`unit_testing` testsuite with the `testcase.yml` file below, all other platforms pass.

`testcase.yaml`
```yaml
common:
  filter: not CONFIG_NATIVE_LIBC
  tags:
    - libc
  # 1 tier0 platform per supported architecture
  platform_key:
    - arch
    - simulation
tests:
  libraries.time_t.size: {}
  libraries.time_t.size.armclang_std_libc:
    toolchain_allow: armclang
    extra_configs:
      - CONFIG_ARMCLANG_STD_LIBC=y
  libraries.time_t.size.arcmwdtlib:
    toolchain_allow: arcmwdt
    extra_configs:
      - CONFIG_ARCMWDT_LIBC=y
  libraries.time_t.size.minimal:
    extra_configs:
      - CONFIG_MINIMAL_LIBC=y
  libraries.time_t.size.newlib:
    filter: TOOLCHAIN_HAS_NEWLIB == 1
    extra_configs:
      - CONFIG_NEWLIB_LIBC=y
  libraries.time_t.size.picolibc:
    tags: picolibc
    filter: CONFIG_PICOLIBC_SUPPORTED
    extra_configs:
      - CONFIG_PICOLIBC=y
```

cfriedt, 2025-05-15:
Just to unblock myself, I might move the test to a non-`unit_testing` testsuite, and add `native_sim/native` in `platform_exclude`.

It's also possible to filter using `filter: not CONFIG_NATIVE_APPLICATION` or `filter: not CONFIG_NATIVE_LIBC`, but I'm hoping there is another way to ensure that `time_t` is 64-bit everywhere.

cfriedt, 2025-05-17:
Ah, just realized this is a [known issue](https://blogs.gentoo.org/mgorny/2024/09/28/the-perils-of-transition-to-64-bit-time_t/) that is still [not completely solved](https://wiki.debian.org/ReleaseGoals/64bit-time) in Linux distributions.

cfriedt, 2025-05-26:
There are many good reasons for requiring a 64-bit `time_t`, but the only reason I can think of that the two platforms above do not use a 64-bit `time_t` is because `__USE_TIME_BITS64` is not set in when the compiler parses e.g. 

* `/usr/include/x86_64-linux-gnu/bits/types/time_t.h`
* `/usr/include/bits/types/time_t.h`
* `/usr/include/i386-linux-gnu/bits/types/time_t.h`

Of course, these are toolchain headers that just happen to exist on my Ubuntu build machine.

There is no guarantee that the same headers or macros exist on whatever computer, Linux distribution, etc, that happens to be building targeting `native_sim/native`.

This is a comment just to reiterate the points here https://github.com/zephyrproject-rtos/zephyr/issues/90029#issuecomment-2888514093.

It should still be noted that every other target architecture supported by at least the Zephyr SDK seem to use a 64-bit `time_t`, and given the reasoning for using a 64-bit `time_t`, it's probably best to ultimately require that also for `native_sim/native` and `unit_testing`.

aescolar, 2025-06-27:
@cfriedt as you mentioned, time_t being just 32bits with glibc for 32bit builds is a known issue, whose main derived problem is the Y2038 issue.
I would have certainly been happier if glibc defaulted already to setting time_t to be 64bits, and all the 32bit libraries distributed with the corresponding distros where rebuilt with it. But unfortunately that is not the case, and I can understand why they haven't bitten that bullet yet: ABI incompatibility with precompiled libraries.
I have considered long ago to either force glibc to 64bit time_t by default, or provide the option for users to select it (see #75567), but I was quite wary of going that way (specially about setting it by default).
My thinking being that a) an ABI mismatch would be really hard to diagnose for users, b) the Y2038 issue is not that close yet, and c) I expected glibc to eventually do the switch, which would mean the distros' libraries would be recompiled with it and therefore the ABI issue would be greatly minimized, and that switch of glibc's default would have increased awareness for users and the amount of information online.

That being said, I believe the issue how it is described here is not correct. The issue is not that unit_testing or the 32bit native_sim (when using glibc) have an "incompatible" time_t, but that time_t is defined by the C library, and we cannot really assume its size. (This would apply to the size of any C library type beyond stdint types). Or at the very least that we should really try to avoid assuming its size, and if we need to, do it very carefully.

Meaning, when we build with pico, or minimal we known we have a 64 bit time_t. But when building with another C library we are at that library's mercy for that given architecture.
Note that we support proprietary toolchains and C libraries, and other architectures beyond the ones in tree.
These, if they were using a 32bit signed time_t like glibc may have also not yet have adapter for the Y2038 issue, as many embedded systems will not be using time_t for anything.
We could also expect a C library for a 24bit DSP to have a time_t of 48bits which would not have any Y2038 issue.
(Being pedantic I don't even think we can assume the C library time_t to be signed or even an integer type, but I would not expect our code to avoid issues with a non integer time_t).

So I think that instead of forcing all our CI platforms to a 64bit time_t, we should avoid assuming the size of time_t, and be happy that this allowed us to realize this assumption was not holding for all C libraries. Otherwise I believe we will be creating a problem for some of our users.