zephyrproject-rtos/zephyr #90037: Fujitsu MB85RS64 device driver issue

KrishKittur, 2025-05-15:
**The bug**
When using the MB85RS64 and MB85RS512, RDID verification fails. The manufacturer's IDs are correct but the product IDs aren't. This appears to be a bug because in this example driver that worked -- https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/drivers/spi_fujitsu_fram/src/main.c#L88-L96 -- they use the same product IDs as the ones I observed.

More information: 
- Target platform: STM32f429Xi
- What I've tried: Looking for more justification around why the product IDs chosen in the zephyr driver were chosen, but couldn't find any. 
- Regression? Doesn't appear to be.

**To Reproduce**
```
&spi2 {

    status = "okay";

    fram: MB85RS64-FRAM@0 {
        compatible = "fujitsu,mb85rsxx";
        reg = <0>;
        size = <8192>; 
        spi-max-frequency = <20000000>; 
        status = "okay";
    };
};
```
Use device tree overlay with EEPROM_SHELL set to y. 

**Expected behavior**
Can correctly perform RDID verification

**Impact**
Doesn't allow me to use the driver. 

**Logs and console output**
```
<err> mb85rsxx: invalid device ID: 04 7F 03 02
```

**Environment (please complete the following information):**
 - OS: MacOS
 - Zephyr SDK
 - v4.1.0

**Additional Notes**
Got a modified version of the driver working locally, with the correct RDID IDs used. Could make a PR for this.

henrikbrixandersen, 2025-06-22:
This driver was added by @fkokosinski in https://github.com/zephyrproject-rtos/zephyr/pull/79910 - perhaps they can shed some light on the product IDs chosen?

fkokosinski, 2025-06-23:
Hi @KrishKittur, thanks for reporting this issue.

> What I've tried: Looking for more justification around why the product IDs chosen in the zephyr driver were chosen, but couldn't find any.

Do you mean these defines, especially the first byte: https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/eeprom/eeprom_mb85rsxx.c#L48-L49? The first byte is a mask, not a value (as described in the comment).

> <err> mb85rsxx: invalid device ID: 04 7F 03 02

As for this, I can’t seem to reproduce it. I’m testing this with the `myra_sip_baseboard` platform:

```
west build -p -b myra_sip_baseboard samples/subsys/shell/shell_module – -DCONFIG_EEPROM=y -DCONFIG_EEPROM_SHELL=yw
west flash
```

And in the UART shell I can do the following just fine:
```
uart:~$ eeprom write mb85rs1mt@0 0 0xaa 0xaa 0xaa 0xaa
uart:~$ eeprom read mb85rs1mt@0 0 0x100
```

closed 2025-09-08 by github-actions[bot]