zephyrproject-rtos/zephyr #90055: Port ADB (Android Debug Bridge) client and server into Zephyr

ngphibang, 2025-05-16:
## Introduction

Enabling ADB support in Zephyr would help us to work across Android and Zephyr platforms, open the door to new applications and usecases.

### Problem description

In some applications, we need to communicate between Zephyr and Android. An example is the demo below:

https://www.youtube.com/watch?v=c6CALZsSDW8

This is working as a "demo" but not as a real-world application. The reason is:

- To control the phone, we need to inject touch events into Android system which require privileged permissions. This can be obtained by launching the Android application in a shell with adb. However, without adb support in Zephyr, we need to use a (Linux) laptop to launch the application which makes it impractical for use in real-world scenerio.

### Proposed change

- Porting of adb into Zephyr RTOS.

## Detailed RFC

Porting of all adb features into Zephyr seems very hard (or maybe impossible). We first need some basic things like:

- Authentication the device with the phone when plugging
- Pushing data (application) into the phone
- Launching application (in a shell) 

### Dependencies

- Android
- USB subsystem
- TCP/IP

### Concerns and Unresolved Questions

- Any comment and help on difficulties / feasibilities as well as interests on this work is welcome.

## Alternatives

- For the above usecase, without adb, we can customize Android but it will be hard and closed-source.
- For other usecases, adb seems is always practical.

josuah, 2025-05-19:
For the server part, here is where to keep track of the progress on Zephyr USB Host support: https://github.com/zephyrproject-rtos/zephyr/issues/42066

For the client part, this looks like a convenient alternative to `west` to keep managing after it booted, from a tool [packaged everywhere](https://repology.org/project/android-tools/versions) and well-known by developers.

This can be an interesting combo with runtimes like [LLEXT](https://docs.zephyrproject.org/latest/services/llext/index.html) or [WASM](https://github.com/bytecodealliance/wasm-micro-runtime/blob/main/product-mini/platforms/zephyr/simple/README.md) and [MicroPython](https://micropython.org/) (though they use [mpremote](https://docs.micropython.org/en/latest/reference/mpremote.html)) to keep managing the application after it booted, and allow basic communication with the application (i.e. send a shell command to capture an image from a camera to a file and download it then check the logs...).

This can also be a convenient way to load application assets such as TensorFlow models or configuration.

raveious, 2025-05-20:
Some very helpful references to get the naming convention correct: https://android.googlesource.com/platform/system/adb/+/refs/heads/main/OVERVIEW.TXT

In terms of Zephyr, I think that a Android Debug Bridge Daemon (adbd) (aka the thing that runs on the Android phone) implementation can be developed for Zephyr. Interfacing with it can be done by either USB, TCP, or both. The main usecase for having ADBD running on an embedded system is the ability to debug and interact with it using existing ADB tooling. I've poked at this as a side project in my non-existent amounts of free time.

As for the client/server side, it might also be possible to do so, but might not provide as much utility as the daemon implementation would. The main usecase for this is to issue ADB commands to Android phones for things like automation. This might get a lot more difficult as many of the capabilities might just make more sense for a full Linux environment. You could create a subset of capability, like pushing files, starting apps, shell commands, but port forwarding and screen capture will take a lot of work.

ngphibang, 2025-05-20:
> In terms of Zephyr, I think that a Android Debug Bridge Daemon (adbd) (aka the thing that runs on the Android phone) implementation can be developed for Zephyr.

I think the adbd is on the Android side, on the phone. What we need to implement is the client and server on the Zephyr (side), the host.

> You could create a subset of capability, like pushing files, starting apps, shell commands, but port forwarding and screen capture will take a lot of work. It doesn't require a USB HOST API, since it could communicate with TCP, but a USB HOST implementation would greatly enhance the usability of the client/server implementation.

Port warding and screen capture is not needed, it's done in the application

raveious, 2025-05-20:
> I think the adbd is on the Android side, on the phone. What we need to implement is the client and server on the Zephyr (side), the host.

I've seen a few custom embedded Linux images include ADB implementations to give a developer all of the utilities that ADB provides (shell, file access, port forwarding, etc), and I think same value can be added to Zephyr.

closed 2026-09-19 by github-actions[bot]