zephyrproject-rtos/zephyr #90057: GPIO subsystem support for more than 32 pins per port

Ayush1325, 2025-05-16:
## Introduction

Current Zephyr driver API only supports 32 pins per port. May Ti chipsets (such as AM62x) have GPIO ports with more than 32 pins (MAIN_GPIO0 in AM62x is 92 GPIOs).

Thus, there needs to be a way to use these GPIO pins.

The Linux kernel driver already does support all the pins, so it can serve as a good starting point.

### Proposed change

I have divided the changes into multiple phases to make the transition as smooth as possible.

#### Phase 1

Add new function pointers to the `struct gpio_driver_api`. The Zephyr pin based GPIO functions such as `gpio_pin_get_raw` and `gpio_pin_set_raw` currently use `gpio_port_{set/get}_masked` internally.

I propose adding the following function pointers to `struct gpio_driver_api`

```
	int (*pin_set_raw)(const struct device *port, gpio_pin_t pin, bool value);
	int (*pin_get_raw)(const struct device *port, gpio_pin_t pin);
	int (*pin_toggle)(const struct device *port, gpio_pin_t pin);
```

The `gpio_pin_*` based functions will use these new driver API functions if defined. Else will fall back to the old way. 

There is no user-facing change in phase 1.

#### Phase 2

I do not have much idea what to do about the `gpio_port_*` functions. In case of Ti chipsets, you can generally perform operations on all pins in a single bank at once. So, if other vendor GPIO chipsets work similarly, we can introduce `gpio_port_v2_*` functions that also take the bank as input.

However, since I do not have experience outside of Ti chipsets, this needs more consideration.

MihiraMadhava, 2025-07-18:
@Ayush1325 , I think it's also a good time to think about the interrupt support here for the Davinci GPIO. there is an Interrupt per GPIO Bank. As per your offsets-based implementation there are 2 interrupts per offsets i.e., with the pins grouped. 

the interrupt enable per bank is a single register that sits right before the DIR01 register and the interrupt status is per "offset" that shows which pin triggered bank interrupt.

Ayush1325, 2026-09-12:
Am using gpio-proxy nodes now.

closed 2026-09-12 by Ayush1325