zephyrproject-rtos/zephyr #90070: doc: Documentation issue in 'boards/nxp/lpcxpresso55s69/doc/index'

Gerriko, 2025-05-16:
**Describe the bug**
When following either the generic get started guide, or the hello_world example as shown on the product page, I get the following message when running "west flash":

**The module for runner "nios2" could not be imported (No module named 'runners.nios2'). This most likely means it is not handling its dependencies properly. Please report this to the zephyr developers.**

<< Please describe the issue here >>
<< You may also want to update the automatically generated issue title above. >>

Please can you update the lpcxpresso55s69 product page to instruct what to do or what prerequisites to install. There might be some additional install requirements for NXP hardware.

**Environment**
Linux Mint OS. 21 (Vanessa)

* Page: `boards/nxp/lpcxpresso55s69/doc/index`
* Version: main
* SHA-1: de13103f0b67296b045978fd51836c9101514a0f

Gerriko, 2025-05-16:
To illustrate incorrect instruction in product documentation; if I then use `west flash --runner pyocd` instead of just `west flash` it works.

kartben, 2025-05-17:
The nios2 warning thing has just been fixed
W.r.t. link server, did you follow the instructions to install it https://docs.zephyrproject.org/latest/boards/nxp/lpcxpresso55s69/doc/index.html#configuring-a-debug-probe ?

Gerriko, 2025-05-17:
Thanks for the update. No I hadn't. The following text is full of jargon, which did not make sense to me at the time of reading.

> LinkServer is the default runner for this board. A debug probe is used for both flashing and debugging the board. This board is configured by default to use the integrated [MCU-Link Onboard Debug Probe](https://docs.zephyrproject.org/latest/develop/flash_debug/probes.html#mcu-link-onboard-debug-probe) in the CMSIS-DAP mode. To use this probe with Zephyr, you need to install the [LinkServer Debug Host Tools](https://docs.zephyrproject.org/latest/develop/flash_debug/host-tools.html#linkserver-debug-host-tools) and make sure they are in your search path. Refer to the detailed overview about [Application Debugging](https://docs.zephyrproject.org/latest/develop/debug/index.html#application-debugging) for additional information.

I like simple. If it said something like "west flash" will not work unless you have LinkServer installed, I would have paid attention.

I now have it installed and the `west flash` command now works.

DerekSnell, 2025-05-22:
> I now have it installed and the `west flash` command now works.

Hi @Gerriko ,
Since you resolved your issue by following the steps in the existing documentation, can you close this issue?  Thanks

closed 2025-05-22 by Gerriko