zephyrproject-rtos/zephyr #90084: Cellular sample fails to bring back L4 connection after modem restart

olalonde, 2025-05-16:
**Describe the bug**

After running the cellular sample, I am getting "L4 was not connected in time" after the modem restarts.

**To Reproduce**

Run the cellular sample.

**Expected behavior**

L4 should connect again after the modem is restarted.

**Logs and console output**

I tested with a [new driver (a76xx)](https://github.com/zephyrproject-rtos/zephyr/pull/90081). I tested both with and without a shutdown script.

https://gist.github.com/olalonde/17f56d92015cb9b08371209fb2ebb761#file-modem_cellular_a76xx_shutdown_script-log-L4999

https://gist.github.com/olalonde/2cd36af3502571906fa453942920f050#file-cellular_modem_a76xx-log-L5010

I doubt it's a bug with my driver as it's just a small modification of the existing simcom 7080 driver and it works fine up until the [modem is restarted](https://github.com/zephyrproject-rtos/zephyr/blob/2fc54ae3bb74cdb2d2bc81d9248a809276cd33e4/samples/net/cellular_modem/src/main.c#L362). The init chat script and dial chat scripts complete successfully. The issue is that afterwards nothing happens, whereas the PPP driver should be initiating a new session.

Can anyone try to run the cellular sample and see if L4 connects after the modem restarts?

**Environment (please complete the following information):**

 - OS: macos
 - Toolchain: west
 - Commit SHA or Version used: 2fc54ae3bb74cdb2d2bc81d9248a809276cd33e4

olalonde, 2025-05-17:
I was able to fix this issue in this commit: https://github.com/zephyrproject-rtos/zephyr/pull/90081/commits/8628eb68a32b9933aadac52fc3f748b04b7ed590

I am not 100% sure if it's the correct fix but it fixes this issue.

vytvir, 2025-05-19:
The `net_if_dormant_off()` should do nothing, because we never call `net_if_dormant_on(...)`.

The `net_if_carrier_off(...)` should be called when leaving CARRIER_ON state in `modem_cellular_carrier_on_state_leave(...)`. Basically, in the same state it was previously turned on. 

The `net_if_carrier_on(...)` is called in the CARRIER_ON state.

It would appear that `notify_iface_up(...)` in `net_if.c` is never called. The PPP driver expects that to start the LCP negotiation (see `ppp_open_async(...)`. 

As we are calling `net_if_carrier_on(...)`, we never set `net_if_dormant_on(...)` anywhere, there seems to be some other reason why this is happening. Could you please enable `CONFIG_NET_IF_LOG_LEVEL_DBG=y` so we understand the cause why?

olalonde, 2025-05-20:
> The net_if_carrier_off(...) should be called when leaving CARRIER_ON state in modem_cellular_carrier_on_state_leave(...). Basically, in the same state it was previously turned on.

Maybe that's the root issue? It is not called in `net_if_carrier_off ()` is not called in `modem_cellular_on_carrier_on_state_leave`.

https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/modem/modem_cellular.c#L1134

I will try that fix later,  see if it works.

olalonde, 2025-05-20:
Yep that also worked. I updated the PR with this fix instead.

https://github.com/zephyrproject-rtos/zephyr/pull/90081/commits/b1b57050bd7693317c3d1940e85f1b8da45532e2

vytvir, 2025-05-20:
I was looking at an older version of `main`:
```c
static int modem_cellular_on_carrier_on_state_leave(struct modem_cellular_data *data)
{
    modem_cellular_stop_timer(data);
    net_if_carrier_off(modem_ppp_get_iface(data->ppp));
    modem_chat_release(&data->chat);
    modem_ppp_release(data->ppp);
    return 0;
}
```
I'm not sure _when_ it was changed from that to:
```c
static int modem_cellular_on_carrier_on_state_leave(struct modem_cellular_data *data)
{
	modem_cellular_stop_timer(data);

	return 0;
}
```

olalonde, 2025-05-20:
It seems that this PR removed the body of the modem_cellular_on_carrier_on_state_leave function. It looks accidental: https://github.com/zephyrproject-rtos/zephyr/pull/89820

MarkusLassila, 2025-05-20:
The removal was not accidental. The idea is that when network is lost, then instead of directly closing the carrier, we would set the PPP connection dormant, which would cause it to be terminated. This would then give the driver a possibility to move to the correct DLCI channel for the dial script.

I'll take a look at this.

olalonde, 2025-05-20:
> The removal was not accidental. The idea is that when network is lost, then instead of directly closing the carrier, we would set the PPP connection dormant, which would cause it to be terminated. [...]

Ok got it. The problem seems to be that when the modem is restarted, it goes straight from `carrier on` to `init power off` so `net_if_carrier_off(modem_ppp_get_iface(data->ppp));` never gets called.

MarkusLassila, 2025-05-20:
Thanks. Accidental removal after all. Apologies. This should fix it:
https://github.com/zephyrproject-rtos/zephyr/pull/90214

@olalonde, I don't have suitable modem available, can you give it a try?

olalonde, 2025-05-21:
> Thanks. Accidental removal after all. Apologies. This should fix it: [#90214](https://github.com/zephyrproject-rtos/zephyr/pull/90214)
> 
> [@olalonde](https://github.com/olalonde), I don't have suitable modem available, can you give it a try?

Yep it does 👍

olalonde, 2025-05-21:
> Thanks. Accidental removal after all. Apologies. This should fix it: [#90214](https://github.com/zephyrproject-rtos/zephyr/pull/90214)
> 
> [@olalonde](https://github.com/olalonde), I don't have suitable modem available, can you give it a try?

Yep it works 👍

closed 2025-05-23 by kartben