zephyrproject-rtos/zephyr #90088: Not getting frame data using OV7670 with nucleo-f746zg on zephyr IDE

R-ch2030, 2025-05-17:
I am integrating Ov7670 camera with nucleo-f746zg using zephyr ide in vs code.
I referred capture sample code and created patch using drivers like Ov7670.c and video_stm32_dcmi_data.
In prj.conf i added configuration for video, dcmi.
I got successful logs for product id i.e 0x76, DMA started successfully, capture started.
But for video_dequeue code stuck after : Before K_FIFO_GET
Pixel formate is 176×144

henrikbrixandersen, 2025-05-19:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

R-ch2030, 2025-05-22:
Now I received the logs like Got Frame. But when tried to print this data. It's zeros (Null).

josuah, 2025-05-26:
@R-ch2030 thank you for trying!

One detail to watch for is that for DCMI, you need at least 2 video buffers enqueued in total. This might not have been in the the default config of the capture sample. But it seems like the default of [`CONFIG_VIDEO_BUFFER_POOL_NUM_MAX`](https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_VIDEO_BUFFER_POOL_NUM_MAX) values were suitable since you had these messages:
```
Enqueue video buffer [0]
Enqueue video buffer [1]
```

The OV7670 was not tested with the ST DCMI driver yet, only with the FRDM-MCXN947 AFAIK, so it is quite possible that some adjustments to either the OV7670 driver either the DCMI driver are required.

josuah, 2025-05-26:
Could you describe the setup a bit more?

-> Which patch did you apply to `ov7670.c`? (a git diff output would help)

-> Which module did you try? This one? https://www.olimex.com/Products/Components/Camera/CAMERA-OV7670/

-> Did you use the FPC connector of the board connected to an [ST-CAMS-OMV](https://www.st.com/en/evaluation-tools/b-cams-omv.html) adapter, or jumper wires?

-> Did you also try with 320x240? Maybe that was too high-res to fit in memory.

-> What do you see if you try to probe the OV7670 data/clock/hsync/vsync pins with an oscilloscope or logic analyzer?

Thanks!

R-ch2030, 2025-05-27:
Thank you for your response.
I'm able to retrieve the frame data in a buffer.
However, how can I convert this data into an image?
When I dump the binary memory and try to convert it to PNG or JPEG format, the image doesn't display correctly.

josuah, 2025-05-27:
> convert it to PNG or JPEG format

This is a good method, but mean you have to figure out the correct ffmpeg/... parameters correctly, which is error prone.
Feel free to paste the binary you extracted, as well as the `.png` file you obtained.

The method I typically use is USB Video: turn the board into a webcam. This is not merged yet but you can use it following the instructions at the bottom of the first post here:
<https://github.com/zephyrproject-rtos/zephyr/pull/76798>

zacck, 2025-06-04:
Hey @R-ch2030  can I ask for a minimal example of your overlay and project conf please?

closed 2025-09-17 by github-actions[bot]