zephyrproject-rtos/zephyr #90103: http_server: Wildcard resource matching reports incorrect path length

tpambor, 2025-05-18:
**Describe the bug**
When the HTTP server handles requests for resources defined with wildcard characters, the reported length of the matching path is incorrect. Instead of calculating the length of the requested URI, the server was incorrectly using the length of the wildcard resource definition itself.

**To Reproduce**
For a resource defined as `/foo/*`. If `/foo/bar` is requested the length returned is 6 and not 8.

**Expected behavior**
For a resource defined as `/foo/*`. If `/foo/bar` the length returned should be 8.

**Impact**
This bug fix ensures that the HTTP server correctly identifies the length of the matched path for wildcard resources. This resolves potential issues related to incorrect path length information being used in subsequent request processing.

**Logs and console output**

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 4.1

**Additional context**

tpambor, 2025-05-18:
This issue has been resolved in main branch by PR #86648. Wildcard resource support was introduced in v3.7. To fully address this bug and a related deprecation warning (present since v3.7 as well), please backport both commits from the pull request 33cf7dc78a976677f8693018eae90c0d0024d07b and 0f90affcdfd376f0360317d34b44ce85f28c4294 to v3.7, v4.0, and v4.1 branches. I would suggest backporting the entire pull request as the second commit resolves a deprecation warning that also affects these versions.

jukkar, 2025-06-19:
PRs related to this are already merged -> closing

PR #90155 (2025-05-28): [Backport v3.7-branch] net: http: server: The detail length of wildcard detail was wrong
Backport two commits from #86648

Fixes #90103
PR #90106 (2025-06-03): [Backport v4.1-branch] net: http: server: The detail length of wildcard detail was wrong
Backport 0f90affcdfd376f0360317d34b44ce85f28c4294~2..0f90affcdfd376f0360317d34b44ce85f28c4294 from #86648.
Fixes #90103
closed 2025-06-19 by jukkar