zephyrproject-rtos/zephyr #90107: sysbuild: impossible to include an overlay for all projects in a multi-image build

jameswalmsley, 2025-05-18:
**Describe the bug**
I'm trying to create a concrete sysbuild example including bootloader, factory image and main image to help people get started with zephyr, and especially using sysbuild.

Either I'm not understanding how it works, or sysbuild is missing some fundamental features!

My issue is I need to override the partition layout of a nucleo board, and I cannot do this without manually specifying the same overlay file to each project on the commandline. This seems like an anti-pattern.

Is there something I'm missing?

I was expecting that I'd be able to add a `sysbuild/boards/nucleo_f413zh.overlay` file or similar that would be picked up and applied to all the projects, but thats not the case.

**My command line**
```
west build --sysbuild sysbuild-example \
    -DDTC_OVERLAY_FILE=sysbuild-example/overlays/nucleo_f413zh_partitions.overlay \
    -Dmcuboot_DTC_OVERLAY_FILE=sysbuild-example/overlays/nucleo_f413zh_partitions.overlay \
    -Dmfg_image_DTC_OVERLAY_FILE=sysbuild-example/overlays/nucleo_f413zh_partitions.overlay \
    -Ddfu_app_DTC_OVERLAY_FILE=sysbuild-example/overlays/nucleo_f413zh_partitions.overlay
```

**Impact**
This makes the use of sysbuild rather limited. Sysbuild should support this scenario for companies using zephyr to build entire systems.

I'd be happy to provide a PR to make this work, if someone could give me some pointers.

nordicjm, 2025-05-19:
If you need to define a different partition layout then you should be extending the board instead https://docs.zephyrproject.org/latest/hardware/porting/board_porting.html#board-extensions or creating a new board which has the expected partition layout, this is not a sysbuild bug/problem and is by design

jameswalmsley, 2025-05-20:
Thanks for the pointer @nordicjm..

That should do what I need, I'll close this ticket.