zephyrproject-rtos/zephyr #90111: ble adv fails with -ENOMEM after disconnect

StefJar, 2025-05-18:
**Describe the bug**
The device boots and enables bt(peripheral role). After it starts advertising a connection with std. advertising and scann response data. The connection gets taken. The peripheral gets connected with the central. All good so far. Then the central terminates the connection(0x13 - REMOTE USER TERMINATED CONNECTION).
At the disconnected callback I try to restart the ble advertising  by calling `bt_le_adv_start`. This fails with -12(-ENOMEM)

**To Reproduce**
I have an own project. The ble code follows that example

https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/bluetooth/peripheral/src/main.c#L254

by adding `bt_le_adv_start` into the disconnect callback.

Guess that will prob do the job

**Expected behavior**
In general I am missing somehow some documentation how the different states are cycled through. Something like
adv - con - discon - adv ...
and how the peripheral implements that.

Imho a ble connection could be lost from various issues like temporally  out of range, EMC, etc. so connection recovery is one important issue to handle.

In my world the peripheral that can be connected always follows that simple states: advertising - connected - disconnected - back to advertising

So I won't expect an OOM error.

**Impact**
I don't like to reset my device at ble disconnect

**Logs and console output**
```
<dbg> BT: disconnected: Disconnected (reason 0x13)
<dbg> BT: adv_start: start advertising
<err> BT: Advertising failed to start (err -12)

```

**Environment (please complete the following information):**

 - OS: Ubuntu
 - Toolchain Zephyr SDK,
 - Commit SHA: b5934e8dfe5c8dd4c1a9f2221341aa46e3dce72a

tested against bluez & android 11 via bluetoothctr and nrfconnect

dottspina, 2025-05-19:
The connection object is not actually freed (re-usable) when the Bluetooth Host stack triggers the `disconnected()` callback.
To resume advertising, prefer the `recycled()` callback.

StefJar, 2025-05-19:
The bt core spec does not tell about a "recycled" BLE Link Layer State(Section: 4.1 - State Overview).

@dottspina thx for the comment - sounds like from your comment that disconnect callback has an still a "connection" object that will later gets recycled. Out of pure interest - when does this happens?

anyway calling  `bt_le_adv_start` from  `recycled` callback fixes that issue.

Maybe some examples could be updated to reflect that "workaround"?

@henrikbrixandersen should I close that issue?

closed 2025-08-12 by github-actions[bot]