zephyrproject-rtos/zephyr #90139: npcx9m6f & npcx4m8f : test: arch.arm.interrupt.no_optimizations fails

majunkier, 2025-05-19:
**Describe the bug**
Bug appear on two boards  :
npcx4m8f_evb/npcx4m8f
npcx9m6f_evb/npcx9m6f

**To Reproduce**
 twister -p npcx4m8f_evb/npcx4m8f -s arch.arm.interrupt.no_optimizations -ll debug

**Logs and console output**
```
*** Booting Zephyr OS build v4.1.0-4186-gb5934e8dfe5c (delayed boot 500ms) ***

START - test_arm_interrupt
Available IRQ line: 58
E: Unhandled IRQn: 58
E: >>> ZEPHYR FATAL ERROR 1: Unhandled interrupt on CPU 0
E: Current thread: 0x200c1b88 (test_arm_interrupt)
Caught system error -- reason 1
E: r0/a1:  0x00000003  r1/a2:  0xe000e100  r2/a3:  0x00000003
E: r3/a4:  0x00000003 r12/ip:  0x00000000 r14/lr:  0x100930bd
E:  xpsr:  0x2100004a
E: Faulting instruction address (r15/pc): 0x10090a42
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x200c1b88 (test_arm_interrupt)
Caught system error -- reason 3
E: r0/a1:  0x00000004  r1/a2:  0xe000e100  r2/a3:  0x00000004
E: r3/a4:  0x00000004 r12/ip:  0x00000000 r14/lr:  0x100930bd
E:  xpsr:  0x2100004a
E: Faulting instruction address (r15/pc): 0x10090a70
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x200c1b88 (test_arm_interrupt)
Caught system error -- reason 4
ASSERTION FAIL [0] @ WEST_TOPDIR/**/tests/arch/arm/arm_interrupt/src/arm_interrupt.c:215
	Intentional assert

E: r0/a1:  0x00000004  r1/a2:  0x000000d7  r2/a3:  0x00000001
E: r3/a4:  0x00000004 r12/ip:  0x00000000 r14/lr:  0x100a6c8f
E:  xpsr:  0x2100004a
E: Faulting instruction address (r15/pc): 0x100a6ce4
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x200c1b88 (test_arm_interrupt)
Caught system error -- reason 4
E: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0xff54c2de r14/lr:  0x00000000
E:  xpsr:  0x01000000
E: Faulting instruction address (r15/pc): 0x10090d66
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x200c1b88 (test_arm_interrupt)
Caught system error -- reason 2
 PASS - test_arm_interrupt in 0.163 seconds
===================================================================
START - test_arm_null_pointer_exception
E: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E:   Data Access Violation
E:   MMFAR Address: 0x200c483c
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x200c1b88 (test_arm_null_pointer_exception)
Caught system error -- reason 2
Was not expecting a crash

```
**Environment**:
OS: Ubuntu 22.04
Toolchain: zephyr-sdk-0.17.0
Zephyr version: v4.1.0-4186-gb5934e8dfe5c

katgiadla, 2025-05-19:
Similar symptoms are observed on `nrf52840dk/nrf52840` and `nrf52dk/nrf52832` (SHA: 6353ba88b6cd5c2969215d601947bd89f651375d)

anangl, 2025-05-22:
> Similar symptoms are observed on `nrf52840dk/nrf52840` and `nrf52dk/nrf52832` (SHA: [6353ba8](https://github.com/zephyrproject-rtos/zephyr/commit/6353ba88b6cd5c2969215d601947bd89f651375d))

#90344 fixes the problem on nRF platforms. @ssekar15 or @majunkier could you check if it also solves the issue for NXP platforms?