zephyrproject-rtos/zephyr #90146: Mesh crypto supports hardware ECC

tzy0002088, 2025-05-19:
There is currently no good way to connect to hardware crypto. For example, functions such as bt_mesh_dhkey_gen directly call software ecc. If I change this function, it seems to be a breaking change.

tzy0002088, 2025-05-19:
@alxelax @alwa-nordic @jhedberg FYI

alxelax, 2025-05-19:
Hi @tzy0002088, I'm not crypto library expert.

Nevertheless, at the current moment, mbedtls PSA is a kind of default crypto approach in Zephyr:
https://github.com/Mbed-TLS/mbedtls?tab=readme-ov-file
The library itself allows implementing custom drivers with hardware accelerations:
https://github.com/Mbed-TLS/TF-PSA-Crypto/blob/development/docs/psa-driver-example-and-guide.md

If you have very specific crypto hardware, then you need to implement such driver.
If you use some mcu chip, it is better to ask chip vendor if PSA driver is supported for mbedtls.
I'm quite sure some vendors already have such drivers.

The principal ability of adding hardware acceleration exists. This is mostly a question about supporting of exact HW.

alxelax, 2025-05-19:
> so why not just use the crypto drivers that are already registered in the kernel?

@tzy0002088 , by mistake, updated your comment. Hope everything has been restored back. Sorry

tzy0002088, 2025-05-19:
> > so why not just use the crypto drivers that are already registered in the kernel?
> 
> [@tzy0002088](https://github.com/tzy0002088) , by mistake, updated your comment. Hope everything has been restored back. Sorry

It's okay. Integrating hardware ECC with mbedtls feels a bit too heavy. I think it would be better to mark the functions in `crypto_tc.c` or `crypto_psa.c` as `weak`, so that users can provide their own implementations.

closed 2026-09-19 by github-actions[bot]