zephyrproject-rtos/zephyr #90166: static random address is not being used despite being set

tailoredsoft, 2025-05-19:
**Describe the bug**
I am trying to enhance the iso_broadcast sample app to use static random address so that it always uses the same address to help with sniffing. When it is not static the sniffer ends up with another device each time I reset the device.

Despite specifying a static random address as per the following code snippet, on-air it is not as I specify and for example I see an address value :- 13:41:8A:E3:51:88

`
static const bt_addr_t static_addr = {{0xC0, 0xDE, 0xCA, 0xFE, 0xBA, 0xCE}};
bt_addr_le_t addr_le = {
	.type = BT_ADDR_LE_RANDOM,
	.a = static_addr
};

int id = bt_id_create(&addr_le, NULL);
if (id < 0) {
	printk("bt_id_create() failed: %d\n", id);
} else {
	printk("bt_id_create() succeeded. ID: %d\n", id);
}    

/* Initialize the Bluetooth Subsystem */
err = bt_enable(NULL);
if (err) {
	printk("Bluetooth init failed (err %d)\n", err);
	return 0;
}
`
**To Reproduce**
Add the code specified abve to main() 

**Expected behavior**
On-air static random address should be used given the startup log provided belo

**Impact**
minimal

**Logs and console output**
`*** Booting Zephyr OS build ROOT_FEATURE_BLE_ISO_GROUP_CHAT-201-gfb5fdab8ac27 ***
Starting ISO Broadcast Demo
bt_id_create() succeeded. ID: 0
[00:00:00.258,422] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.258,453] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.258,514] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 4.1 Build 99
[00:00:00.258,880] <inf> bt_hci_core: HCI transport: Controller
[00:00:00.259,002] <inf> bt_hci_core: Identity: CE:BA:FE:CA:DE:C0 (random)
[00:00:00.259,033] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1
[00:00:00.259,063] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff
Waiting for BIG complete chan 0...
ISO Channel 0x20000144 connected
ISO Channel 0x20000158 connected
BIG create complete chan 0.`

**Environment (please complete the following information):**
N/A

**Additional context**
Trying to make the sample app easier to use with a bluetooth sniffer

cvinayak, 2025-05-19:
Did you try with `BT_LE_ADV_OPT_USE_IDENTITY` ?

tailoredsoft, 2025-05-19:
No, I have not ☹
Simple reason is that I have not yet learned how to know if config key exists for something.
I will do so now.

Best Regards,
Mahendra Tailor
Tailored Technology Ltd

tailoredsoft, 2025-05-19:
Hi
Your suggestion of using BT_LE_ADV_OPT_USE_IDENTITY has solved the issue.
Thank you so much for the hint.

Best Regards,
Mahendra Tailor
Tailored Technology Ltd