zephyrproject-rtos/zephyr #90180: samples: wifi: shell: xiao_esp32c6 does not boot

mariopaja, 2025-05-19:
I recently rebased to latest commit and I noticed that  the board did not boot when flashed with this sample. 
Unfortunately, I didnt keep notice of the last working commit.

ESP32S3 & ESP32C3 are unaffected.

ESP32C6:
```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0x1e (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002f44c
SPIWP:0xee
mode:DIO, clock div:2
load:0x40800000,len:0x20788
load:0x408207a0,len:0x3af0
SHA-256 comparison failed:
Calculated: adba8169b59b120d6286b42217a15437e4ae3d0d64d0eaf3d42739220882c076
Expected: 0000000040bd0000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x408035e0
I (49) soc_init: ESP Simple boot
I (49) soc_init: compile time May 19 2025 20:24:48
I (50) soc_init: chip revision: v0.1
I (50) flash_init: SPI Speed      : 80MHz
I (52) flash_init: SPI Mode       : DIO
I (56) flash_init: SPI Flash Size : 4MB
I (59) boot: DRAM: lma 0x00000020 vma 0x40800000 len 0x20788  (133000)
I (65) boot: DRAM: lma 0x000207b0 vma 0x408207a0 len 0x3af0   (15088)
I (72) boot: IRAM: lma 0x000242b8 vma 0x00000000 len 0xbd40   (48448)
I (78) boot: IMAP: lma 0x00030000 vma 0x42800000 len 0x19c3c  (105532)
I (84) boot: IRAM: lma 0x00049c44 vma 0x00000000 len 0x63b4   (25524)
I (90) boot: IMAP: lma 0x00050000 vma 0x42000000 len 0x5eb20  (387872)
I (96) boot: Image with 6 segments
I (100) boot: IROM segment: paddr=00050000h, vaddr=42000000h, size=5EB1Eh (387870) map
I (107) boot: DROM segment: paddr=00030000h, vaddr=42800000h, size=19C40h (105536) map
I (115) boot: libc heap size 246 kB.
I (118) spi_flash: detected chip: generic
I (122) spi_flash: flash io: dio

```

dkalowsk, 2025-05-20:
@mariopaja can you please use the bug template?  Also have you ensured that you've run `west update`?

sylvioalves, 2025-05-20:
@mariopaja I tested with latest main using `esp32c6_devkitc` board and it works. Let me know if you get it or if there is indeed some issue with that specific board.

```
I (95) soc_init: ESP Simple boot
I (95) soc_init: compile time May 20 2025 13:14:52
I (96) soc_init: chip revision: v0.0
I (96) flash_init: SPI Speed      : 80MHz
I (98) flash_init: SPI Mode       : DIO
I (102) flash_init: SPI Flash Size : 8MB
I (105) boot: DRAM: lma 0x00000020 vma 0x40800000 len 0x208bc  (133308)
I (112) boot: DRAM: lma 0x000208e4 vma 0x408208d0 len 0x3ad0   (15056)
I (118) boot: IRAM: lma 0x000243c8 vma 0x00000000 len 0xbc30   (48176)
I (124) boot: IMAP: lma 0x00030000 vma 0x42800000 len 0x19cc8  (105672)
I (130) boot: IRAM: lma 0x00049cd0 vma 0x00000000 len 0x6328   (25384)
I (137) boot: IMAP: lma 0x00050000 vma 0x42000000 len 0x5e8a4  (387236)
I (143) boot: Image with 6 segments
I (146) boot: IROM segment: paddr=00050000h, vaddr=42000000h, size=5E8A4h (387236) map
I (154) boot: DROM segment: paddr=00030000h, vaddr=42800000h, size=19CD0h (105680) map
I (173) boot: libc heap size 246 kB.
I (173) spi_flash: detected chip: gd
I (173) spi_flash: flash io: dio

*** Booting Zephyr OS build 4.1.99 ***
uart:~$ wifi scan
Scan requested

Num  | SSID                             (len) | Chan (Band)   | RSSI | Security             | BSSID             | MFP     
1    | TP-LINK_Extender_8CB340          23    | 11   (2.4GHz) | -19  | WPA2-PSK             | 18:A6:F7:8C:B3:40 | Disable 
...
...
Scan request done
```

mariopaja, 2025-05-20:
@sylvioalves I was able to pin the issue down to  `zephyr,console = &usb_serial`  & `zephyr,shell-uart = &usb_serial` node.
I swapped `usb_serial` with `uart0` and then everything worked fine

closed 2025-05-20 by sylvioalves