zephyrproject-rtos/zephyr #90215: shell_print() drops messages when used with logger

tokangas, 2025-05-20:
**Describe the bug**

`shell_print()` has started silently dropping messages. We started seeing this in nRF Connect SDK main after the last Zephyr upmerge, when running tests in CI using nRF91 DKs. We're using the shell UART backend.

The issue seems to be regression from commit b0a0febe589662bddb4bfe40c0faaf5795812fb1, and it's connected to logger and shell log backend usage. The commit changes `k_mutex_lock()` timeout in `shell_vfprintf()` from `K_FOREVER` to 50 ms. While logs are being flushed, another thread may call `shell_print()`. Flushing takes some time, during which locking the mutex times out, and the message gets dropped here: https://github.com/zephyrproject-rtos/zephyr/blob/6bdd19bda32887330003c196ce4d9947ba1db01f/subsys/shell/shell.c#L1555

**To Reproduce**

The issue can be reproduced using a simple shell application which uses logger. I used the following code to reproduce it on a nRF9151 DK:

```
	for (int i = 0; i < 50; i++) {
		LOG_INF("logger %d", i);
		shell_print(sh, "shell_print() %d", i);
	}
```

The output is always missing one shell_print() message after the log buffer has been flushed (see below for the output).

**Expected behavior**

No messages are lost with `shell_print()`.

**Impact**

Severe impact on shell usage.

**Logs and console output**

```
shell_print() 0
shell_print() 1
shell_print() 2
shell_print() 3
shell_print() 4
shell_print() 5
shell_print() 6
shell_print() 7
shell_print() 8
shell_print() 9
[00:00:01.349,700] <inf> shell_test: logger 0
[00:00:01.352,203] <inf> shell_test: logger 1
[00:00:01.356,475] <inf> shell_test: logger 2
[00:00:01.360,076] <inf> shell_test: logger 3
[00:00:01.363,647] <inf> shell_test: logger 4
[00:00:01.367,248] <inf> shell_test: logger 5
[00:00:01.371,520] <inf> shell_test: logger 6
[00:00:01.375,122] <inf> shell_test: logger 7
[00:00:01.378,692] <inf> shell_test: logger 8
[00:00:01.382,263] <inf> shell_test: logger 9
shell_print() 11
shell_print() 12
shell_print() 13
shell_print() 14
shell_print() 15
shell_print() 16
shell_print() 17
shell_print() 18
shell_print() 19
[00:00:01.386,566] <inf> shell_test: logger 10
[00:00:01.436,798] <inf> shell_test: logger 11
[00:00:01.463,928] <inf> shell_test: logger 12
[00:00:01.468,200] <inf> shell_test: logger 13
[00:00:01.471,801] <inf> shell_test: logger 14
[00:00:01.475,372] <inf> shell_test: logger 15
[00:00:01.479,675] <inf> shell_test: logger 16
[00:00:01.483,276] <inf> shell_test: logger 17
[00:00:01.486,846] <inf> shell_test: logger 18
[00:00:01.491,119] <inf> shell_test: logger 19
shell_print() 21
shell_print() 22
shell_print() 23
shell_print() 24
shell_print() 25
shell_print() 26
shell_print() 27
shell_print() 28
shell_print() 29
[00:00:01.494,720] <inf> shell_test: logger 20
[00:00:01.544,921] <inf> shell_test: logger 21
[00:00:01.573,516] <inf> shell_test: logger 22
[00:00:01.577,087] <inf> shell_test: logger 23
[00:00:01.581,359] <inf> shell_test: logger 24
[00:00:01.584,960] <inf> shell_test: logger 25
[00:00:01.588,562] <inf> shell_test: logger 26
[00:00:01.592,834] <inf> shell_test: logger 27
[00:00:01.596,435] <inf> shell_test: logger 28
[00:00:01.600,708] <inf> shell_test: logger 29
shell_print() 31
shell_print() 32
shell_print() 33
shell_print() 34
shell_print() 35
shell_print() 36
shell_print() 37
shell_print() 38
shell_print() 39
[00:00:01.604,309] <inf> shell_test: logger 30
[00:00:01.654,479] <inf> shell_test: logger 31
[00:00:01.683,074] <inf> shell_test: logger 32
[00:00:01.686,676] <inf> shell_test: logger 33
[00:00:01.690,246] <inf> shell_test: logger 34
[00:00:01.694,519] <inf> shell_test: logger 35
[00:00:01.698,120] <inf> shell_test: logger 36
[00:00:01.702,392] <inf> shell_test: logger 37
[00:00:01.705,993] <inf> shell_test: logger 38
[00:00:01.709,564] <inf> shell_test: logger 39
shell_print() 41
shell_print() 42
shell_print() 43
shell_print() 44
shell_print() 45
shell_print() 46
shell_print() 47
shell_print() 48
shell_print() 49
[00:00:01.713,867] <inf> shell_test: logger 40
[00:00:01.764,068] <inf> shell_test: logger 41
[00:00:01.791,931] <inf> shell_test: logger 42
[00:00:01.796,203] <inf> shell_test: logger 43
[00:00:01.799,804] <inf> shell_test: logger 44
[00:00:01.804,077] <inf> shell_test: logger 45
[00:00:01.807,678] <inf> shell_test: logger 46
[00:00:01.811,248] <inf> shell_test: logger 47
[00:00:01.815,551] <inf> shell_test: logger 48
[00:00:01.819,122] <inf> shell_test: logger 49
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK v0.17.0
 - Commit SHA 6bdd19bda32887330003c196ce4d9947ba1db01f

**Additional context**

tokangas, 2025-05-20:
Tagging @jakub-uC who authored the bug fix commit b0a0febe589662bddb4bfe40c0faaf5795812fb1 and @AaronFontaine-DojoFive who reported the original issue (https://github.com/zephyrproject-rtos/zephyr/issues/84274).

The current 50 ms timeout seems to be too short, because log processing can take longer when `CONFIG_SHELL_LOG_BACKEND` is enabled. Would it be possible to make the timeout considerably longer, while still being short enough to fix the original issue? Another option would be make the timeout configurable, but I think it would be hard for an user to figure out that this timeout needs to be increased to avoid dropped messages with `shell_print()`.

In our CI, we've seen that at least three consecutive messages get dropped. This probably means that the mutex can be locked in our case for more than 150 ms.

jakub-uC, 2025-05-21:
Thank you for spotting that side effect of the change — good catch!
I agree that making the timeout configurable is a good idea.
@tokangas, would you like to implement it? If not, I can take care of it in the next couple of days.

tokangas, 2025-05-21:
> I agree that making the timeout configurable is a good idea.
> @tokangas, would you like to implement it? If not, I can take care of it in the next couple of days.

@jakub-uC Sure, I can implement it. As this is presumably only related to using the shell log backend, maybe the default value could be higher when `CONFIG_SHELL_LOG_BACKEND` is enabled? It would be good if it could be several hundred ms or even 1s. That should ensure that no messages are lost when the shell log backend is used. And the user would not have to figure out that they need to change some Kconfig option to avoid message loss.

closed 2025-08-12 by github-actions[bot]