zephyrproject-rtos/zephyr #90224: drivers: video: internal video protocol

josuah, 2025-05-20:
## Introduction

A protocol to control a Zephyr video driver from Linux: a Linux V4L2 (Cortex-A) to Zephyr Video (Cortex-M) OpenAMP gateway.

### Problem description

Some video-capable SoCs have a Linux core and a Zephyr core. When the Linux core is full or to use fewer power, it can be tempting to offload more functions to the Zephyr core, and eventually turn off the Linux core.

**Example 1:**: algorithms performing auto-exposure/auto-focus/auto-white-balance can continuously be done on Zephyr every frame without bothering the Linux core busy with other computing. Linux which gets high-level controls such as color temperature rather than raw R/G/B gain, or auto-exposure strategy (MeteringCentreWeighted, MeteringSpot, MeteringMatrix).

**Example 2:**: only power a sensor in ultra-low-power mode like [Himax hm11b1](https://www.himax.com.tw/products/cmos-image-sensor/always-on-vision-sensors/hm11b1/) with a wake-up cascade: `sensor only` (motion detector) => `zephyr core` (wake-up decision) => `linux core` (full wake-up). In such case the Zephyr core can implement the video drivers and let the Linux core access it via a low-overhead internal protocol during runtime.

**Example 3:** A system with a Cortex-M4 and DSP core for processing implements the data processing on the DSP core but the control algorithms on the Cortex-M4 core. A simple control protocol is needed.

### Proposed change

Review existing protocols present on Linux. If none exist introduce a new one with implementation on both side.
Essentially provides the features of [UVC](https://fr.wikipedia.org/wiki/USB_video_device_class) without the overhead of USB.

## Detailed RFC

### Proposed change (Detailed)

Proposed design goals:

- Reduced scope, reuse existing OS components (Zbus, V4L2/Video API, libcamera, Gstreamer).
- Bi-directional: a single protocol for both direction concurrently.
- Low overhead, zero-copy via sharing memory pages for data, MCU-friendly control protocol.
- Allow control-only operation without video I/O,
- Self-describing data formats and discovery protocol for supported controls and capabilities.
- V4L2/[Libcamera](https://git.libcamera.org/libcamera/libcamera.git/tree/src/libcamera/control_ids_core.yaml) control IDs, fourcc for formats.
- Define a serialized version to export it over UART/TCP/... when the Linux and Zephyr part are two discrete chips.

Out of scope for initial implementation:

- Linux to Linux implementation, not part of Zephyr Project bug tracker,
- VirtIO driver between a VM and a Hypervisor, but keep use-case in mind as Zephyr can be used as i.e. Xen Dom0
- Generic IPC protocol, but it can be based on one like RPMsg/OpenAMP.

### Dependencies

A Linux and Zephyr proposal might need to be updated concurrently.

### Concerns and Unresolved Questions

- Is the overhead going to be too much?
- Is the use-case really there?

Only one way to find out: implement and try? :)

## Alternatives

- Run UVC or an internet-based protocol internally at the cost of higher overhead.

- Implement as a 3rd-party library on top of existing APIs.

josuah, 2025-05-20:
A good illustration would be a Video counterpart to https://www.sofproject.org/.
- A superset: not exclusive to DSPs offloading from Linux use-cases.
- A subset: start small and grow later once figuring out what needs extension.

josuah, 2025-05-21:
If the protocol is split in a high-level transport agnostic and a low-level transport (like USB/Bluetooth HID), it becomes possible to map the control protocol to i.e. I2C, I3C, UART, SPI... in addition to an IPC mechanism.

Chips running Zephyr can have automatic integration with Linux systems using a generic "video control" driver on both end. For example, a small MCU controlling a servo for optical zoom or P/T/Z controls.

A bit like MIPI CCS but based on Zephyr/Linux control IDs instead.

josuah, 2025-07-17:
A bit like MIPI CCS ~~but based on Zephyr/Linux control IDs instead.~~

Maybe exposing a MIPI-CCS interface directly is not such a bad idea, as this would allow using the MIPI-CCS Linux driver directly.

It seems like something Basler and Allied Vision are doing with their product line of industrial camera:
https://www.imveurope.com/news/mipi-alliance-releases-camera-command-set-specification

Being a register-based interface, this should work reasonably well between a Linux system and Zephyr system.
Linux-side and Zephyr side, this becomes a small "virtual I2C/I3C adapter" (or equivalent) when two cores are on the same chip: very small patch on both Linux and Zephyr side!

Then, if this is effectively two different chips, the "virtual I2C adapter" becomes an actual I2C/I3C adapter and nothing needs to be re-designed.

zacck, 2025-08-06:
> Then, if this is effectively two different chips (a Linux MPU and a Zephyr MCU/FPGA), the "virtual I2C/I3C adapter" is replaced by an actual I2C/I3C adapter and nothing else must be changed.

Hmmm, I like this part, perhaps this is a good experiment with the open mv cam