zephyrproject-rtos/zephyr #90239: Overhaul toolchain configuration and make ZEPHYR_TOOLCHAIN_VARIANT scalable

nashif, 2025-05-20:
`ZEPHYR_TOOLCHAIN_VARIANT` is now a mix of 

- SDK and Compiler environments, Toolchain collections, and
- Standalone Compilers

Instead, `ZEPHYR_TOOLCHAIN_VARIANT` should be used to specify the environment, product or collection of compilers.

Goal is not to introduce more complexity and inconsistency when we introduce LLVM into the Zephy SDK and to cleanup current usage with existing toolchains.

When a *variant* provides multiple compilers, those need to be selected using some other variable or other method with a preset default.

An example of current and confusing usage:

- `ZEPHYR_TOOLCHAIN_VARIANT`=host uses host gcc
- `ZEPHYR_TOOLCHAIN_VARIANT`=llvm uses host llvm

To fix this, use `ZEPHYR_TOOLCHAIN_VARIANT`=host for compilers provided by host with gcc set as the default compiler to use.

The following options should be considered:

1. Introduce a new variable (?) for selecting the compiler, i.e. `XXX_COMPILER`=<compiler>. In the host variant case, this can be either 
    - gcc, or 
    - llvm

2. Overload the `ZEPHYR_TOOLCHAIN_VARIANT` variable, so that `ZEPHYR_TOOLCHAIN_VARIANT`=zephyr-gcc will select gcc, `ZEPHYR_TOOLCHAIN_VARIANT`=zephyr-llvm will select llvm, `ZEPHYR_TOOLCHAIN_VARIANT`=zephyr will select a default.

LoveKarlsson, 2025-05-21:
IAR uses the `ZEPHYR_TOOLCHAIN_VARIANT` to set IAR usage, and another variable, `IAR_TOOLCHAIN_PATH`, to point to the toolchain. Cmake then figures out which one is used and sets everything up accordingly.

nashif, 2025-05-27:
> LLVM per nature is a cross-compiler, so setting variant to llvm is not only targetting `host`.

thats where the terminology is confusing. host variant is not targetting building for the host, i.e. building an ELF that can run on Linux (like we do with native_sim), this is merely saying we are using the host compiler, how we build and what we build for is up to us, depending on what the compiler is able to do.

You can have multiple llvm builds part of different variants, all able to build for the same target, i.e.  one provided by your OS (covered by host), one we provide as part of the zephyr SDK, another procided by ARM, etc.

tejlmand, 2025-06-06:
> In other words: host signifies toolchain provided by the host, not necessarily targeting building for the host.

but what defines "provided by the host".
If I have the toolchain in my path, then it can be considered to be provided by the host, and that means having Arm Toolchain for Embedded (ATfE) in my path means provided by host.

And what if user downloads and build llvm by himself, is that toolchain considered a host toolchain ?

If we are to keep the `host` as a toolchain variant, then we must decide what value `host` gives us, and why.

> one provided by your OS (covered by host)

and why is it important whether the toolchain was provided by your distribution ?

If it's because you want to avoid setting `<toolchain>_TOOLCHAIN_PATH`, then why not just have that if this setting is empty then we look for the tools in path.
Such as:
```
# Use tools in $PATH
ZEPHYR_TOOLCHAIN_VARIANT=llvm
```
and:
```
# Use tools from LLVM_TOOLCHAIN_PATH
ZEPHYR_TOOLCHAIN_VARIANT=llvm
LLVM_TOOLCHAIN_PATH=/opt/atfe
```

nashif, 2025-06-10:
> but what defines "provided by the host". If I have the toolchain in my path, then it can be considered to be provided by the host, and that means having Arm Toolchain for Embedded (ATfE) in my path [...]

any toolchain, nevermind the source, including the zephyr SDK is in your path and on your host. 'host' is what we have now used for `ZEPHYR_TOOLCHAIN_VARIANT` and it means compilers provided by the host, i.e. the compiler you use to build anything on that host or OS environment. Now,  such compilers might be suitable for building something like zephyr, i.e. you are not building for the host in that case, you are just using whatever compilers are offered and come as part of that host environment.

If we have to do that all over again, we probably would not have picked 'host', we continue to use it with a small tweak for compatibility. I do not mind changing this to something else.

ATfE or any other specialized compilers are their own variants, sure you can point to the binary and use ZEPHYR_TOOLCHAIN_VARIANT=llvm, but IMO this is going to be very confusing, because the VARIANT here becomes the compiler and when you set ZEPHYR_TOOLCHAIN_VARIANT=zephyr right now, zephyr is not a compiler. Thats why ZEPHYR_TOOLCHAIN_VARIANT=llvm does not fit at all and was moved under the host (or whatever we want to call it) variant.

> 
> And what if user downloads and build llvm by himself, is that toolchain considered a host toolchain ?

yes, unless it comes as part of an environment we target and support directly.
btw, this is usally the case, most distros have outdated llvm, so you have to go and download llvm yourself, but it installed just like any other package.

> 
> If we are to keep the `host` as a toolchain variant, then we must decide what value `host` gives us, and why.
> 
> > one provided by your OS (covered by host)
> 
> and why is it important whether the toolchain was provided by your distribution ?

It is not important, this is to distinguish between 'generic', general purpose toolchains, that are put under one umbrella (host) and things like the Zephyr SDK, or Metaware or Cadence toolchains or ARMclang or ATfE, which are supported as variants.

> If it's because you want to avoid setting `<toolchain>_TOOLCHAIN_PATH`, then why not just have that if this setting is empty then we look for the tools in path. Such as:
>
> ```
> # Use tools in $PATH [...]