zephyrproject-rtos/zephyr #90245: The serial port of the CH32V208 will lose data when it is received

Colt-Ma, 2025-05-21:
It may be necessary to increase the receive interrupt priority of the serial port. In addition, according to the description of WCH engineers, if the program runs outside the 128KB address range, the flash will not have zero waits, and the program will run ten times slower, so is there a way to ensure that all related programs involved in interrupts are linked to the 128KB address range

Colt-Ma, 2025-05-24:
When will the serial port support CONFIG_UART_USE_RUNTIME_CONFIGURE

VynDragon, 2025-11-30:
Tried adding the interrupt priority today, it seems that wasnt the issue, also put serial driver in RAM, this didnt fix it either. This may be a usart driver issue.

Colt-Ma, 2025-12-04:
> -----------------------------------
>
>
> I was only able to reproduce it running micropython, @Colt-Ma could you provide the code you used to reproduce?

The test code path is: zephyr/samples/drivers/uart/echo_bot

The baud rate is set to 115200

VynDragon, 2025-12-04:
I cannot reproduce, I tried this when testing and it didn't fail using echo_bot, not even a single time at that baudrate.

Colt-Ma, 2025-12-04:
I made some modifications to this program, counting the number of bytes received in the receive interrupt and outputting this number through the log. Then, using another device, I sent data to this UART port at full load. By comparing the number of bytes sent by the other device with the number of bytes received in the UART interrupt as shown in the log, errors can be seen after a few minutes.

-----------------------------------

Colt-Ma, 2025-12-04:
>>> Are characters '~' printed on the output?

Sorry, I do not understand what you mean.

 &gt; 收件人："zephyrproject-rtos/zephyr" ***@***.***&amp;gt;; 
 &gt; 抄送："Colt-Ma ***@***.***&amp;gt;;"Mention ***@***.***&amp;gt;; 
 &gt; lose data when it is received (Issue #90245) 
 &gt; ----------------------------------- 
 &gt; 
 &gt; 
 &gt; I cannot reproduce, I tried this when testing and it didn't fail using 
 &gt; echo_bot, not even a single time at that baudrate. 
 &gt; 
 &gt; — 
 &gt; ***@***.***&amp;gt; 
 &gt; 
 &gt; — 
 &gt; <https://github.com/zephyrproject-rtos/zephyr/issues/90245#issuecomment-3610384959&gt;, 
 &gt; . 
 &gt; ***@***.***&gt; 
 &gt;

SoftwareArchitekt, 2025-12-04:
I took a look at the test program.

I don't think it's suitable for the following reason:

If we constantly receive 1 byte, store it in a buffer, and then send this buffer within the interrupt, data loss can occur if sending takes longer than receiving 1 byte.

The solution: The buffer must be copied to another buffer after a `\n` command, and a flag must be set, which then triggers sending in a separate loop (outside the interrupt).

Therefore, this test program is only suitable for receiving data from a human because it's slow.

closed 2026-09-17 by github-actions[bot]