zephyrproject-rtos/zephyr #90253: Update UDP report sharing packet sending method

abhinavnxp, 2025-05-21:
**Is your enhancement proposal related to a problem? Please describe.**
 Failed to send UDP is printed if 2 report packets are missed by client.

**Describe the solution you'd like**
 Either count can be improved or ack can be added to these packets if protocol permits.

**Describe alternatives you've considered**
 NA

**Additional context**
 NA

rlubos, 2025-05-21:
I don't think we can add ACK since the protocol needs to be compatible with iperf. I guess we could make retransmission count configurable so users can increase this value in lossy networks.

abhinavnxp, 2025-05-21:
Yes it is correct we cant add ack since it is udp. Configuration option seems logical and appealing. Please help adding the change so I can test at my end.