zephyrproject-rtos/zephyr #90269: Build failure for rpi pico 2 after west update: cmsis_core.h not found

hanan619, 2025-05-21:
I was able to build Zephyr for the rpi pico 2 successfully. However, after performing a west update, the build now fails with the following error:

[image]

It seems cmsis_core.h was removed or not correctly linked after the update

nordicjm, 2025-05-21:
Do you have `cmsis_6` in your allow list?

shiftee, 2025-05-21:
This issue also happens on the Arduino Opta and starts with this commit: https://github.com/zephyrproject-rtos/zephyr/commit/f726cb5123e

metratec-mkoehler, 2025-05-22:
I have the same issue with a stm nucleo_f103rb dev board- 

I do not have cmsis_6 on my list but cmsis. I am a bit confused why a change would be needed.

nordicjm, 2025-05-22:
```
commit f726cb5123eda12feb8746e3acb34a1f7d7b0340
Author: Sudan Landge <sudan.landge@arm.com>
Date:   Tue Apr 29 15:05:15 2025 +0100

    modules: CMSIS_6: Switch to CMSIS_6 for Cortex-M
    
    What is the change?
    Switch to module CMSIS_6 for Cortex-M while continuing to use
    cmsis module (version 5.9.0) for Cortex-A/R.
    
    Why do we need this change?
    
    The current cmsis module consists of:
    - Cortex-M support from upstream cmsis 5.9.0
    - Cortex-A/R support which was not upstreamed to CMSIS
    
    Upstream cmsis 5.9 was deprecated so we should be using CMSIS_6 however,
    it seems due to lack of Cortex-A/R support in upstream and other reasons,
    this was pushed back.
    
    While upstreaming Cortex-A/R support to CMSIS_6 could take its time,
    this shouldn't stop Cortex-M to start using CMSIS_6.
    
    Also, if we do not use CMSIS_6 for Cortex-M then using the
    newer GCC 14.2 toolchain will return below compiler error:
    ```
    zephyrproject/modules/hal/cmsis/CMSIS/Core/Include/core_cm85.h:4406:10:
    fatal error: pac_armv81.h: No such file or directory
     4406 | #include "pac_armv81.h"
    ```
    Using CMSIS_6 for Cortex-M will fix this.
    
    Signed-off-by: Sudan Landge <sudan.landge@arm.com>
```
So this not an issue, this is expected, you need to add `cmsis_6` to your allow list

hanan619, 2025-05-22:
After adding cmsis_6, issue is resolved.
@nordicjm thanks for the clarification.
this issue can be closed

closed 2025-05-22 by hanan619

liquidhiter, 2025-06-10:
My local zephyr repo was on the 4.1 branch, running a `west update` after switching to the `main` branch resolved this issue.